LarsK1/hass_solvis_control #1: Sensors not loading + Cryptic error while adding IP

SevenNines, 2024-04-22:
The heating is already implementet via modbus, so it can't be a wrong modbusconfig.

In my opinion the scrennshots say everything, if you need more info let me know.

[image]

[image]

SevenNines, 2024-04-22:
Its Solvis Ben
[image]
[image]

LarsK1, 2024-04-22:
Okay, how's you're Ben connected (WiFi or LAN)? Please try to press the button under the control panel  on oyur solvis to turn it off, afterwards wait 10s, repress the button to turn it on. Then wait 5 minutes, usually it then works again.

SevenNines, 2024-05-01:
Hello,
sorry for the late reply.
Apparently only one connection is allowed at a time. After a restart, the sensors were supplied with measured values.
Thank you

LarsK1, 2024-05-02:
Hi,
thanks for confirming.

closed 2024-05-02 by LarsK1