LarsK1/hass_solvis_control #5: Frage zur aktuellen Solvis SC3 Version

Patrick762, 2024-06-21:
Hat hier noch jemand Probleme mit der aktuellen Version der SC3 und modbus? Es scheint fast so als ob Solvis die Schnittstelle geschlossen hat

DittelHome, 2024-06-22:
Na, so halb..
Eigentlich funktioniert noch alles, nur das auslesen der drei Register:
33536 Laufzeit Brennerstufe 1
33538 Laufzeit Brennerstufe 2
33537 Brennerstarts Stufe 1
funktioniert bei mir nicht mehr, seit MA3.19.47 ..

LarsK1, 2024-06-22:
@Patrick762 das Probleme tritt immer mal wieder auf. Ein Ausschalten der Heizung am Hauptschalter und ein anschließendes wieder einschalten hat bisher bei fast allen geholfen!

DittelHome, 2024-06-22:
Mh, bei mir half nur, die SC3 wieder mit der alten Firmware MA3.18.14. zu flashen.
Dann gibt es auch wieder die o.g Werte ..

closed 2024-07-30 by LarsK1