LarsK1/hass_solvis_control #6: Home Assistant failed to connect

ThomasWollstadt, 2024-06-26:
Latest version of  solvis control is installed.
HomeAssistant-Version latest is installed running on a PI4
Solvis VC3 is connected via LAN and communicates properly with Solvis portal. It is connected via a switch with my Fritzbox.
IP Address and port 502 is set properly in the configuration of Solvis Control

Nevertheless I get a failed to connect after reboot of heating system or Homeassistant every time.

Any idea where to search for the root cause?

LarsK1, 2024-06-27:
Hi @ThomasWollstadt ,
did you enable the modbus interface one the Solvis in the "Installateur"-Menu?

ThomasWollstadt, 2024-06-27:
… yes… 

 

Interestingly: I switched from LAN to WLAN connection on the SOLVIS side yesterday and now it works without any other adaptations… not sure how to inteprete this…

LarsK1, 2024-06-27:
Hi,
never experienced such an issue. I'm personally using my Solvis over LAN without any troubles...

closed 2024-06-27 by LarsK1