LarsK1/hass_solvis_control #7: [BUG] Sensor values are inverted

tinkerberlin, 2024-06-30:
**Version of the plugin**
1.0.1

 **Home-Assistant Version**
2024.7.0b5

**Enviroment (Docker, HAOS, ...)**
HAOS

**Describe the bug**
Some sensors are being pulled in inverted from the value being reported on the Solvis Control unit. 
For example: 
[image]
The value here is reported as nearly negative 70°C. In the SC unit directly the value is correctly reported at nearly positive 70°C.

The sensor of the example here is the "lower" ("unten") Pufferspeicher-Sensor. The "upper" ("oben") Sensor is correctly reporting the positive temperature value - as expected. 

**To Reproduce**
Not sure how to reproduce. This is simply how the sensor's value is being displayed for me...

**Expected behavior**
I would expect the integration to show the values as reported by the SC unit.

**Screenshots**
[image] (same screenshot as above)

LarsK1, 2024-06-30:
Hallo!
Danke für deine Meldung. Das ist ja lustig. Ich kenne diese Werte derweil in x-verschiedenen Varianten. Welches Modell von Heizung bzw. welche Firmwareversion hast du denn?

tinkerberlin, 2024-06-30:
Hallo! Danke für die schnelle Antwort.
Es handelt sich um die SolvisMax.
Ich glaube die FW Version ist 3.18.14 - kommt das hin?

HLueken, 2024-07-08:
Ich habe das gleiche Problem, mit "sensor.solvis_heizung_heizungspuffertemperatur_unten".
Ich habe Solvis Max mit Wärmepumpe Lea Premium, SC3, FW Version 3.19.47

LarsK1, 2024-07-09:
Bugfix is available in: https://github.com/LarsK1/hass_solvis_control/commit/07cee47fe7f7a52233543b7807cb0a8a83460704

LarsK1, 2024-07-09:
Gerne einmal mit dem neuen Release testen.

HLueken, 2024-07-09:
Dein Fix hat mein Problem mit sensor.solvis_heizung_heizungspuffertemperatur_unten bereinigt, der Wert passt jetzt!
Vielen Dank!

LarsK1, 2024-07-10:
@tinkerberlin kannst du bitte auch noch einmal testen und bestätigen, dass das Problem bei dir auch behoben ist?

LarsK1, 2024-07-30:
Hi @tinkerberlin,
nein. Leider hat uns schon ein Solvis-Techniker gesagt, dass die Modbus-Schnittstelle normalerweise nur im 30 Minuten-Takt genutzt wird. Da wir diese aber alle 30s abfrage kommt es immer mal wieder zu kleinen Problemen. Aber so häufig wie bei dir kann ich es nicht feststellen.

closed 2024-08-19 by LarsK1