LarsK1/hass_solvis_control #9: Values of heat pump

HLueken, 2024-07-30:
This software works great in my installation. I have installed a SolvisMax, SC-3, and a Lea Premium heat pump.

All values fron SolvisMax are reported, this is cool.

Would it be possible to show also values from the Lea Premium heat pump? Unfortunately I did not find any Modbus documentation for these values.

I would be interested in the following values:
- elektrische Energie Wärmepumpe
- Wärmemenge Wärmepumpe
- Startvorgänge Wärmepumpe
- Leistung Wärmepumpe
- Laufzeit Wärmepumpe
- Wärmemenge Heizstab
- Startvorgänge Heizstab
- Laufzeit Heizstab

LarsK1, 2024-07-30:
Hallo @HLueken ,
danke erst einmal für dein Lob. Werden den die Werte über das Solvis Portal angezeigt?
https://solvisportal.de/account

HLueken, 2024-07-30:
Werden sie, teils bei den Sensoren, teils im Überblick, teils bei den Bilanzen

[image]
[image]
[image]

kwithus, 2024-08-19:
Moin,
Ich hänge mich hier gerne an diesen Wunsch an, auch wenn es bei mir die Solvis Mia Wärmepumpe ist.

LarsK1, 2024-08-19:
Hi,
über Modbus ist das leider nicht realisierbar, da Solvis diese Daten (nach meinem Wissen) nicht freigibt. Es wird aber in der Zukunft eine V2 geben über das Solvis Portal, dort würde das dann gehen.

kwithus, 2024-08-19:
Ich habe einmal bei Solvis nachgefragt, ob und wenn ja wie die Daten der Wärmepumpe über Modbus abfragbar sind, werde dann berichten. (Wenn Solvis antwortet)

HLueken, 2024-08-24:
Wenn du irgendwann deine Lösung erweitern kannst (V2, Modbus, …), biete ich mich gerne zum Testen an).

kwithus, 2024-08-29:
Laut Solvis wird das Register derzeit überarbeitet. Mehr Auskunft war leider nicht zu bekommen

zefu59, 2024-09-03:
Die besonders interessanten Werte für Wärmepumpenbesitzer habe ich experimentell herausgefunden und mit dem Solvisportal verifiziert:
address=33544 => Wärmepumpenleistung
address=33545 => elektrische Wärmepumpenleistung
Hatte die auch schon mal erfolgreich in const.py eingebaut, aber aus Angst vor Update-Konflikten, mit dieser von mir sehr geschätzten HA-Integration, wieder entfernt. Vielleicht kann das ja ins nächste Update integriert werden ;-)

zefu59, 2024-09-03:
> Laut Solvis wird das Register derzeit überarbeitet. Mehr Auskunft war leider nicht zu bekommen

Ich hatte die auch mal angeschrieben und höflich gefragt - keine Antwort seit mehreren Wochen (muss man mindestens als unhöflich bezeichnen). In verschiedenen Foren habe ich über Monate verschiedene Berichte über nicht beantwortete Anfragen gelesen, ebenso (schon vor Monaten), dass man die Doku gerade überarbeitet ...

zefu59, 2024-09-03:
> Hi, über Modbus ist das leider nicht realisierbar, da Solvis diese Daten (nach meinem Wissen) nicht freigibt. Es wird aber in der Zukunft eine V2 geben über das Solvis Portal, dort würde das dann gehen.

Das Solvis Portal zu benutzen, setzt aber voraus, dass die Anlage eine permanente Internetverbindung hat. Das wollte ich persönlich gerne gerade durch den Einsatz des HA vermeiden. Für die Störungssuche oder Softwareupdates würde ich natürlich (temporär) den Internetkontakt erlauben.

LarsK1, 2024-09-03:
@zefu59 in welcher Einheit liegen denn die beiden Modbus-Werte vor. Bzw. in welcher Einheit sollte dieser Wert ausgegeben werden? So wie im PR?

zefu59, 2024-09-03:
@LarsK1 beide werden in kW gemessen (siehe "Übersicht" im Solvis-Portal, was ich zur Verifizierung der Adressen genutzt habe)

kwithus, 2024-09-03:
> Ich hatte die auch mal angeschrieben und höflich gefragt - keine Antwort seit mehreren Wochen (muss man mindestens als unhöflich bezeichnen). In verschiedenen Foren habe ich über Monate verschiedene [...]

Ja, Antworten dauern etwas, aber ich habe jeweils nach 1-2 Wochen eine Antwort bekommen, wenn auch nicht wirklich aussagekräftig ;-)

> Die Register für die Zeitfenster von WW (34174-34215) und Zirku (34216-34257), sowie ECO-Funktion (34258-34299) sind fehlerhaft. Somit gibt es Adressen die noch nicht zur Verfügung stehen und welche, die geprüft werden.
Es wird mit Sicherheit ein Release geben, wo das alles gefixt ist und auch vollständig ist. Die Prog. Prio liegt aktuell jedoch auf anderen Dingen, daher wird es dieses Jahr mit Sicherheit nicht mehr vervollständigt und getestet.

> 

Klingt aber zumindest für die Zukunft vielversprechend

closed 2024-09-03 by LarsK1

HLueken, 2024-09-21:
Ich habe es jetzt auch vor ein paar Tage installiert, scheint gut zu funktionieren. Zwei Änderungen waren notwendig in const.py: 
- Unit in kW geändert
- device_class in power geändert

Vielen Dank für den Einbau!!!

[image]

PR #15 (2024-09-03): Hinzufügen von Features für die Wärmepumpe
fixes #9