LarsK1/hass_solvis_control #10: [FEATURE] Ability to use own temperature sensors

sneekerzzz, 2024-08-11:
Add support for custom sensors, such as temperature sensors, for integration with Solvis systems.

In our case, we use our own sensors in Home Assistant rather than the thermostats provided by Solvis.

LarsK1, 2024-08-15:
Hi, 
sorry. This is currently not supported by Solvis.

So unfortunately not doable.

sneekerzzz, 2024-11-12:
Hi LarsK1,

This is definitely possible—I’m already using it this way. However, my goal is to use your integration, but that limits my ability to create custom automations and prevents me from writing temperature values to the register every 30 seconds.

Would there be any chance you could add this feature?

[image]

LarsK1, 2024-11-12:
Hi @sneekerzzz,
which registers need to be written? Or can you may be share your current code?

sneekerzzz, 2024-11-12:
> Hi @sneekerzzz, which registers need to be written? Or can you may be share your current code?

I'm not very tech-savvy :) but I managed to get it working with the automation code below. Solvis expects it every 30 seconds. I've replaced 3 Solvis thermostats with my own Philips Hue multi-sensors, and it's working great! 

`alias: Solvis | Temperatuur | Radiators
description: ""
triggers:
  - trigger: time_pattern
    seconds: /30
conditions: []
actions:
  - data:
      hub: SolvisRemote
      slave: 1
      address: 34305
      value: "{{ (states('sensor.hue_ms_livingroom_temperature') | float * 10) | int }}"
    enabled: true
    action: modbus.write_register
mode: single`

[image]

LarsK1, 2024-11-26:
Hi @sneekerzzz ,
thanks for providing that snippet. That should be implementable. Will check on the weekend and report back.

sneekerzzz, 2024-11-28:
> Hi @sneekerzzz , thanks for providing that snippet. That should be implementable. Will check on the weekend and report back.

That would be great! I’ve also contacted Solvis to confirm the timing for when the SC3 expects a temperature update. However, I’ve found that a consistent 10-second interval works perfectly.

LarsK1, 2024-12-29:
@sneekerzzz did you get a reply from Sovlis?

sneekerzzz, 2024-12-29:
> @sneekerzzz did you get a reply from Sovlis?

Unfortunately, not yet, but I’ve followed up on it and also asked our installer to look into this question. He has promised to get back to me with the information as soon as possible.

In the meantime, I’ve had the chance to run multiple tests, and anything between 5 to 10 seconds seems to be working fine.

sneekerzzz, 2025-01-08:
Hi @LarsK1 

I received the following response today from our installer, who had inquired again with Solvis:

They will not provide a time because Modbus is not a priority, and they are developing a new software system for EMS systems.

Therefore, I have not received an official timeframe.

As previously indicated, writing intervals between 5 and 10 seconds work well and do not cause errors on the SC3; I have been running this setup for a few weeks now.

LarsK1, 2025-02-23:
@sneekerzzz Please test with 2.0.0-alpha.9 once released

closed 2025-02-23 by LarsK1

sneekerzzz, 2025-02-23:
@LarsK1 

Cool, thanks for looking into it! Sorry, but how do I use this update to add my own temperature sensors?

fkhr79, 2025-02-23:
Just had a look into the new implementation: I think you (two) had a misunderstanding:

@LarsK1 implemented - as far as I understand it - that you now can set the number entitiy of the integration to a value you like. It's then written to the device via modbus.

@sneekerzzz might have wished to configure the integration via config_flow to directly use an entity which yields the temp value to write via modbus. - What do you think: can you live with the implementation as is?

LarsK1, 2025-02-24:
Hi @sneekerzzz ,
You'll need to add an automation, that set's the value of the room temperature according to your sensor.

fkhr79, 2025-02-25:
Hey again, @sneekerzzz ,
just thought to give you an automation example... and found a little bug while putting it together. 😇 
Will push a commit shortly which will fix the issue. After this you can plz test with either the next release or with the current main branch version (see instructions for manual installtion).

The following automation should work for you (I hope at least because I cannot test it in my setup):

```
alias: Set HKR1 Raumtemperatur Example
description: "Test the write feature of HKR1 Raumtemperatur - Solvis Control integration"
triggers:
  - trigger: time_pattern
    seconds: /5
    hours: "*"
    minutes: "*"
actions:
  - action: number.set_value
    data:
      value: >-
        {{ (states('sensor.ms_underfloor_heating_upstairs_temperature') | float
        * 10) | int }}
    target:
      entity_id: number.solvis_heizung_hkr1_raumtemperatur
mode: single
```

commit 7154cb28 (2025-02-23): Feat: restructured feature_checking and allow for and conditions ins entity feature check

Fixes #10
PR #168 (2025-02-25): Fix: hkr1_room_temperature write feature
- changed hkr1_room_temp to input_type numer and edit=True ind const.py
- added corresponding entries in icons.json and de.json
- fixed manual installation instructions in readme.md

fix: #10