LarsK1/hass_solvis_control #11: Frage zu Wert SollTemp HK1

kwithus, 2024-08-19:
Welcher Wert verbirgt sich hinter diesem Wert? Ich kann den angezeigten Wert (bei mir 50°) nicht zuordnen.

LarsK1, 2024-08-19:
Das ist die Solltemperatur des Heizkreises 1

larsneo, 2024-09-21:
Habe das gleiche Problem mit "2820 / HKR1 Solltemperatur Tag" (in der Kombi Solvis Max mit Mia 14) - da wird leider konstant 50C geliefert, Änderungen an der tatsächlichen Solltemperatur sind damit also leider nicht möglich.

JoeUrs, 2024-09-25:
Der Modbus Wert für die HKR1 Solltemperatur Tag ist die 2822.
Es hängt auch davon ab, ob du mehrere Temperaturen für bis zu 3 Tagen im SC-3 eingetragen hast.

larsneo, 2024-09-25:
@JoeUrs Der Wechsel des Modbus-Wertes hat perfekt funktioniert, vielen lieben Dank!

Beim Check muss man halt dran denken, dass die Solvisportal-App quasi nicht auf die extern gesetzten Werte reagiert, lediglich die Remoteschnittstelle direkt auf die SC-3 Schnittstelle zeigt die aktuellen Daten. Als Gegenstück zum bereits automatisierten Absenkmodus bei EXPENSIVE und VERY_EXPENSIVE kann ich mich damit nun an eine 1C-Erhöhung von HKR1 Tag und Absenktemperatur bei VERY_CHEAP machen :-)

larsneo, 2024-10-06:
@LarsK1 : danke für den commit, aber ich denke, dass beide adressen - je nach steuerung - ihre berechtigung haben. es wäre wahrscheinlich besser, neben den beiden fix-werten 2820 und 2821 die hkr1 (ggfs. auch hkr2 und hkr3)-werte für tag1/tag2/tag3 und abesnktemperatur zusätzlich aufzunehmen, um alle anforderungen abzudecken. letztere braucht man halt für die zeitprogramme.
[image]

kwithus, 2024-10-06:
Maybe we should add
2832 B10 HKR1 Heizkurve Steilheit 20 250 in 10tel
as well for those that do not set a fixed temperature but use the curve.

for HKR2 the respective addresses are

3076 C04 HKR2 Fix Temperatur Tag 5 75 °C
3077 C05 HKR2 Fix Absenktemperatur 5 75 °C
3078 C06 HKR2 Heizkurve Tag Temperatur 1 5 50 °C
3079 C07 HKR2 Heizkurve Tag Temperatur 2 5 30 °C
3080 C08 HKR2 Heizkurve Tag Temperatur 3 5 30 °C
3081 C09 HKR2 Heizkurve Absenktemperatur 5 30 °C
3088 C10 HKR2 Heizkurve Steilheit 20 250 in 10tel

And for HKR3

3332 D04 HKR3 Fix Temperatur Tag 5 75 °C
3333 D05 HKR3 Fix Absenktemperatur 5 75 °C
3334 D06 HKR3 Heizkurve Tag Temperatur 1 5 50 °C
3335 D07 HKR3 Heizkurve Tag Temperatur 2 5 30 °C
3336 D08 HKR3 Heizkurve Tag Temperatur 3 5 30 °C
3337 D09 HKR3 Heizkurve Absenktemperatur 5 30 °C
3344 D10 HKR3 Heizkurve Steilheit 20 250 in 10tel

I will try to implement those and add a pull request.

kwithus, 2024-10-06:
I do have an additional (slightly off topic) questions for those that use the fix temperature settings:
What kind of logic / automations to you use to set the temperature? 
What are the advantages compared to using a curve setting?

LarsK1, 2024-10-07:
Die Werte sind nun alle hinterlegt. Testet gerne einmal Version 1.1.0-rc1

larsneo, 2024-10-08:
sieht jetzt echt gut aus, VIELEN dank für die arbeit. ein kleines problem habe ich aber nun beim setzen der HKR-werte (unabhängig davon ob über automation oder direkt): 
"Die Aktion number/set_value konnte nicht ausgeführt werden. object NoneType can't be used in 'await' expression"

LarsK1, 2024-10-09:
@larsneo den Bug habe ich gefunden. Ist mit dem nächsten Release 1.1.0-rc2 behoben

larsneo, 2024-10-09:
sieht jetzt prima aus (direkte einstellung funktioniert, automatisierung muss ich bis zum nächsten tibber-preiswechsel warten), vielen dank dafür :-)

kwithus, 2024-10-09:
kann es leider erst nächste Woche ausprobieren, bin zur Zeit abwesend und möchte keinen Frust der Familie riskieren, falls Neustart aus irgendwelchen Gründen hängt 😂
Ich würde es bevorzugen, wenn Temperatur Einstellungen nicht per slider, sondern als Eingabefeld konfiguriert werden können, Slider sind gerade auf Mobilgerät ziemlich unpräzise

LarsK1, 2024-10-10:
@kwithus ob Slider oder nicht liegt nicht in meiner Hand. Du kannst auf den Dashboards eigene Widgets erstellen, da kannst du es einstellen, in der Integration geht es aber nicht

JoeUrs, 2024-10-10:
@larsneo -  Ja, das mit der Heizkurve ist ja eher eine einmalige Sache. Die Vorgehensweise ist sehr gut in der Bedienungsanleitung für Kunden, aber auch spezifischer in der Bedienungsanleitung für Fachpersonal beschrieben. Hier findet man auch alle Angaben zu den Sensoren und Ausgängen. 
[https://solvis-files.s3.eu-central-1.amazonaws.com/downloads-fk/regelung/sc-3/32432_BAL-SBSX-3-I.pdf](url)

LarsK1, 2024-10-16:
@JoeUrs ja, kenne ich von meiner Heizung auch. Passiert ungefähr einmal im Monat 😞

JoeUrs, 2024-10-16:
@LarsK1  - Nächsten Dienstag kommt bei mir der Installateur vorbei, schaut sich das mal an und will daraufhin mit dem Solvis Kundendienst das Problem angehen. Mal sehen was dabei heraus kommt. Ich gebe bescheid.

kwithus, 2024-10-16:
> beim setting für "heizkurve steilheit" bin ich mir aber tatsächlich auch noch nicht sicher, ob man das via HA überhaupt sinnvoll setzen kann. das sehe ich aber auch eher als einmalige einrichtung in der grundkonfiguration

grundsätzlich ja, wenn die Kurve einmal stimmt, dann braucht man sie nicht zu verändern. Allerdings kann man hier natürlich einmal eine Zeitangabe ‚ausprobieren‘ welche Kurve am optimalsten ist. Das benötigt längere Zeit, da sich die Außentemperaturen ja nicht täglich ändern. 
und die Veränderung der Kurve geht über HA dann einfacher, als sich immer die Solvis App aufzurufen. Ist aber sicherlich nicht das dringendste tägliche To-Do

kwithus, 2024-10-16:
> @kwithus ob Slider oder nicht liegt nicht in meiner Hand. Du kannst auf den Dashboards eigene Widgets erstellen, da kannst du es einstellen, in der Integration geht es aber nicht

Bei einem Template kann man für number m.E. Nach Mode=box setzen (default ist slider) 
Lässt sich das in den Definitionen für die Integration nicht auch einbinden?

LarsK1, 2024-10-20:
Ja, doch das ist möglich. Ich bin mir nur nicht sicher, ob das nicht auch sehr vom Nutzer abhängt. Zudem sagt der HA-Dev-Guide (https://developers.home-assistant.io/docs/core/entity/number/) auch eindeutig, dass "auto" zu wählen ist, und nicht box/slider...

commit 70c013a3 (2024-10-06): Minor fix

Closes #11