LarsK1/hass_solvis_control #12: [FEATURE] values for other HK

kwithus, 2024-08-19:
Wäre schön, wenn man (optional?) auch weitere HK anzeigen/steuern könnte. Ich habe drei verschiedene Heizkreise.

LarsK1, 2024-08-19:
Hi @kwithus,
ich schließe aus deinem Diskussionsthema, dass du die zugehörigen Modbus-ID kennst - wenn du mir diese verrätst pflege ich diese gerne mit ein.

kwithus, 2024-08-19:
Das meiste ergibt sich ja aus der offiziellen Dokumentation, ein paar Adressen habe ich zusätzlich zuordnen können.
Wobei gerade bei den TempFühlern das von Installation zu Installation unterschiedlich sein kann, dh. Das Beste wäre, wenn die Integration Optionen hätte, um auszuwählen welche Adressen abgefragt / beschrieben werden sollten, wobei dabei die 'offiziellen' Bezeichnungen verwendet werden und man dann die Bezeichnungen individuell anpasst.
[SC-3_ModBus_Schnittstellenbeschreibung 2.pdf](https://github.com/user-attachments/files/16664797/SC-3_ModBus_Schnittstellenbeschreibung.2.pdf)
[Solvis Modbus Adressen.pdf](https://github.com/user-attachments/files/16664849/Solvis.Modbus.Adressen.pdf)

kwithus, 2024-10-05:
Well, I managed to add HKR2 and 3 to my installation, and I tried to create a pull request for that by uploading the changed files, however, I havent done a pull request for long time and I am not sure, whether I have done it correctly.
If not, please give me so,e instructions what to do

closed 2024-10-06 by LarsK1