LarsK1/hass_solvis_control #13: [FEATURE]

kwithus, 2024-08-19:
Es wäre schön, wenn es möglich wäre, die Schaltpunkte für HK, Warmwasser / Zirkulationspumpe einsehen und einstellen könnte.

LarsK1, 2024-10-23:
HI @kwithus ,
can you maybe update this issue, which excat switching points you are referring to? Some have already been added lately.

kwithus, 2024-10-23:
I am talking about the weekplan schedules (34048 ff)
However, at least some of those are not correctly documented for RC3 (that’s what Solvis told me)
I used 34174-34178 under RC2 and they don’t give any values currently (in my config)

timmost, 2024-12-07:
Would be a nice feature to determine the "to-be value" and compare it with the "as-Is" value to see how well the control of temperatures work. However, this would require some additional logic to record then:
- read all values for the timing from Solvis
- interpret the times stored in the field "start" / "stop" to match current time with the right time window programmed in Solvis
- get the corresponding to be temperature value
- i.e. Current time Monday 17:15 -> compare with "Wochenplan Tag x Start / Stop y" -> determine time window 1/2/3 for that weekday -> Get corresponding temperature
Is done in the user interface in Solvis automatically, but is not shown in the modbud registers...

fkhr79, 2025-05-31:
@LarsK1 
Denke gerade über die Umsetzung (nur Auslesen der Register - zunächst keine weitere Logik) nach. Das sind ja eine MENGE Register. Wenn wir die alle standardmäßig hinzufügen, wird das sehr wild...
M. E. bleibt eigentlich nur, den Config-Flow zu erweitern und das Hinzufügen der Entitäten dem User zu überlassen. Die Optionen könnten sein:

- HKR1 Wochenplan (Register 34048 bis 34089)
- HKR2 Wochenplan (Register 34090 bis 34131)
- HKR3 Wochenplan (Register 34132 bis 34173)
- Warmwasser Wochenplan (Register 34174 bis 34215)
- Zirkulation Wochenplan (Register 34216 bis 34257)

Den Wochenplan für HK Eco-Betrieb würde ich mir erstmal sparen: fraglich ob den überhaupt jemand braucht.

Dann ist da noch die Frage, wie "schön" das mit einer number-Entity wird. Input-Number ist - soweit ich recherchieren konnte - den Nutzern vorbehalten. Ansonsten sehe ich keine Optionen. Du?
Problem ist an der Stelle ebenfalls, dass die Uhrzeit in den Registern in 15-Minuten-Intervallen codiert ist... 

LG
Frank

LarsK1, 2025-06-01:
Hi Frank,
jain, es gibt es Variante die ich persönlich schöner finde. 

[image]

Das ist ein Textfeld aus Z2M. Vielleicht könnten wir so etwas für unsere Schaltzeiten ja auch nachbauen, würde da etwas Aufwand bedeuteen, dass per REGEX zu prüfen. Halte ich aber von der Usability für die beste Option die wir haben.

LarsK1, 2025-06-01:
Aber auch hier haben wir das Problem https://github.com/LarsK1/hass_solvis_control/issues/301#issuecomment-2926455616

kwithus, 2025-06-01:
> [@LarsK1](https://github.com/LarsK1) Denke gerade über die Umsetzung (nur Auslesen der Register - zunächst keine weitere Logik) nach. Das sind ja eine MENGE Register. Wenn wir die alle standardmäßig [...]
Ich persönlich sehe keinen großen Mehrwert darin, die Schaltzeitpunkte in HA zu sehen, solange ich sie nicht auch von HA aus beeinflussen/verändern kann. Letzteres geht leider nur, wenn Solvis das zum Beschreiben freigeben würde.
Es gibt auch andere Regelpunkte, die ich gerne zum Schreiben zur Verfügung hätte, z.B. die Einstellungen für den Zirkulationsmodus. Keine Ahnung, ob Solvis plant da etwas zu verändern.

kwithus, 2025-06-01:
> Ich war gerade irritiert, dass die Register nicht schreibbar sein sollen und habe testweise mal 34048 beschrieben: funktioniert. Insofern würde ich davon ausgehen, dass das Schreiben für alle o. a. Register funktioniert.>

das muss dann neu sein, in der Doku, die ich kenne steht read only.
Aber gut, wenn es jetzt klappt

fkhr79, 2025-06-04:
> Konkret soll laut Doku das Register 34174 WW am Tag 1 Start 0 beschreiben. Zumindest vor der Umstellung auf Remote 3 steuerte dies Register bei mir tatsächlich die Zirkulation Tag 1 Start 0.
> Seit Umstellung auf Remote 3 ändert sich dies Register gar nicht mehr, wenn ich versuche es zu beschreiben.
> Das Gleiche gilt für die danach folgenden Registerzu WW bzw.zu Zirkulation

Ah.. oookay! - Ich habe erstmal nur den Schreibtest gemacht. Ob die interne Zuordnung stimmt, natürlich nicht. Das kann ja lustig werden...