LarsK1/hass_solvis_control #17: [FEATURE] Schalter für 2322 / WW Nachheizung Start

larsneo, 2024-09-21:
Es wäre schön, wenn man die Nachheizung über HA nicht nur anzeigen, sondern auch auslösen könnte (z.B. in Kombination mit Erhöhung von 33025 / Warmwassertemperatur bei günstigen Strompreisen oder PV Überschuß)

LarsK1, 2024-09-25:
Hallo, 
vielen Dank für deinen Feature-Wunsch. Aktuell ist es aber leider nicht umzusetzen, da (meines Wissens) Solvis kein Modbus-Register dafür bereitstellt.

Solltest du aber die Nummern kennen, lass es mich wissen, dann setze ich das gerne mit um 😄

JoeUrs, 2024-09-25:
Der Wert für die WW Nachheizung Start ist 2328.
Hierzu muss in der SC-3 unter Modus auf "write" gestellt werden, damit man den Wert auch setzen kann (0 / 1).

LarsK1, 2024-10-09:
Gelöst in https://github.com/LarsK1/hass_solvis_control/commit/ec61ed5a5f61be1056b931c14047865715b3b02f

larsneo, 2024-10-11:
Vielen Dank für die Switch-Erweiterung, in der 1.1.0-rc3 sehe ich jetzt auch den schalter - scheinbar kann man damit WW-Aufbereitung aber noch nicht aktivieren - mache ich einen Denkfehler oder funktioniert das noch nicht?
[image]

LarsK1, 2024-10-11:
@larsneo Bei mir ist der Wert zwar dauerhaft an, aber ich mir nicht ganz sicher - wie sich das Solvis gedacht hat. Da meine (Heizung) immer 1 per Modbus ausgibt?

JoeUrs, 2024-10-11:
@LarsK1 -  Da wir der Staus ausgegeben (2328). 1 für an und 0 für aus. Hierbei handelt es sich um ein Register, welches auch beschrieben werden kann. 
Also wird hier festgelegt ob die Nachheizung aktiviert ist oder nicht.

nevyen, 2024-10-15:
Ich habe mir auch das RC-3 installiert. Vielen Dank für diese super Integration :-) 

Das mit dem Nachheizen scheint bei mir auch nicht zu gehen. Der Wert ist immer 1 und selbst wenn ich den "Schalter" betätige, ändert sich nichts.

LarsK1, 2024-10-16:
Alles klar, könntet Ihr einem den Debug-Modus einschalten und mir die Logs nach ca. einer halben Stunde zukommen lassen?
[image]

nevyen, 2024-10-16:
Anbei die Log Datei. 
Relativ am Anfang habe ich auch einmal den Switch betätigt. 
[error_log.txt](https://github.com/user-attachments/files/17390886/error_log.txt)

Vielen Dank

kwithus, 2024-10-16:
Also ich kann den Wert per Modbus ein/ausschalten

[image]

nevyen, 2024-10-17:
> @nevyen leider sehe ich in der Datei auch keine Fehler, welchen Wert hat den bei Dir Version SC3, bzw. Version NBG?

Version Netzbaugruppe N3.2.1
Anlagen Version MA3.19.47

LarsK1, 2024-10-17:
Hmm, es scheint also irgendwie mit der Anlagenversion zusammenzuhängen. Meine ist deutlich älter. Ich versuche mal jemanden bei Solvis zu erwischen.

nevyen, 2024-10-17:
Vielen vielen Dank. 

Ich probiere gerade das ganze mal "manuell" über Modbus und die Konfiguration einzubinden.

kwithus, 2024-10-17:
in meinem screenshot oben wird übrigens gerade ausgeschaltet (0), einschalten natürlich über value :1 
wobei ich in der Regel nur einschalte und die Steuerung dann entsprechend nach Erreichen der Temperatur ausschaltet

nevyen, 2024-10-17:
> in meinem screenshot oben wird übrigens gerade ausgeschaltet (0), einschalten natürlich über value :1 wobei ich in der Regel nur einschalte und die Steuerung dann entsprechend nach Erreichen der Temperatur ausschaltet

Kannst du mir den Auszug aus der config.yaml schicken? Ich bekomme den Sensor nicht eingerichtet

kwithus, 2024-10-17:
[image]

schicke es als screenshot, bekomme den code nicht formatiert reinkopiert (vielleicht weil ich derzeit nur am iphone bin)

[image]

nevyen, 2024-10-17:
@kwithus Das ist ja nur die Sequenz ich meine den "Sensor" selber. 

@LarsK1 ja aber das ist ja nicht die fertige Struktur des Sensors. Das baust du dann ja in der switch.py noch zusammen. 

Bei mir sieht es gerade so aus:
`modbus:
  - name: SC3
    type: tcp
    host: 192.168.15.30
    port: 502
    delay: 0
    message_wait_milliseconds: 30
    timeout: 5
    switches:
      - name: WW Nachheizung
        address: 2328
        command_on: 1
        command_off: 0`

Aktuell kommt aber immer der Fehler:` Pymodbus: SC3: Error: device: 0 address: 2328 -> Modbus Error: [Input/Output] ERROR: No response received after 3 retries`

wenn ich den Schalter betätige.

kwithus, 2024-10-17:
Ok, mir fällt gerade auf, dass Ihr die Adresse 2328 nutzt, steht so auch in der Doku.
Ich nutze in meinem Script 2322, das Script stammt noch aus der Zeit bevor bei mir im Sommer die Wärmepumpe und damit die RC3 installiert wurde. Beim Umstieg von RC2 auf RC3 habe ich einiges anpassen müssen, da dies Script lief, habe ich es nicht hinterfragt.
Offenbar stimmt hier die Doku der RC3 nicht (Solvis hatte mir gegenüber auch schon gesagt, die Doku hätte Fehler und würde ‚demnächst‘ überarbeitet)
Wenn ich jetzt in der cons.py die Adresse 2328 mit 2322 ersetze, kann ich die Nachheizung über die Integratin ein/ausschalten.
Vielleicht probiert ihr das auch einfach mal

larsneo, 2024-10-18:
mit #2322 in der 1.1.1 funktioniert die schaltung jetzt perfekt, vielen dank!
sidenote: in der manifest.json ist es wohl noch die 1.1.0 ;-)

LarsK1, 2024-10-18:
@larsneo Danke für das Lob. Dann schließe ich diesen Bug 🪲

closed 2024-10-18 by LarsK1