LarsK1/hass_solvis_control #18: [FEATURE] Make all values from the Modbus Spezifikation available ... and correct some wrong units/device-classes

MAPSOFT2, 2024-09-27:
I love the features the integrations has right now ... but I am missing some values e.g. HK2 ... 
how about adding "all" values from the Modbus-Spezifikation?
e.g. https://solvis-files.s3.eu-central-1.amazonaws.com/downloads-fk/regelung/sc-3/SC-3_ModBus_Schnittstellenbeschreibung.pdf

You tried to give the values the speaking names ... instead of using the name from the specification.
I am not sure that this is a good idea. I believe that at least some values are moved around depending on your configuration.

You started the "Ausgang"-Values e.g. with "A05_" ... Why didn't you do that with the Sensors e.g. "S01_"

I believe all the "Ausgang"-Values e.g. with "A01_" have the wrong unit "V" and device class "voltage" .... 
the values are between 0 and 100.
So the unit should be none or "%" ... and the device class could be "POWER_FACTOR" 

Same for "Ionisationsstrom" ... the unit from the documentation is "mA" and the device class would be "CURRENT"

 In my Modbus specification adress 32770 is "Version SC3" and not "Version SC2". ;-)

LarsK1, 2024-09-27:
Hi @MAPSOFT2 
many thanks for your greatful words. I'm currently totally out of time. If you would prepare a PR for this I definitely would appreciate it and definitely try to review and merge it as fast as my rate time allows it 😉 

Thanks ,
Lars

kwithus, 2024-10-04:
Hi, Just want to express my thanks to all working on this integration.I am currently use a YAML Modbus config to get the values and do really appreciate to be able using a specific integration .
However, as MAPSOFT2 mentioned, depending on the specific configuration the actual mapping f the values I; the modbus specification is not necessarily identical to the installation.
So, yes if possible it would be good, if there is a way to easily change the names. In my YAML definition I do use the modbus adress for the unique ID and a ‚speaking name‘ as entity ID, that way it is easier for me to find ‚differences‘ between the documentation and my installation.
I would also appreciate to have all modbus adresses available in the integration and have an easy way to change the naming (actually this can be done within the entities definition in HA)

Question: you do have the entity ‚Elektrische Wärmepumpenleistung‘, this seems to be the actual kWh the Heatpump is using , which modbus adress is this? I was looking for it but couldn’t find it in my installation.

LarsK1, 2024-10-06:
I've pushed some changes. Will be available from the next release. Further changes are coming

kwithus, 2024-10-07:
I added most of the Temp settings for HKR1-3 and done some other changes, after some trial and error I managed to create a pull request for it ;-)

[image]
[image]

JoeUrs, 2024-10-10:
@kwithus - All outputs on my heating system are configured as relay switching outputs. This means that no values ​​can be read out here.

LarsK1, 2024-10-11:
@kwithus I also don't get values for the "Vorlaufart". The manual also doesn't know about 2819? (https://solvis-files.s3.eu-central-1.amazonaws.com/downloads-fk/regelung/sc-3/SC-3_ModBus_Schnittstellenbeschreibung.pdf).

Do you have a different version, or is this something that's related to the specific model?

LarsK1, 2024-10-16:
Bei mir nur leider nicht. Meine Heizung kennt diese Werte nicht 😜

kwithus, 2024-10-16:
[image]

unterschiedliche Software-Versionen?

LarsK1, 2024-10-17:
Ja, scheint so. Meine NBG hat 3.1.0, dafür die Anlage aber nur 3.5.1. Ich werde mal unseren Installateur fragen, ob ein Update möglich ist.

closed 2024-10-17 by LarsK1

MAPSOFT2, 2024-10-18:
Since yesterday I got my Solvis up and running ;-)
And I looked into the new Version.
Thanks for the new version!!!
There are some Values that have a different meaning and some I don't have at all.
I liked your Idea with the optionsmenue ... but I think that that is not constructive.
How about a using the register and the Address as ID ... and then make the name editable by the user ... e.g. a file or somethink like the optionsmenue.
I would suggest to use the names from the modbus as initial names.
These names should stay when updating. 
I would also suggest a parameter for showing the value.
Maybe it is possible to automatically check this parameter when values are received for this value. 
Something like this would make the integration as flexable as the solvis system.

I would like to help now that I have a system that is up to date and just configured by the customer service of the manufacturer.
I transferred the Modbus-Dokumentation to Excel and created formulas that "build" the ModbusFieldConfig. 
These have to be tested ... but they might be a good start. 
If you want, I can send it to you.
To make the const.py easier to extend, I would suggest to put he ModbusFieldConfig-Entries in the Order of the Register and Address ... like in the ModBus-Dokumentation. Ideally I would like to see all entries with all configuration parameters in it .... commented out until tested.
That way me (or others) can change the const.py and restart the integration and compare the values with the official Solvis app received.
I know that I want much ... but I also want to help get a good and flexable integration.

kwithus, 2024-10-19:
@MAPSOFT2 I generally agree. I already discovered discrepancies between the Solvis Modbus Documentation and the actual setting. I am not sure, whether all configurations are sticking to the official documentation.
So yes, maybe having the address as part of the entity name would be helpful.
Otherwise, for all those with a standard configuration of their Solvis, it is much easier to have it kind of ‚set and go‘.
Anyhow, both ways will create situations where the entity name needs to be adjusted by the user and those adjustments should be as easy as possible and should last over any updates of the integration.
I have now idea, what the best way is to program this into the integration.
For now, I am just happy, that @LarsK1 is doing all this work for us ;-)

commit 3b17b3b4 (2024-10-17): Fixes #18