LarsK1/hass_solvis_control #31: [BUG] coorect unit for "Elektrische Wärmepumpenleistung" is kW (not kWh)

larsneo, 2024-10-07:
**Version of the plugin**
1.1.0-beta5

 **Home-Assistant Version**
2024.10.1

**Enviroment (Docker, HAOS, ...)**
Home Assistant green

**Describe the bug**
Wrong unit for adress #33545 in last update

**To Reproduce**
Update to latest beta-version

**Expected behavior**
Display of "elektrische Wärmepumpenleistung" in kW (not kWh)

**Additional context**
    ModbusFieldConfig(  # elektrische Wärmepumenleistung
        name="elek_waermepumpe_leistung",
        address=33545,
        unit="kW",

LarsK1, 2024-10-07:
@kwithus @larsneo I'm pretty unsure about this one. 

Can you please give me detailed information about, why you want which unit? Since I don't have a heat pump I can't text myself.

larsneo, 2024-10-08:
verkürzt gesagt: Kilowatt beschreibt die Leistung, Kilowattstunden die Arbeit.
Langfassung: https://www.bmwk-energiewende.de/EWD/Redaktion/Newsletter/2016/12/Meldung/direkt-erklaert.html

Die elektrische Aufnahmeleistung p(el) und die thermische 
Leistung p(therm) der Wärmepumpe sind also beide jeweils in kW anzugeben, der "Verbrauch" bzw. die "Erzeugung" über eine bestimmte Zeit wäre dann kWh.

[image]

kwithus, 2024-10-08:
I am sure, you are absolutely correct. However, HA expects the kWh in order to include an entity into the energy statistics, e.g. to be shown on the energy dashboard

larsneo, 2024-10-08:
nope, heat pump and e.g. wallbox should use "watt" (or kilowatt) in the energy dashboard
[image]

kwithus, 2024-10-08:
that’s strange, as my HA tells me something differently 

[image]

kwithus, 2024-10-08:
[image]

I use this config for the heatpump and was able to add it
    `ModbusFieldConfig(  # elektrische Wärmepumenleistung
        name="elek_waermepumpe_leistung",
        address=33545,
        unit="kWh",
        device_class="energy",
        state_class="total",
        register=2,
        edit=False,
        enabled_by_default=True,
    ), `

larsneo, 2024-10-08:
we are talking about different things - the ha energy dashboards shows "Energieverbrauch" over the time (day, week, month), therefore kWh is the correct unit in this case. nevertheless the unit for wallbox and heatpump itself is kW ;-)

see my dashboard an compare to the energy flow diagram in my previous post

[image]

larsneo, 2024-10-08:
see e.g. the Tibber entities as another example for the correct mixup of kW and kWh:

[image]

kwithus, 2024-10-08:
Well, again, you are absolutely right and I agree that kW would be the correct setting, however
For the power flow card it doesn’t make a difference whether kW or kWh, both works.
I still don’t understand why it makes a difference for the energy dashboard, which accepts only kWh.
Well if you prefer to have the setting in the integration to kW, that’s fine, for usage in Energy Dashboard I will then have to setup a template sensor to convert it to kWh ;-)

HLueken, 2024-10-08:
The implementation with kW is correct since this is the **power** used by the heat pump. The Energy dashboard shows the **consumption** which is kWh. If you want to show the consumption of the heat pump you should define a integral sensor (which calculates power to consumption using time)

commit 7529a137 (2024-10-08): Bugfix for wrong unit

Closes #31
closed 2024-10-08 by LarsK1