LarsK1/hass_solvis_control #39: [Enhancement] Individual names for HKR

kwithus, 2024-10-17:
Would be great if the HKR could be named individually (during setup)
Having three HKRs it would be easier to understand the dashboard with ‚speaking‘ names. Assigning them during setup would be easier than changing all entities manually afterwards.

LarsK1, 2024-10-17:
Hi,
thanks for your requests. I'm not shure, if that is even possible. Will leave it open for now

kwithus, 2024-10-18:
Maybe an alternative would be to set up different entities, one for each HKR, on for Water and a ‚general‘ one for the heater incl. Solar and heat pump , or alternatively for the heating side also different entities for the Solvis Heater, for Solar and for Heat pump.
Both ways would give a better overview. Having three HKR, Solar and Heatpumps creates currently a very large entity with many attributes, not so easy to structure on the dashboard.
I know it’s not the most prominent feature request, sorting out all addresses and potential bugs is more important. However, adding ‚speaking‘ names, and/or setting up different entities is potentially a breaking change, so maybe nice to have upfront.

fkhr79, 2025-02-23:
Since the number of entities in version 2.x is getting even bigger I also thought about a bit more structure. 
One possibility (probably the one I like the most and nearly matches your suggestion) is to use the via_device property of devices (see https://developers.home-assistant.io/docs/device_registry_index/) which would also resemble the modularity of solvis devices ver well. 
But this would be a lot of work. Don't know if I would put so much effort into implementing this because the entities can quite easily grouped and structured on dashboards.

kwithus, 2025-02-23:
HiI am currently traveling with limited internet access, so I cannot see this thread on GitHub and I cannot see whether you already posted anything in regards to the upcoming changes in 2.0. However, first at all, as you do the vast majority of work, it’s your decision what to and how to implement.That said, sure, having more structure would be nice.Maybe splitting the one Solvis device into multiple devices, which are set up separately? Looking into the Solvis Portal, they structure it by heating/solar/warmwater/heatpump (and maybe other topics depending on setup)I am not sure whether this is what you talk about when mentioning the via-device structure, although I looked at the link ;-).So yes, better structure would be nice, probably it would reduce work to group on the user side, however as you said, it will definitely mean more work on your side. Your decision ThanksKarl

LarsK1, 2025-02-23:
I see the issue here as well. Let us focus on releasing a stable 2.0.0 and for 2.1 we try to get this done...

fkhr79, 2025-06-04:
Was a small fight... but think I did it. See the dev-branch in my fork if you wanna try 🙂 
But: once named individually one has to re-config (re-install) the integration to change the individual hkr-names. Acceptable, I think.

fkhr79, 2025-06-04:
There you are, @kwithus 
Enjoy! 🙂

PR #311 (2025-06-04): Feat: add config option for custom hkr* naming
Fix #39
commit 5a0bff28 (2025-06-04): Feat: add config option for custom hkr* naming

Fix #39