LarsK1/hass_solvis_control #46: [Feature] Add more entites to const.py

LarsK1, 2024-10-19:
Since yesterday I got my Solvis up and running ;-)
And I looked into the new Version.
Thanks for the new version!!!
There are some Values that have a different meaning and some I don't have at all.
I liked your Idea with the optionsmenue ... but I think that that is not constructive.
How about a using the register and the Address as ID ... and then make the name editable by the user ... e.g. a file or somethink like the optionsmenue.
I would suggest to use the names from the modbus as initial names.
These names should stay when updating. 
I would also suggest a parameter for showing the value.
Maybe it is possible to automatically check this parameter when values are received for this value. 
Something like this would make the integration as flexable as the solvis system.

I would like to help now that I have a system that is up to date and just configured by the customer service of the manufacturer.
I transferred the Modbus-Dokumentation to Excel and created formulas that "build" the ModbusFieldConfig. 
These have to be tested ... but they might be a good start. 
If you want, I can send it to you.
To make the const.py easier to extend, I would suggest to put he ModbusFieldConfig-Entries in the Order of the Register and Address ... like in the ModBus-Dokumentation. Ideally I would like to see all entries with all configuration parameters in it .... commented out until tested.
That way me (or others) can change the const.py and restart the integration and compare the values with the official Solvis app received.
I know that I want much ... but I also want to help get a good and flexable integration.

_Originally posted by @MAPSOFT2 in https://github.com/LarsK1/hass_solvis_control/issues/18#issuecomment-2423187029_

LarsK1, 2024-10-19:
Hi,
feel free to add a PR for this. You'll need to add the corresponding entities to const.py and the name to strings.json (in the right section, according to the input_type. Afterwards copy them over to de.json. Also you'll need to add the conf_option to const.py, currently there are HKR2 (1), HKR3 (2), solar collector (3), heat pump (4)

LarsK1, 2024-10-19:
There is also an list of supported attributes: https://github.com/LarsK1/hass_solvis_control/blob/main/supported-entites.md

MAPSOFT2, 2024-10-19:
I just looked through that document and I was flashed that so many of the values are allready in the integration :-)
Thank you so much!!!

Since I have never done a PR so far ... that makes it harder than I would like it to be ;-)

MAPSOFT2, 2024-10-19:
until I got that figured out I will look through the already available values...
1) A02.Pumpe Warmwasser
In my configuration it is supposed to be "Ladepumpe WP".
But that's not what I found out.
The Value of A02 is shown as 100% in the SOLVIS App ... but your Integration shows 1,56% ... that is 4 * your multiplier 0,00390625
Strange. I have no idea.
I have tried to varify those ModBus-Values .... but so far I haven't found a program working for me. Once in a while I get values but most of the time not :-(
Can I have only one Modbus Master connected to SOLVIS?

2) DigIn Störungen
It looks like you got a multiplier 0.1 in there ....
I see Values of 0.1, 0.2 and 0.3 in the Integration.
In the Modbus Documentation it looks like it is a bit-switch ... that would make a 
0 -> everything fine
1 -> "Sicherung Netzbaugruppe"
2 -> "Brennerfehler"
3 -> "Sicherung Netzbaugruppe" and "Brennerfehler"

3) HKR1 Heizkurve Steilheit / HKR2 Heizkurve Steilheit
In the Dokumentation it says ... values from 20 to 250 "in 10tel".
What does that mean .... real values from 2.0 to 25.0 ?
If I look in the Solvis App I can enter values from 0.2 up to 2.5
So (I think) the multiplier shoud be 0.01.
But that still doesn't fit to my Steilheit values in the Solvis-App 0.8 / 1.0 ... and 20 / 20 in your Integration.
Same problem with verifying the Modbus-Values with a program.

That's all for now ... and again thank you for your effort :-)

LarsK1, 2024-10-19:
@MAPSOFT2 fixed 2 / 3.
You'll only can connect one device using modbus at a time. Everything else must raise errors. Usually you'll need to hard shut-down the solvis device, afterwards.

MAPSOFT2, 2024-10-20:
How about a Solvis-Dashboard?
[image]

LarsK1, 2024-10-20:
Looks pretty nice, is this from you?

MAPSOFT2, 2024-10-20:
Just looked at your fix for my 2. ... did you see the second point ... the bit-switch part?
That way more than one  DigIn Störungen is possible at the same time.
I believe the easiest way to solve this would be to do something like this in strings.json:
"0":""
"1":"Sicherung Netzbaugruppe"
"2":"Brennerfehler"
"3":"Sicherung Netzbaugruppe,Brennerfehler"
"4":"STB1 Fehler"
"5":"Sicherung Netzbaugruppe,STB1 Fehler"
"6":"Brennerfehler,STB1 Fehler"
"7":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler"
"8":"STB2 Fehler,STB2 Fehler"
"9":"Sicherung Netzbaugruppe,STB2 Fehler,STB2 Fehler"
"10":"Brennerfehler,STB2 Fehler,STB2 Fehler"
"11":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,STB2 Fehler"
"12":"STB1 Fehler,STB2 Fehler,STB2 Fehler"
"13":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,STB2 Fehler"
"14":"Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler"
"15":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler"
"16":"Brenner CM424"
"17":"Sicherung Netzbaugruppe,Brenner CM424"
"18":"Brennerfehler,Brenner CM424"
"19":"Sicherung Netzbaugruppe,Brennerfehler,Brenner CM424"
"20":"STB1 Fehler,Brenner CM424"
"21":"Sicherung Netzbaugruppe,STB1 Fehler,Brenner CM424"
"22":"Brennerfehler,STB1 Fehler,Brenner CM424"
"23":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Brenner CM424"
"24":"STB2 Fehler,STB2 Fehler,Brenner CM424"
"25":"Sicherung Netzbaugruppe,STB2 Fehler,STB2 Fehler,Brenner CM424"
"26":"Brennerfehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"27":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"28":"STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"29":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"30":"Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"31":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"32":""
"33":"Sicherung Netzbaugruppe"
"34":"Brennerfehler"
"35":"Sicherung Netzbaugruppe,Brennerfehler"
"36":"STB1 Fehler"
"37":"Sicherung Netzbaugruppe,STB1 Fehler"
"38":"Brennerfehler,STB1 Fehler"
"39":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler"
"40":"STB2 Fehler,STB2 Fehler"
"41":"Sicherung Netzbaugruppe,STB2 Fehler,STB2 Fehler"
"42":"Brennerfehler,STB2 Fehler,STB2 Fehler"
"43":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,STB2 Fehler"
"44":"STB1 Fehler,STB2 Fehler,STB2 Fehler"
"45":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,STB2 Fehler"
"46":"Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler"
"47":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler"
"48":"Brenner CM424"
"49":"Sicherung Netzbaugruppe,Brenner CM424"
"50":"Brennerfehler,Brenner CM424"
"51":"Sicherung Netzbaugruppe,Brennerfehler,Brenner CM424"
"52":"STB1 Fehler,Brenner CM424"
"53":"Sicherung Netzbaugruppe,STB1 Fehler,Brenner CM424"
"54":"Brennerfehler,STB1 Fehler,Brenner CM424"
"55":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Brenner CM424"
"56":"STB2 Fehler,STB2 Fehler,Brenner CM424"
"57":"Sicherung Netzbaugruppe,STB2 Fehler,STB2 Fehler,Brenner CM424"
"58":"Brennerfehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"59":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"60":"STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"61":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"62":"Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"63":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424"
"64":"Unbekannt"
"65":"Sicherung Netzbaugruppe,Unbekannt"
"66":"Brennerfehler,Unbekannt"
[... 164 lines cut ...]
"231":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Unbekannt,Anlagendruck"
"232":"STB2 Fehler,STB2 Fehler,Unbekannt,Anlagendruck"
"233":"Sicherung Netzbaugruppe,STB2 Fehler,STB2 Fehler,Unbekannt,Anlagendruck"
"234":"Brennerfehler,STB2 Fehler,STB2 Fehler,Unbekannt,Anlagendruck"
"235":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,STB2 Fehler,Unbekannt,Anlagendruck"
"236":"STB1 Fehler,STB2 Fehler,STB2 Fehler,Unbekannt,Anlagendruck"
"237":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,STB2 Fehler,Unbekannt,Anlagendruck"
"238":"Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler,Unbekannt,Anlagendruck"
"239":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler,Unbekannt,Anlagendruck"
"240":"Brenner CM424,Unbekannt,Anlagendruck"
"241":"Sicherung Netzbaugruppe,Brenner CM424,Unbekannt,Anlagendruck"
"242":"Brennerfehler,Brenner CM424,Unbekannt,Anlagendruck"
"243":"Sicherung Netzbaugruppe,Brennerfehler,Brenner CM424,Unbekannt,Anlagendruck"
"244":"STB1 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"245":"Sicherung Netzbaugruppe,STB1 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"246":"Brennerfehler,STB1 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"247":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"248":"STB2 Fehler,STB2 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"249":"Sicherung Netzbaugruppe,STB2 Fehler,STB2 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"250":"Brennerfehler,STB2 Fehler,STB2 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"251":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,STB2 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"252":"STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"253":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"254":"Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424,Unbekannt,Anlagendruck"
"255":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,STB2 Fehler,Brenner CM424,Unbekannt,Anlagendruck"

MAPSOFT2, 2024-10-20:
Did you see the second part from 2. ... regarding the bit-switch?

I didn't see this in strings.json .... how about something like this:
"0":""
"1":"Sicherung Netzbaugruppe"
"2":"Brennerfehler"
"3":"Sicherung Netzbaugruppe,Brennerfehler"
"4":"STB1 Fehler"
"5":"Sicherung Netzbaugruppe,STB1 Fehler"
"6":"Brennerfehler,STB1 Fehler"
"7":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler"
"8":"STB2 Fehler"
"9":"Sicherung Netzbaugruppe,STB2 Fehler"
"10":"Brennerfehler,STB2 Fehler"
"11":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler"
"12":"STB1 Fehler,STB2 Fehler"
"13":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler"
"14":"Brennerfehler,STB1 Fehler,STB2 Fehler"
"15":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler"
"16":"Brenner CM424"
"17":"Sicherung Netzbaugruppe,Brenner CM424"
"18":"Brennerfehler,Brenner CM424"
"19":"Sicherung Netzbaugruppe,Brennerfehler,Brenner CM424"
"20":"STB1 Fehler,Brenner CM424"
"21":"Sicherung Netzbaugruppe,STB1 Fehler,Brenner CM424"
"22":"Brennerfehler,STB1 Fehler,Brenner CM424"
"23":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Brenner CM424"
"24":"STB2 Fehler,Brenner CM424"
"25":"Sicherung Netzbaugruppe,STB2 Fehler,Brenner CM424"
"26":"Brennerfehler,STB2 Fehler,Brenner CM424"
"27":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Brenner CM424"
"28":"STB1 Fehler,STB2 Fehler,Brenner CM424"
"29":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Brenner CM424"
"30":"Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424"
"31":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424"
"32":"Solardruck"
"33":"Sicherung Netzbaugruppe,Solardruck"
"34":"Brennerfehler,Solardruck"
"35":"Sicherung Netzbaugruppe,Brennerfehler,Solardruck"
"36":"STB1 Fehler,Solardruck"
"37":"Sicherung Netzbaugruppe,STB1 Fehler,Solardruck"
"38":"Brennerfehler,STB1 Fehler,Solardruck"
"39":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Solardruck"
"40":"STB2 Fehler,Solardruck"
"41":"Sicherung Netzbaugruppe,STB2 Fehler,Solardruck"
"42":"Brennerfehler,STB2 Fehler,Solardruck"
"43":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Solardruck"
"44":"STB1 Fehler,STB2 Fehler,Solardruck"
"45":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Solardruck"
"46":"Brennerfehler,STB1 Fehler,STB2 Fehler,Solardruck"
"47":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,Solardruck"
"48":"Brenner CM424,Solardruck"
"49":"Sicherung Netzbaugruppe,Brenner CM424,Solardruck"
"50":"Brennerfehler,Brenner CM424,Solardruck"
"51":"Sicherung Netzbaugruppe,Brennerfehler,Brenner CM424,Solardruck"
"52":"STB1 Fehler,Brenner CM424,Solardruck"
"53":"Sicherung Netzbaugruppe,STB1 Fehler,Brenner CM424,Solardruck"
"54":"Brennerfehler,STB1 Fehler,Brenner CM424,Solardruck"
"55":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Brenner CM424,Solardruck"
"56":"STB2 Fehler,Brenner CM424,Solardruck"
"57":"Sicherung Netzbaugruppe,STB2 Fehler,Brenner CM424,Solardruck"
"58":"Brennerfehler,STB2 Fehler,Brenner CM424,Solardruck"
"59":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Brenner CM424,Solardruck"
"60":"STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck"
"61":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck"
"62":"Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck"
"63":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck"
"64":"Unbekannt"
"65":"Sicherung Netzbaugruppe,Unbekannt"
"66":"Brennerfehler,Unbekannt"
[... 420 lines cut ...]
"487":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"488":"STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"489":"Sicherung Netzbaugruppe,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"490":"Brennerfehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"491":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"492":"STB1 Fehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"493":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"494":"Brennerfehler,STB1 Fehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"495":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"496":"Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"497":"Sicherung Netzbaugruppe,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"498":"Brennerfehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"499":"Sicherung Netzbaugruppe,Brennerfehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"500":"STB1 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"501":"Sicherung Netzbaugruppe,STB1 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"502":"Brennerfehler,STB1 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"503":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"504":"STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"505":"Sicherung Netzbaugruppe,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"506":"Brennerfehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"507":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"508":"STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"509":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"510":"Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"511":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"

kwithus, 2024-10-20:
> @MAPSOFT2 fixed 2 / 3. You'll only can connect one device using modbus at a time. Everything else must raise errors. Usually you'll need to hard shut-down the solvis device, afterwards.

Actually, I currently run the integration parallel with reading/writing with modbus directly, no issues at all doing that parallel. Also the Solvis Max is connected via modbus with the heatpump, which also runs parallel

kwithus, 2024-10-20:
> 3\. HKR1 Heizkurve Steilheit / HKR2 Heizkurve Steilheit
>     In the Dokumentation it says ... values from 20 to 250 "in 10tel".
>     What does that mean .... real values from 2.0 to 25.0 ?
>     If I [...]

Well, for me the values for Steilheit are consistent between the integration and the Solvis App / Solvis Remote.  I don’t see a need for any changes

MAPSOFT2, 2024-10-20:
> Looks pretty nice, is this from you?

seeing your question ... I said "Oh no!"

The "background" is from a presentation I found at:
https://www.solvis-analyse.de/resources/55376_solvimax_messe_slow.exe

I "just" added the values from the integration.
Did you have any contact to Solvis regarding your Integration?
I think it would be better/safer to get their OK  ;-(

MAPSOFT2, 2024-10-20:
> Well, for me the values for Steilheit are consistent between the integration and the Solvis App / Solvis Remote. I don’t see a need for any changes

I don't know which Version you are running ....
LarsK1 changed the Factor to 0.01 allready.

MAPSOFT2, 2024-10-20:
When trying to configure the integration again ... you get the questions for IP and port ... after "BESTÄTIGEN"... the options-Menu appears ... when hitting "BESTÄTIGEN" again an "Unknown Error" appears.

LarsK1, 2024-10-20:
@MAPSOFT2 this error has been fixed in 1.1.2. But please do open new issues, otherwise this thread is getting totally unreadable and "new" user migth loose context.

LarsK1, 2024-10-20:
>
> Actually, I currently run the integration parallel with reading/writing with modbus directly, no issues at all doing that parallel. Also the Solvis Max is connected via modbus with the heatpump, which also runs parallel

Solvis has each modbus-servers for their own devices, but the smarthome-modbus-server (according to solvis) only supports one connection at a time 😁

kwithus, 2024-10-20:
Well, so far I never had any issues running both, your integration and a basic modbus sync parallel.

closed 2024-10-23 by LarsK1