LarsK1/hass_solvis_control #48: [Feature] Graphical presentation fo values

LarsK1, 2024-10-20:
How about a Solvis-Dashboard?
[image]

_Originally posted by @MAPSOFT2 in https://github.com/LarsK1/hass_solvis_control/issues/46#issuecomment-2424758500_

timmost, 2024-11-25:
Reminds me on the online generator of Solvis. there are a ton of different configurations possible already with Solvis components.    see here... [https://www.solvis.de/online-generator/](url)

y17dt, 2024-11-30:
I would love to have something like this. 
Maybe with toggles for Warmwater temp integrated?

MAPSOFT2, 2025-02-17:
Another idea would be to take screenshots of 
[https://solvisportal.de/devices/system-schema](https://solvisportal.de/devices/system-schema)
and add the values at certain positions.

nevyen, 2025-04-30:
I take a look here too. 

@MAPSOFT2  Taking a Screenshot will only allows a static representation. The graphical representation of the portal has some "intelligence". For example, when adding warm water the dashed line to the top of the Max gets straight. And the line for the Heaters gets dashed. And you have different "Screens" one for each heating source, and then warm water and maybe another one for Solar panels if installed. 

Taking a look to the portal with a developer console shows, that the graphic is represented as an <path> element. So coping all path elements do an empty html Canva will give the Graphic without any values. 

But I didn't get those Canva integrated into HA. 
And then I did run out of time to get further on this problem. 

Maybe it's possible to render some Canva into HA. 

Another Idea was to work with the remote control of the SC3. But this isn't a simple html page. And I run into the same time problem. 

I would like to have this feature too But this could be a little complicated.

LarsK1, 2025-12-28:
This is now being actively worked on and a first release migth soon be available under: 
https://github.com/LarsK1/hass_solvis_frontend

Any feedback or help is highly appreciated