LarsK1/hass_solvis_control #50: DigIn-Störungen from #46

MAPSOFT2, 2024-10-20:
DigIn Störungen
In the Modbus Documentation it looks like it is a bit-switch ... that would make a
0 -> everything fine
1 -> "Sicherung Netzbaugruppe"
2 -> "Brennerfehler"
3 -> "Sicherung Netzbaugruppe" and "Brennerfehler"

That could look like the following:
"0":""
"1":"Sicherung Netzbaugruppe"
"2":"Brennerfehler"
"3":"Sicherung Netzbaugruppe,Brennerfehler"
"4":"STB1 Fehler"
"5":"Sicherung Netzbaugruppe,STB1 Fehler"
"6":"Brennerfehler,STB1 Fehler"
"7":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler"
"8":"STB2 Fehler"
"9":"Sicherung Netzbaugruppe,STB2 Fehler"
"10":"Brennerfehler,STB2 Fehler"
"11":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler"
"12":"STB1 Fehler,STB2 Fehler"
"13":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler"
"14":"Brennerfehler,STB1 Fehler,STB2 Fehler"
"15":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler"
"16":"Brenner CM424"
"17":"Sicherung Netzbaugruppe,Brenner CM424"
"18":"Brennerfehler,Brenner CM424"
"19":"Sicherung Netzbaugruppe,Brennerfehler,Brenner CM424"
"20":"STB1 Fehler,Brenner CM424"
"21":"Sicherung Netzbaugruppe,STB1 Fehler,Brenner CM424"
"22":"Brennerfehler,STB1 Fehler,Brenner CM424"
"23":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Brenner CM424"
"24":"STB2 Fehler,Brenner CM424"
"25":"Sicherung Netzbaugruppe,STB2 Fehler,Brenner CM424"
"26":"Brennerfehler,STB2 Fehler,Brenner CM424"
"27":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Brenner CM424"
"28":"STB1 Fehler,STB2 Fehler,Brenner CM424"
"29":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Brenner CM424"
"30":"Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424"
"31":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424"
"32":"Solardruck"
"33":"Sicherung Netzbaugruppe,Solardruck"
"34":"Brennerfehler,Solardruck"
"35":"Sicherung Netzbaugruppe,Brennerfehler,Solardruck"
"36":"STB1 Fehler,Solardruck"
"37":"Sicherung Netzbaugruppe,STB1 Fehler,Solardruck"
"38":"Brennerfehler,STB1 Fehler,Solardruck"
"39":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Solardruck"
"40":"STB2 Fehler,Solardruck"
"41":"Sicherung Netzbaugruppe,STB2 Fehler,Solardruck"
"42":"Brennerfehler,STB2 Fehler,Solardruck"
"43":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Solardruck"
"44":"STB1 Fehler,STB2 Fehler,Solardruck"
"45":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Solardruck"
"46":"Brennerfehler,STB1 Fehler,STB2 Fehler,Solardruck"
"47":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,Solardruck"
"48":"Brenner CM424,Solardruck"
"49":"Sicherung Netzbaugruppe,Brenner CM424,Solardruck"
"50":"Brennerfehler,Brenner CM424,Solardruck"
"51":"Sicherung Netzbaugruppe,Brennerfehler,Brenner CM424,Solardruck"
"52":"STB1 Fehler,Brenner CM424,Solardruck"
"53":"Sicherung Netzbaugruppe,STB1 Fehler,Brenner CM424,Solardruck"
"54":"Brennerfehler,STB1 Fehler,Brenner CM424,Solardruck"
"55":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Brenner CM424,Solardruck"
"56":"STB2 Fehler,Brenner CM424,Solardruck"
"57":"Sicherung Netzbaugruppe,STB2 Fehler,Brenner CM424,Solardruck"
"58":"Brennerfehler,STB2 Fehler,Brenner CM424,Solardruck"
"59":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Brenner CM424,Solardruck"
"60":"STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck"
"61":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck"
[... 425 lines cut ...]
"487":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"488":"STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"489":"Sicherung Netzbaugruppe,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"490":"Brennerfehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"491":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"492":"STB1 Fehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"493":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"494":"Brennerfehler,STB1 Fehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"495":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"496":"Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"497":"Sicherung Netzbaugruppe,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"498":"Brennerfehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"499":"Sicherung Netzbaugruppe,Brennerfehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"500":"STB1 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"501":"Sicherung Netzbaugruppe,STB1 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"502":"Brennerfehler,STB1 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"503":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"504":"STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"505":"Sicherung Netzbaugruppe,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"506":"Brennerfehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"507":"Sicherung Netzbaugruppe,Brennerfehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"508":"STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"509":"Sicherung Netzbaugruppe,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"510":"Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"
"511":"Sicherung Netzbaugruppe,Brennerfehler,STB1 Fehler,STB2 Fehler,Brenner CM424,Solardruck,Unbekannt,Anlagendruck,Kondensat"

MAPSOFT2, 2024-10-20:
I just about it a little more ....

I think it would be better to have 8 values that are boolean.

For example: 
DigIn Störung - Brennerfehler
DigIn Störung - STB1 Fehler 
....

MAPSOFT2, 2024-10-23:
I believe it is just 8 Bits ...

876543210 Bit
000000001 -> 1
000000011 -> 3
...
100000001 -> 257

But ... your integration shows me that I have values of 1, 2 or 3 [image]
Meaning that I have a Störung all the time.
And I expect to have no Störung at all ... since it is a new system and the "Anlagenstatus" says "in Ordnung".

fkhr79, 2025-02-08:
The English-German-mixture makes it a very funny reading! 😆 

I experience the same behaviour as @MAPSOFT2 : continuous switching of the entity between 1, 2 and 3. I also do not have any failures in my system (doublechecked in SC3-Installateursmenü).
Therefore I think the register ist flawed - at least in SC3. (Never used the register in SC2 so I cannot say if it ever functioned).

Furthermore I  fear, that - even if the register would be OK - we won't be able to read the register without greater changes of the code: it's a 16bit register, from which we are only needing the first 8 bit...

fkhr79, 2025-02-16:
investigated furher via another modbus client:
register is int16 and switches between values 
- 0x0000 (binary 0000000000000000, dec. 0)
- 0x0001 (binary 0000000000000000, dec. 1)
- 0x0002 (binary 0000000000000010, dec. 2)
- 0x0003 (binary 0000000000000011, dec. 3)
so, the interpretation is wrong since the errors are represented in the first 9 bits (according to the sc3 modbus spec.)

since we're reading the register as int16 (and cannot change this due to the limitation of pymodbus) we need to extract the first 9 bits (for example) like so:

`bin_first_9_bits = bin(data_from_register)[2:].zfill(16)[:9]`

this delivers an 9 char string of the first 9 bits of the int16 register, which can be evaluated according to the spec:

[image]

fkhr79, 2025-02-16:
@LarsK1 : will you or shall I try?

LarsK1, 2025-02-16:
@fkhr79 Please try, I'm currently working on renaming all entities to a consistent scheme.

fkhr79, 2025-02-16:
Thought about the implementation (and representation in ha) for a while:
Thankfully never got an error like this, but i think the different errors don't exclude each other. So afaik neither a sensor nor a select entity is fitting. I would suggest turning it into a binary_sensor only showing false/true (no error/error) with the errors stored in it's attributes.
Any other suggestion/idea? Otherwise I'd try to do it that way.

LarsK1, 2025-02-16:
@fkhr79 yes, that's probably best. So we convert the current one to a binary_one, which is in overall False / True, and in the _attr_extra_state_attributes we add the details. Sound logical to me.

fkhr79, 2025-02-16:
Solved.

Mind: Solvis seems to count the bits from left to right. This is the direct opposite of the bit counting of a big endian int16.
If one wants to test the code: add (for example) `response_data = 0b0011111100000000` (which is decimal 16128) directly after the case-statement and let the bit-magic happen! 🤪

I would have left the debug statements but lost the battle with princess lint. 😠I'm just not good enough for her.

LarsK1, 2025-02-16:
@fkhr79 thanks for that PR will check tomorrow.

Regarding your linting issues. In Python most devs use PyCharm, which is able to automatically format the file using black according to this Repositories rules... Then you don't need to format anything yourself nor delete helpful comments....

closed 2025-02-16 by LarsK1