LarsK1/hass_solvis_control #51: [BUG] Heizkurve Steilheit has Value 0 since last version

kwithus, 2024-10-21:
Hi
I downloaded and installed the latest version, for Heizkurve Steilheit I now get 0 for all HKR
I am not sure, but believe the changes to multiplier are causing this

(Leistung Solarthermie does work, as expected and shows consistent values.)

LarsK1, 2024-10-21:
@kwithus the change isn't released yet. It will in 1.1.3...

LarsK1, 2024-10-21:
Also the value "Leistung Solarthermie" isn't released yet...

kwithus, 2024-10-21:
Ok, found it, it’s not related to the multiplier, but to a bug in the Solvis Modbus Documentation.
The correct addresses should be 2826, 3082 and 3338 for HKR 1-3 Steilheit.
At least those addresses work in my config.

however, in my view the multiplier and step settings are not optimized too, compared to Multiplier 1 
First of all, the stepsons within the SolvisPortal App is 5, not 10.
Secondly with Multipler 0,01 and range 20-250, any curve setting in the normal range ends up close to the left side of the slider.
I tried some different settings for my three HKR
HKR1 as is in latest version
HKR2 multiplier 0.01 with range 0,2 - 2,5
HKR3 Multiplier 1 with range 20-250
Results see screenshot

[image]

kwithus, 2024-10-21:
Maybe even though the ‘official’ range is 20,250, setting a shorter range would improve representation on dashboard without limiting the reasonable, realistic range of the curve.
I doubt whether a curve below 50 or above 180 will be reasonable in any circumstances

LarsK1, 2024-10-22:
@kwithus many thanks for your deep dive into your topic. I personally choose option 2. From my side, this integration should allow all possible values, therefore I entered the complete possible range, since everybody should decide on their own, what they think is correct.

closed 2024-10-22 by LarsK1

kwithus, 2024-10-22:
Sorry for re-opening this.
I am absolutely fine with the representation, however, with multiplier other than 1, I am not able to change the setting of the curve. 
not only that the slider ‚jumps‘ to values .2 or 1. or 2 (regardless of different setting for step-size I tried) it also doesn’t send any values to Solvis.
With Multiplier 1 I am able to set values, however, there is an issue with the step size, as the value should change only in steps multiplied by 5. again, I tried different step size, nothing worked.
Funny enough, as I said, at least with Multiplier 1 I am able to change the settings, but always steps size 1 and even more strange Solvis does accept steps ‚inbetween‘ where the remote itself only does steps in 5.
Example: I have a curve of .75 (with multiplier 1 showing in HA as 75) I change this by slider to 76, looking via remote on Max I do see setting 76, when I then increase within the remote I do get 81 (step 5 increase)
So somehow we should find combination that makes it possible to change the settings in steps of 5.
There is a different issue with the SolvisPortal App, in case you are using this. Changes to the curve setting in the Portal are reflected immediately in the remote Control, changes made by the remote Control RC3 are not or at least only with very long delay shown in the Portal.
So anyone trying to find a solution should check directly in RC3 
Fallback Solution: making those addresses for the curve a readonly Sensor and using the Portal or RC3 to make changes. Not as nice, but better than a not working number entity