LarsK1/hass_solvis_control #53: [BUG] A02.Pumpe Warmwasser

LarsK1, 2024-10-23:
1) A02.Pumpe Warmwasser
In my configuration it is supposed to be "Ladepumpe WP".
But that's not what I found out.
The Value of A02 is shown as 100% in the SOLVIS App ... but your Integration shows 1,56% ... that is 4 * your multiplier 0,00390625

_Originally posted by @MAPSOFT2 in https://github.com/LarsK1/hass_solvis_control/issues/46#issuecomment-2424023179_

LarsK1, 2024-10-30:
Hi @MAPSOFT2,
can you please enable debug-logging and send it to me. Please try to include 0% to100% states...

MAPSOFT2, 2024-11-10:
I did a debug-log ...

```
...
2024-11-10 18:27:40.492 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 0
2024-11-10 18:27:40.500 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a02_pumpe_warmwasser
2024-11-10 18:27:40.500 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 400
2024-11-10 18:27:40.508 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a03_pumpe_hkr1
...
2024-11-10 18:27:40.941 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Returned data: {'analog_out_1_status': 0.2, 'analog_out_2_status': 0.0, 'analog_out_3_status': 0.0, 'analog_out_4_status': 0.0, 'analog_out_5_status': 0.0, 'analog_out_6_status': 0.0, 'gas_power': 0.0, 'laufzeit_brennerstufe_1': 0.6, 'laufzeit_brennerstufe_2': 0.0, 'outdoor_air_temp': 8.8, 'roof_air_temp': 6.6, 'cold_water_temp': 19.5, 'hkr1_flow_water_temp': 23.2, 'hkr2_flow_water_temp': 24.9, 'hkr3_flow_water_temp': 19.7, 'domestic_water_temp': 49.4, 'domestic_water_reheat_start': 0, 'solar_water_temp': -0.1, 'solar_heat_exchanger_in_water_temp': 20.4, 'solar_heat_exchanger_out_water_temp': 20.1, 'tank_layer1_water_temp': 25.9, 'tank_layer2_water_temp': 48.3, 'tank_layer3_water_temp': 49.1, 'tank_layer4_water_temp': 49.3, 'cold_water_temperatur': 10.0, 'number_gas_burner_start': 44, 'ionisation_voltage': 0.5, 'a01_pumpe_zirkulation': 0.0, 'a02_pumpe_warmwasser': 1.56, 'a03_pumpe_hkr1': 0.0, 'a04_pumpe_hkr2': 0.0, 'a05_pumpe_hkr3': 0.0, 'a12_brennerstatus': 0.0, 'solar_water_flow': 0, 'solar_leistung': 0.0, 'domestic_water_flow': 0.0, 'hkr1_betriebsart': 2, 'hkr1_warmwasser_vorrang': 1, 'hkr1_vorlaufart': 0, 'hkr1_fix_vorlauf_tag': 50, 'hkr1_fix_vorlauf_nacht': 35, 'hkr1_heizkurve_temp_tag_1': 20, 'hkr1_heizkurve_temp_tag_2': 20, 'hkr1_heizkurve_temp_tag_3': 20, 'hkr1_heizkurve_temp_absenkung': 16, 'hkr1_heizkurve_steilheit': 1.0, 'raumtemperatur_hkr1': 0.0, 'hkr2_betriebsart': 2, 'hkr2_vorlaufart': 0, 'hkr2_warmwasser_vorrang': 1, 'hkr2_fix_vorlauf_tag': 50, 'hkr2_fix_vorlauf_nacht': 35, 'hkr2_heizkurve_temp_tag_1': 20, 'hkr2_heizkurve_temp_tag_2': 20, 'hkr2_heizkurve_temp_tag_3': 20, 'hkr2_heizkurve_temp_absenkung': 16, 'hkr2_heizkurve_steilheit': 0.8, 'raumtemperatur_hkr2': 0.0, 'hkr3_betriebsart': 2, 'hkr3_vorlaufart': 0, 'hkr3_warmwasser_vorrang': 1, 'hkr3_fix_vorlauf_tag': 50, 'hkr3_fix_vorlauf_nacht': 35, 'hkr3_heizkurve_temp_tag_1': 20, 'hkr3_heizkurve_temp_tag_2': 20, 'hkr3_heizkurve_temp_tag_3': 20, 'hkr3_heizkurve_temp_absenkung': 20, 'hkr3_heizkurve_steilheit': 0.8, 'raumtemperatur_hkr3': 0.0, 'ww_solltemperatur': 45, 'version_sc3': 32004, 'version_nbg': 30201, 'digin_error': 1, 'zirkulation_betriebsart': 1, 'waermepumpe_leistung': 5.8, 'elek_waermepumpe_leistung': 2.0}
...
```

So a value of 400 is equal to 100% (the value shown in the Solvis-App).

The modbus-documentation says:
33281 8201 Ausgang A2 0,25%
which is different from all the aller A-Values .... e.g. 
33282 8202 Ausgang A3 0,50%

fkhr79, 2025-02-08:
At least in my config I have the same issue. Using the HA Modbus-Integration I get a constant reading of "400".
Turning the pump on via service menu does not change the value.
Might be, that the register is flawed!? - At least in my solvis-configuration. Unfortunately I cannot check with SC2.

Version: 2.0.0-alpha5
on
Home-Assistant Version: 2025.2.1 
Enviroment: HA OS 14.2
Solvis Version NBG: 3.1.0
Solvis Version SC3: 3.19.47

LarsK1, 2025-02-16:
Hi @MAPSOFT2 ,
can you please retest with 2.0.0-alpha.7?

It should be fixed.

fkhr79, 2025-02-17:
Today I started to dig deeeep into the docs...:

Yes, the value itself should be fixed.

Some more details on this topic which need to be adressed:
The raw value of the register (input, 33281) should be 200 (on) or 0 (off)... if everything is OK and the register is actually used.
Depending on the setup Output A2 is either unused (Gas/Öl/Pia/Lino/solo) or it is the Ladepumpe PLAS-WP (WP). See https://solvis-files.s3.eu-central-1.amazonaws.com/downloads-fk/solvismax7/27350_ALS-MAX-7.pdf
In my setup I do have a heatpump (WP) and the register is at 400 (constantly). I already stated that I fear the register being flawed. Meanwhile I think, that, if the register is unused, the value is simply doubled (to indicate the not-usage). Don't know: only Solvis can tell.
But we can use another register for the WP-Ladepumpe: it's Analog Out O4 (input 33297). It contains a genuine % value of the PWM signal.
@LarsK1 : So far for now. I'll check the implementation and raise issues if necessary... but I think we could add some more useful registers 🙂

fkhr79, 2025-02-17:
Ahh... I forgot. @MAPSOFT2 : Could You please tell us about your Solvis setup (SC2/SC3, Max/Ben/... , WP and so on)? Thx!

MAPSOFT2, 2025-02-17:
in the App under "Sensoren und Aktoren" my system shows ....
- A2 is named "A2 Ladepumpe WP" and has a permanent value of 100.00% for me.
- LP is named "LP Ladepumpe WP" and has changing values of 0% up to at least 81%

fkhr79, 2025-02-17:
> in the App under "Sensoren und Aktoren" my system shows ....
>     * A2 is named "A2 Ladepumpe WP" and has a permanent value of 100.00% for me.
>     * LP is named "LP Ladepumpe WP" and has changing values of 0% up to at least 81%

Perfect! - So it's exatly the same behaviour I'm observing.

fkhr79, 2025-02-21:
Just reviewed the issue to check if there's anything to work on yet. Conclusion (branch main, 2025-02-01):

- Current reading of the modbus register via pymodbus is OK (doublechecked again) ✅ 
- Current assignment to "Ladepumpe WP" is OK (consistent in SC3, Solvis Portal, Solvis Docs) ✅ 
- Value stays constantly at 400 which is in contradiction to the SC3 Modbus Doc. Until further notice we cannot find out if the value is to be expected or if the register is flawed 🟨 

Suggestions:
- Actual Bug is solved: code is OK -> issue could be closed or labeled "solved"
- Until claryfication entity could be disabled by default
- Issue could get the label "need-help-from-solvis"; alternatively/additionally we could create a discussion (or something more fitting at another place) to collect all the cases where help from solvis is needed: this would clean up the issue-section (since it's no issue of the integration itself); furthermore a "known limitations" paragraph could be added to readme.md to inform the users

fkhr79, 2025-03-02:
Evaluation of Excel table by Solvis (see #173): 
- no new info; register is as it is: on or off (output switch)

Since there are no further indications for anything being out of order:
- close issue: there's nothing to be done at the moment (might change with new versions)
- register is ok: should stay enabled
- will leave a hint at supported entities.md

closed 2025-03-02 by fkhr79