LarsK1/hass_solvis_control #55: [FEATURE] Register für elektrische Leistung "PV2Heat"

larsneo, 2024-11-01:
**Is your feature request related to a problem? Please describe.**
Kennt jemand vielleicht das Register für die elektrische Leistung von PV2Heat? 

**Describe the solution you'd like**
Nachdem die Anzeige von elektrischer und thermischer Leistung der Wärmepumpe ja bereits möglich ist, wäre es toll, wenn man auch den Verbrauch von Tim bzw. Tom abbilden könnte - im Solvisportal wird das ja abgebildet.

**Describe alternatives you've considered**
Anfrage bei Solvis nach aktueller Modbus Dokumentation - selbstredend unbeantwortet :-(

**Additional context**
[image]

churge82, 2025-02-13:
If you need any infos from a running Solvis system, I've got a Solvis Ben with a heat pump and a TIM installed and the Modbus enabled.

fkhr79, 2025-02-21:
Till now we found the following undocumented values:

_Input registers_

| Adresse | Solvis Beschreibung                | Unterstützt seit V | Sensortyp | Conf.-Option | Bemerkung | Min | Max    | Einheit                  |
|---------|------------------------------------|--------------------|-----------|--------------|-----------|-----|--------|--------------------------|
| 33550   | Wärmemengenzähler Leistung         | 2.0.0              | 0         | 0            |           |     |        | hW                       |

_Holding registers_

| Adresse | Solvis Beschreibung                | Unterstützt seit V | Sensortyp | Conf.-Option | Bemerkung | Min | Max    | Einheit                  |
|---------|------------------------------------|--------------------|-----------|--------------|-----------|-----|--------|--------------------------|
| 33543   | Leistung Solarthermie             | 1.1.3              | 0         | 3            | not in Solvis Documentation         | 0    |       | kW       |
| 33544   | Wärmepumpe Leistung               | 0.9                | 0         | 4            | not in Solvis Documentation         |      |       | kW       |
| 33545   | Elektrische Wärmepumpenleistung   | 0.9                | 0         | 4            | not in Solvis Documentation         |      |       | kW       |

I think it's worth a try to just sniff around with a Modbus server tool (e. g. Modbus Master) in the neighbourhood of those registers: perhaps we are lucky!? If you @churge82 or @larsneo would give it a try...? Would be great!

Good Hunting! 🙂

fkhr79, 2025-03-02:
@churge82 and @larsneo : found the register you wanted in the Solvis Excel table (see #173) and will merge the commit to add this entity at once.
Since I don't have a PV2Heat I cannot check if the values are OK. So, please, if you would be so kind and have a look? Please give feedback here. There is also a new entity named "PV2Heat Wärmemenge". Please check/verify it's values too. Thanks!

closed 2025-03-02 by fkhr79

larsneo, 2025-03-02:
nice - but it seems that the commit didn't make it into 2.0.0-alpha.10?

fkhr79, 2025-03-02:
@larsneo 👍 Cool! - Thanks a lot for the fast respone!

> nice - but it seems that the commit didn't make it into 2.0.0-alpha.10?

The last release 2.0.0-alpha.10 - if downloaded via hacs - doesn't contain the recent commits. You have to download the main branch of this repo manually to get them. The next release - don't know when this will happen - will contain these commits and make it more comfortable.

> "pv2heat leistung elektrisch" seems to be the current value

Yes, it is the current power consumption with unit kW. 

> "pv2heat wärmemenge" - i have no clue about the timeframe for this value, it's not todays value

It's the **total** (overall) energy value. Since it can get very big, Solvis apparently uses unit "10 kWh" as with other big values. In my first commit I trusted the Excel sheet and set the factor to 1. But - as can be seen in your screenshots - they use factor 10 with this value too.
I'll push a commit in short time and will change the factor to 10. After this the value should be correct. But, mind, rounded to 10 kWh. 

Unfortunately I couldn't find the daily and yesterday values. But: you can derive them quite easily using HA helpers or custom templates.

PR #185 (2025-03-02): Feat: Add new & change existing entities
add reg. 2 - hkr_number
change reg. 33536 - solar_energy
change reg. 33537 - heatpump_energy_thermal
change reg. 33538 - burner_energy_thermal
change reg. 33540 - warm_water_energy
add reg. 33541 - heating_circuits_energy

partly solves #173
fixes #55
commit 24bdfec3 (2025-03-02): Feat: Add new & change existing entities

add reg. 2 - hkr_number
change reg. 33536 - solar_energy
change reg. 33537 - heatpump_energy_thermal
change reg. 33538 - burner_energy_thermal
change reg. 33540 - warm_water_energy
add reg. 33541 - heating_circuits_energy

partly solves #173
fixes #55
commit a5b0ca27 (2025-03-02): Feat: Add new & change existing entities

add reg. 2 - hkr_number
change reg. 33536 - solar_energy
change reg. 33537 - heatpump_energy_thermal
change reg. 33538 - burner_energy_thermal
change reg. 33540 - warm_water_energy
add reg. 33541 - heating_circuits_energy

partly solves #173
fixes #55