LarsK1/hass_solvis_control #58: [BUG] Unpack Error using Solvis SC2

timmost, 2024-11-23:
**Version of the plugin**
1.1.4

**Home-Assistant Version**
Core
2024.11.3
Supervisor
2024.11.4
Operating System
13.2
Frontend
20241106.2

**Enviroment (Docker, HAOS, ...)**

HAOS
**Describe the bug**
Installed the integration as decribed. System tries to connect to SC2(!). Apparently gets a value back that does nto work with unpack library - see log. 

**To Reproduce**
Install, wait for initialization

**Expected behavior**
Connection and parsing of read values form SC2

**Screenshots / Logs**
2024-11-23 17:53:38.024 ERROR (MainThread) [custom_components.solvis_control.coordinator] Unexpected error fetching solvis_control data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 382, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solvis_control/coordinator.py", line 92, in _async_update_data
    register_value = decoder.decode_16bit_int()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/payload.py", line 403, in decode_16bit_int
    return unpack(fstring, handle)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 2 bytes

**Additional context**
I tried the solvis integation from https://github.com/Patrick762/hassio-solvis-modbus that is also written for SC3 - which worked well. After seeing your hint in the issues tried your integration. Seems to start and try to connect; but apparently there is a slightly different handling of the return values:

in your code on   File "/config/custom_components/solvis_control/coordinator.py", line 92, in _async_update_data
(...)
  decoder = BinaryPayloadDecoder.fromRegisters(
                        result.registers, byteorder=Endian.BIG
                    )
                    register_value = decoder.decode_16bit_int()

(...)

While the corresponding line in the other library seems to test the value first: ([hassio-solvis-modbus](https://github.com/Patrick762/hassio-solvis-modbus/tree/main)/[custom_components](https://github.com/Patrick762/hassio-solvis-modbus/tree/main/custom_components)/[solvis_modbus](https://github.com/Patrick762/hassio-solvis-modbus/tree/main/custom_components/solvis_modbus)
/coordinator.py)

                        **if(len(result.registers) == 1):**
                            d = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.BIG)
                            parsed_data[register.name] = round(d.decode_16bit_int() * register.multiplier, 2)

LarsK1, 2024-11-24:
Hi @timmost ,
I've pushed a fix, that should allow you to configure the device for now, since it just skipps the failed entitys. My integration fails for you, since we are also pulling way more data in. Therefore it is reasonable, that it fails. Later on, we definitly should try to get also the SC2 working perfectly.

timmost, 2024-11-24:
Hi @LarsK1 
super and thank you for you fast reaction. however. still seems to struggle - pls see the output. The device is created, but still is not properly initiated. the debug log shows a slightly different behavior, I also restarted the Solvis Webdevice to ensure that this is ok.

One of the issues seems to be the non-existing register 33543 - not sure what the return value is. I commented the solarleistung out, 

2024-11-24 17:26:05.811 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register solar_leistung/33543
2024-11-24 17:26:05.811 ERROR (MainThread) [custom_components.solvis_control.coordinator] Unexpected error fetching solvis_control data

with that commented out in Const.py, I get the following debuglog:
It seems that the polling now works, but the conversion seems to result in this weiird value of "-300".
Best regards Timmo

```
2024-11-24 17:53:54.184 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Polling data
2024-11-24 17:53:54.185 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Connected to Modbus for Solvis
2024-11-24 17:53:54.557 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_1_status
2024-11-24 17:53:54.557 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 2
2024-11-24 17:53:55.056 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_2_status
2024-11-24 17:53:55.056 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 0
2024-11-24 17:53:55.555 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_3_status
2024-11-24 17:53:55.555 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 0
2024-11-24 17:53:56.054 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_4_status
2024-11-24 17:53:56.054 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 0
2024-11-24 17:53:56.558 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_5_status
2024-11-24 17:53:56.558 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 0
2024-11-24 17:53:57.057 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_6_status
2024-11-24 17:53:57.058 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 0
2024-11-24 17:53:57.556 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register gas_power
2024-11-24 17:53:57.556 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 0
2024-11-24 17:53:58.306 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register laufzeit_brennerstufe_1
2024-11-24 17:53:58.306 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 11036
2024-11-24 17:53:58.805 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register laufzeit_brennerstufe_2
2024-11-24 17:53:58.805 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 0
2024-11-24 17:53:59.309 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register outdoor_air_temp
2024-11-24 17:53:59.309 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 70
2024-11-24 17:53:59.807 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register roof_air_temp
2024-11-24 17:53:59.808 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 67
2024-11-24 17:54:00.307 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register cold_water_temp
2024-11-24 17:54:00.307 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 234
2024-11-24 17:54:00.806 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr1_flow_water_temp
2024-11-24 17:54:00.806 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 289
2024-11-24 17:54:01.305 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr2_flow_water_temp
2024-11-24 17:54:01.305 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 268
2024-11-24 17:54:01.809 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr3_flow_water_temp
2024-11-24 17:54:01.809 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Value from previous register before modification: 2500
2024-11-24 17:54:02.308 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_temp
[... cut ...]
2024-11-24 17:54:31.811 WARNING (MainThread) [custom_components.solvis_control.select] The coordinator failed to fetch data for entity: hkr2_betriebsart
2024-11-24 17:54:31.811 WARNING (MainThread) [custom_components.solvis_control.switch] The coordinator failed to fetch data for entity: domestic_water_reheat_start
2024-11-24 17:54:31.812 WARNING (MainThread) [custom_components.solvis_control.switch] The coordinator failed to fetch data for entity: hkr1_warmwasser_vorrang
2024-11-24 17:54:31.812 WARNING (MainThread) [custom_components.solvis_control.switch] The coordinator failed to fetch data for entity: hkr2_warmwasser_vorrang
```

LarsK1, 2024-11-24:
Hi @timmost ,
-300 means, that the values from your Solvis isn't read from your modbus as 16 bit, but something different. I'm honestly not shure, why your device sends different data, then every other :)

You might wanna try to change coordinator.py to:
```
                    try:
                        value = round(
                            decoder.decode_16bit_int() * register.multiplier, 2
                        )
                    except struct.error:
                        try:
                            value = round(
                            decoder.decode_8bit_int() * register.multiplier, 2
                        )
                        except struct.error:
                            parsed_data[register.name] = -300
                    else:
                        parsed_data[register.name] = (
                            abs(value) if register.absolute_value else value
                        )
```

timmost, 2024-11-24:
Hi Lars, 

this fix did not change the behavior. 
I used the other integrations successfully, and from my reading the Modbus definitions are the same for SC2 / SC§ (besides the extras.)
[https://github.com/Patrick762/hassio-solvis-modbus](url)
[https://smarthome.ms1.de/projekte/solvis-heizung/](url)

I saw that you created two more versions - should I try them?

In any case I will see that I clean everything and restart the process perhaps an old library or so...
B

LarsK1, 2024-11-24:
Hi @timmost ,
yes - please install the latest version. The older ones.had some pretty nasty bugs. You don't need to change the files - HASS takes care of that for you.

Probably, some minor adjustments need to be made for SC2 regarding the modbus ports.

timmost, 2024-11-24:
Works!! Cool, thx!
I leave debug logging on, and will check it a bit more over the next days. 
from what i see, the major differences between SC2 and SC3 are some added registers. 
Plus, at least in my firmware, there are two wrong entries / calculations (from the Solvis hotline)
```
Für die SC2 wurde der Fehler leider noch nicht behoben. Immerhin gibt es einen Workaround:
In der Doku zur Remote steht zwar:
33040 8110 Volumenstrom S17 l/min
33041 8111 Volumenstrom S18 l/min
Leider werden stattdessen aber die Zeit von Puls zu Puls in ms übertragen (bzw in 0,5ms Schritten für S18).
Bei 422l/h (S17) und 1,2l/min werden über Remote die Werte 203 und 204 zurückgegeben.
Die Umrechnung lautet also:
(1 / (value / 60)) * 1000) /42*60 für S17 
bzw.
(1 / (value / 60)) * 1000 / 2 /42*60 für l/h für S18.
Wobei die 42 an den Parameter für die Impulse/Liter angepasst werden muss.
Also z.B. (1 / (203 / 60)) * 1000)/42*60 => 422 l/h
``` 

Last but not least, I think that the techies switched / reassigned one or two sensors in my setup. this could be easily changed in the config files, however this would be overridden with each update. 
thx a lot!!

timmost, 2024-11-24:
@LarsK1 
small item - Warmwasser Nachheizen funktioniert auch (Write function). Das könnte man noch anzeigen - da man das nicth ausschalten kann (geht von selber aus). Die anderen Writefunctions lasse ich mal in Frieden.

timmost, 2024-11-25:
Hi Lars, 
noch einer - die Version der SC wirft einen Fehler.:

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 664, in state
    numerical_value = float(value)  # type:ignore[arg-type]
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: '2.05.08'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 266, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 481, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 176, in async_update_listeners
    update_callback()
  File "/config/custom_components/solvis_control/sensor.py", line 192, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1007, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1132, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1069, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 666, in state
    raise ValueError(
ValueError: Sensor sensor.solvis_heizung_version_sc3 has device class 'None', state class 'None' unit '' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '2.05.08' (<class 'str'>)

``` 

Soll ich einzelne Beobachtungen in neue Threads setzen, oder hier weitermachen? und English oder deutsch? 
VG Timmo

LarsK1, 2024-11-25:
Hi @timmost ,
von mir aus gerne Deutsch 😁.

Hmm, komisch - diesen Fehler hatte ich auch und resultiert daraus, dass die unit dieser Entität nicht "None" ist. Ist deine constants.py identisch zur der aktuellsten Version?

LarsK1, 2024-11-25:
@timmost für den Fehler siehe #60

timmost, 2024-11-25:
Ok. habs nochmals komplett runtergeladen und neu installiert, hatte mal mit den Namen gespielt. 

Allerdings ist der Fehler immer noch drin;

Noch ein Gedanke - Du hast ja die HK2 etc im Setup auch parametrisiert. Kann man den gleichen Mechanismus auch für die SC2 verwenden? Die Wärmepumpe ist auch noch da.
VG Timmo

```
2024-11-25 21:41:00.416 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Polling data
2024-11-25 21:41:00.418 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Connected to Modbus for Solvis
2024-11-25 21:41:00.888 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_1_status
2024-11-25 21:41:01.387 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_2_status
2024-11-25 21:41:01.886 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_3_status
2024-11-25 21:41:02.385 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_4_status
2024-11-25 21:41:02.884 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_5_status
2024-11-25 21:41:03.388 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_6_status
2024-11-25 21:41:03.887 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register gas_power
2024-11-25 21:41:04.386 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register laufzeit_brennerstufe_1
2024-11-25 21:41:04.885 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register laufzeit_brennerstufe_2
2024-11-25 21:41:05.384 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register outdoor_air_temp
2024-11-25 21:41:05.888 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register roof_air_temp
2024-11-25 21:41:06.387 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register cold_water_temp
2024-11-25 21:41:06.886 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr1_flow_water_temp
2024-11-25 21:41:07.385 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr2_flow_water_temp
2024-11-25 21:41:07.884 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr3_flow_water_temp
2024-11-25 21:41:08.402 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_temp
2024-11-25 21:41:08.887 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register domestic_water_reheat_start/2322
2024-11-25 21:41:09.636 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_water_temp
2024-11-25 21:41:10.136 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_heat_exchanger_in_water_temp
2024-11-25 21:41:10.635 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_heat_exchanger_out_water_temp
2024-11-25 21:41:11.167 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer1_water_temp
2024-11-25 21:41:11.638 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer2_water_temp
2024-11-25 21:41:12.138 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer3_water_temp
2024-11-25 21:41:12.635 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer4_water_temp
2024-11-25 21:41:13.134 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register cold_water_temperatur
2024-11-25 21:41:13.638 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register number_gas_burner_start
2024-11-25 21:41:14.136 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register ionisation_voltage
2024-11-25 21:41:14.635 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a01_pumpe_zirkulation
2024-11-25 21:41:15.134 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a02_pumpe_warmwasser
2024-11-25 21:41:15.639 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a03_pumpe_hkr1
2024-11-25 21:41:16.138 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a04_pumpe_hkr2
2024-11-25 21:41:16.649 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a05_pumpe_hkr3
2024-11-25 21:41:17.135 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a12_brennerstatus
2024-11-25 21:41:17.635 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_water_flow
2024-11-25 21:41:18.138 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register solar_leistung/33543
2024-11-25 21:41:18.637 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow
[... cut ...]
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 666, in state
    raise ValueError(
ValueError: Sensor sensor.solvis_heizung_version_sc3 has device class 'None', state class 'None' unit '' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '2.05.08' (<class 'str'>)
```

LarsK1, 2024-11-25:
@timmost das ist mir gerade auch aufgefallen, ist aber mit V1.1.8 beides behoben...

timmost, 2024-11-25:
ahh zu früh bin noch auf 1.1.7. gewesen ;-)

```
2024-11-25 21:53:31.884 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register roof_air_temp
2024-11-25 21:53:32.383 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register cold_water_temp
2024-11-25 21:53:32.882 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr1_flow_water_temp
2024-11-25 21:53:33.381 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr2_flow_water_temp
2024-11-25 21:53:33.885 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_temp
2024-11-25 21:53:34.384 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register domestic_water_reheat_start/2322
2024-11-25 21:53:34.886 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_water_temp
2024-11-25 21:53:35.382 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_heat_exchanger_in_water_temp
2024-11-25 21:53:35.881 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_heat_exchanger_out_water_temp
2024-11-25 21:53:36.636 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer1_water_temp
2024-11-25 21:53:37.135 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer2_water_temp
2024-11-25 21:53:37.634 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer3_water_temp
2024-11-25 21:53:38.882 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer4_water_temp
2024-11-25 21:53:39.385 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register cold_water_temperatur
2024-11-25 21:53:39.884 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register number_gas_burner_start
2024-11-25 21:53:40.383 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register ionisation_voltage
2024-11-25 21:53:40.882 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a01_pumpe_zirkulation
2024-11-25 21:53:41.381 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a02_pumpe_warmwasser
2024-11-25 21:53:41.885 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a03_pumpe_hkr1
2024-11-25 21:53:42.384 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a04_pumpe_hkr2
2024-11-25 21:53:42.883 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a12_brennerstatus
2024-11-25 21:53:43.382 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_water_flow
2024-11-25 21:53:43.881 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register solar_leistung/33543
2024-11-25 21:53:44.385 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow
2024-11-25 21:53:44.884 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_betriebsart/2818
2024-11-25 21:53:45.383 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_warmwasser_vorrang/2817
2024-11-25 21:53:45.883 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_vorlaufart/2819
2024-11-25 21:53:46.381 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_fix_vorlauf_tag/2820
2024-11-25 21:53:46.885 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_fix_vorlauf_nacht/2821
2024-11-25 21:53:47.635 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_heizkurve_temp_tag_1/2822
2024-11-25 21:53:48.134 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_heizkurve_temp_tag_2/2823
2024-11-25 21:53:48.633 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_heizkurve_temp_tag_3/2824
2024-11-25 21:53:49.132 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_heizkurve_temp_absenkung/2825
2024-11-25 21:53:49.636 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr1_heizkurve_steilheit/2826
2024-11-25 21:53:50.135 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register raumtemperatur_hkr1/34304
2024-11-25 21:53:50.634 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr2_betriebsart/3074
2024-11-25 21:53:51.133 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr2_vorlaufart/3075
2024-11-25 21:53:51.632 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr2_warmwasser_vorrang/3073
2024-11-25 21:53:52.136 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register hkr2_fix_vorlauf_tag/3076
[... cut ...]
2024-11-25 21:54:28.220 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Connected to Modbus for Solvis
2024-11-25 21:54:28.636 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_1_status
2024-11-25 21:54:29.134 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_2_status
2024-11-25 21:54:29.633 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_3_status
```

LarsK1, 2024-11-26:
Schau mal hier #61

LarsK1, 2024-11-26:
HI @timmost ,
danke für deinen ausführlichen Kommentar:
- _Benennung "circulation flow rate" -> sollte "Circulation Temperature" sein_. Welche Entität meinst Du? Gerne die Stelle aus const.py angeben.
- Punkt 2 sollte mit #61 erledigt sein. Prüf aber gerne einmal die Konvertierung, ob ich das richtig verstanden habe.
- Kennst Du die Umrechnung für 3.?
- Der Bug ist noch offen (siehe #54 )
- Ist auch noch offen, geht mit SC3 auch noch nicht so richtig (#50)
- Es gibt bereits eine Mehrsprachigkeit. Einige Entitäten sind aber noch gemischt. Da hier aber die Integration definitiv "Schaden" verursacht durch andere Namen, möchte ich das erst in einer V2 angehen...
- Das steht bereits auf meiner Todo :)
- Das sollte bereits jetzt gehen, dass wenn du eine Entität deaktivierst, diese nicht mehr abgefragt wird.

LarsK1, 2024-11-27:
@timmost 
Schau mal in #62 
Ich habe die ganzen Features mal in ein Release gepackt: V1.2.0-alpha1. Gerne testen und Feedback geben.

timmost, 2024-11-28:
Gerne- allerdings wird sicherlich Sonntag

Timmo Sturm
Wittelsbacherstr. 8
85622 Feldkirchen
+49 151 402 71 711

larsneo, 2024-11-28:
kleines feedback zum release: das update funktioniert auch mit sc3 (nach erneuter konfiguration) scheinbar prima.
[grafik](https://github.com/user-attachments/assets/ba325b9a-2337-405d-864c-355cb068d5ce)
könnte aber vielleicht noch eine sprechende bezeichnung der version vertragen. danke für die arbeit!

timmost, 2024-11-30:
Hi @LarsK1 
so, jetzt ist Wochenende ... testen geht weiter. Ich teste die 1.2.0-alpha3

- circulation flow rate - das ist in en.json falsch übersetzt:
```
			"cold_water_temp":{
				"name":"circulation flow rate"
``` 
Flow Hot water - da ist die Umrechung noch etwas mehr als Faktor 10 falsch (also wharscheinlich nicht nur eine 0) - kannst du fürs debuggen mal den rohen Sensorwert mit ins Debuglog schreiben - dann kann ich checken, obs die Umrechnung oder der Sensorwert ist. Der Code sah richtig aus.

Flow Solar - gleiche Formel bis auf einen Faktor 2 - der Sensor gibt laut Support die Zeit zwischen Pulsen in 1 ms oder in 0,5ms aus... 
(1 / (value / 60)) * 1000 / 2 /42*60 für l/h für S18

Re Schaden durch Umbenennung von Sensordaten - klar - wäre es denkbar, die Unique_ID zu nutzen, und dadurch die "Friendly Names" von der eineindeutigen internen technischen Benennung zu trennen? [https://developers.home-assistant.io/docs/entity_registry_index/](url) 

Abschalten / Verändern von Entitäten - mit den Schltern von HKR 2/3 - klar, geht. Mein Gedanke war, eine Möglichkeit zu schaffen, nur noch die Werte zu pollen, die später tatsächlich genutzt werden, um die Abfragezeiten zu reduzieren. Die SC2 ist so lahm, dass sie momentan 28 Sekunden für einen Poll braucht, und dadurch das Webinterface sehr sluggish ist (allerdings benutze ich das nur zum debuggen). Plus - da können man in Racetime issues laufen - der alte Poll läuft noch, wenn ein neuer Poll gestartet wird.  Alternative - pro entität die langsame / schnelle Polltime einstellen.

Vielleicht kann man das über "Pro-Feature machen" - eine "alternative const.py" erlauben, die einzelne Parameter nach der normalen const.py einliest und dann überschreibt. Ähnlich wie bei Docker mit compose-override. Keine Ahnnung wie aufwendig das ist, würde aber vermeiden, 100 zusätzliche Switches in die config einzubauen. (Oder man macht das manuell, und zieht das beim Upgrade nach - kein Problem für die Geeks)

```
DEBUG (MainThread) [custom_components.solvis_control.coordinator] Finished fetching solvis_control data in 26.858 seconds (success: True)
``` 

VG!

LarsK1, 2024-11-30:
Hi @timmost ,
danke für dein Feedback. Das deaktivieren ist ganz einfach (und auch schon implementiert). Einfach die Entitäten deaktivieren, die nicht gebraucht werden. Dann werden diese auch nicht mehr abgefragt...
[image]

Die Übersetzungen werden hier gepflegt. Daher gerne dort ändern und wenn Du kannst / möchtest auch weitere Sprachen mit übersetzen.
https://explore.transifex.com/homeassistant-solvis/homeassistant-solvis-plugin/

timmost, 2024-11-30:
ahh super, schaue ich mir an. Hast Du eine chance mit dem rohwert des Flows - dann schaue ich mir die Umrechnerei nochmals an.

LarsK1, 2024-11-30:
Schau mal in die aktuellste Alpha Nr. 4

timmost, 2024-12-01:
Hi @LarsK1 - schaltest Du mich noch für das Übersetzungsprojekt frei?

LarsK1, 2024-12-01:
@timmost erledigt 👍

timmost, 2024-12-01:
@LarsK1 habe mal die englischen Übersetzungen gemacht. Allerdings bin ich mir bei Gross/Kleinschreibungen nicht sicher, da ja Texte am Anfang in der Benutzeroberfläche gross geschrieben werden - danach aber klein. Habe es jetzt erstmal relativ konsistent klein gemacht, da muss man wahrscheinlich in der App nochmals checken. 
Wie läuft das deployment in eine Version - da musst Du wahrscheinlich ein Commit machen?
vG

timmost, 2024-12-01:
@LarsK1 
zu den Readouts Cold Water - da gibt es zwei, einer ist der Zirkulationsrückfluss:

Der S15 / 33038  ist richtig:
    ModbusFieldConfig(  # Kaltwassertemperatur
        name="cold_water_temperatur",
        address=33038,
        unit="°C",
        device_class="temperature",
        state_class="measurement",
        poll_time=0,
    ),

Der S11 / 33034 sollte korrigiert werden (statt "Zirkulationsdurchfluss" und "cold_water_temp" sollte "Zirkulation Rückfluss Temperatur" und "circulation_reflow_temp" oder so.. 
    ModbusFieldConfig(  # Zirkulationsdurchfluss
        name="cold_water_temp",
        address=33034,
        unit="°C",
        device_class="temperature",
        state_class="measurement",
        multiplier=0.1,
        poll_time=0,
    ),
vg

timmost, 2024-12-01:
@LarsK1  
und jetzt hab ich den Umrechungsfehler glaube ich... 
Die Rohwerte für die Volumenströme werden wie die Temperaturen als 0,1C als Einheit interpretiert. Das ist zumindest für die SC2 Werte für den Flow nicht richtig, da hier 0,5 ms für S18 und 1 ms für S17 als Einheit verwendet wird. 
Siehe debug log:
Beim Pollen kommt das hier:
2024-12-01 10:12:41.643 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow
2024-12-01 10:12:41.643 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 60
2024-12-01 10:12:42.646 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow
2024-12-01 10:12:42.646 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: **63**

Die Umrechung liefert aber das hier:
2024-12-01 10:12:48.647 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Returned data: {'analog_out_1_status': 0.2, 'analog_out_2_status': 0.0, 'analog_out_3_status': 0.0, 'analog_out_4_status': 0.0, 'analog_out_5_status': 0.0, 'analog_out_6_status': 0.0, 'gas_power': 0.0, 'laufzeit_brennerstufe_1': 1111.7, 'laufzeit_brennerstufe_2': 0.0, 'outdoor_air_temp': -1.0, 'roof_air_temp': 1.3, 'cold_water_temp': 36.6, 'hkr1_flow_water_temp': 22.1, 'hkr2_flow_water_temp': 26.6, 'domestic_water_temp': 45.4, 'domestic_water_reheat_start': 0, 'solar_water_temp': 22.2, 'solar_heat_exchanger_in_water_temp': 21.5, 'solar_heat_exchanger_out_water_temp': 22.8, 'tank_layer1_water_temp': 22.8, 'tank_layer2_water_temp': 28.0, 'tank_layer3_water_temp': 38.3, 'tank_layer4_water_temp': 59.0, 'cold_water_temperatur': 10.0, 'ionisation_voltage': 0.0, 'a01_pumpe_zirkulation': 0.0, 'a02_pumpe_warmwasser': 0.0, 'a03_pumpe_hkr1': 100.0, 'a04_pumpe_hkr2': 100.0, 'a12_brennerstatus': 0.0, 'solar_water_flow': 0, 'solar_leistung': -300, 'domestic_water_flow': **6.3**, 'raumtemperatur_hkr1': 0.0, 'hkr2_warmwasser_vorrang': -300, 'raumtemperatur_hkr2': 0.0, 'ww_solltemperatur': 46, 'digin_error': 1, 'zirkulation_betriebsart': 3}

timmost, 2024-12-15:
HI @LarsK1 

Bekomme zur Zeit einen Fehler im Log: (Version [1.2.0-alpha4]). Any view?
VG timmo

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:773
integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 5:36:48 PM (1 occurrences)
Last logged: 5:36:48 PM

Platform solvis_control does not generate unique IDs. ID _domestic_water_flow already exists - ignoring sensor.solvis_heizung_flow_rate_hot_water_circulation

LarsK1, 2024-12-29:
@timmost sorry, für die lange Verzögerung. 

Zu deinen Punkten:
> Platform solvis_control does not generate unique IDs. ID _domestic_water_flow already exists - ignoring sensor.solvis_heizung_flow_rate_hot_water_circulation

--> schaue ich mir noch die Tage an.

Bzgl. der Umrechnung. Füg mal `multiplier=1,` in deine const.py an die jeweilige Stelle ein. Dann nimmt er den Wert eins-zu-eins.

Das doppelte Polling sollte eigentlich nicht sein...

timmost, 2024-12-30:
Hi @LarsK1 schaue ich mir morgen an... vG

timmost, 2025-01-05:
@LarsK1 Die Übersetzungen habe ich jetzt nochmals überarbeitet. Die anderen Punkte habe ich noch nicht gecheck, da momentan die Installation nicht durchgeht (siehe #80 )
VG

timmost, 2025-02-02:
@LarsK1 
Habe mal einzelne Bugs aufgemacht, und schliesse den Thread hier .. wird zu unübersichtlich. VG timmo

PR #59 (2024-11-24): Fixes & Version string parsing
Closes #58
closed 2025-02-02 by timmost