LarsK1/hass_solvis_control #60: [BUG] Problem with update to 1.1.7

larsneo, 2024-11-25:
**Version of the plugin**
1.1.7

 **Home-Assistant Version**
2024.11.3

**Enviroment (Docker, HAOS, ...)**
HA green

**Describe the bug**
Various entities disappear with update from 1.1.4 to 1.1.7

**To Reproduce**
update to 1.1.7

**Expected behavior**
all entites remain available

**Screenshots**
[image]
[image]

**Additional context**
Beim Update auf 1.1.7 verschwinden diverse Entitäten, insbesondere die Angaben zur Wärmepumpe. Auch Neustart von HA bzw. Löschen und Neunanlegen des Gerätes machen keinen Unterschied. Firmware der Steuerung: MA3.19.47

LarsK1, 2024-11-25:
Hi @larsneo ,
danke für deine Rückmeldung.

Das verstehe ich nicht. Bei mir sind alle Entitäten noch sichtbar - derweil auch die gleiche Firmware-Version...

Kannst Du bitte noch einmal upgraden und dann den Debug-Log einschalten?

LarsK1, 2024-11-25:
HI @larsneo ,
alles zurück - der Bug sitzt wie immer vor dem PC... Zieh dir mal die neuste Version, damit sollte der Fehler weg sein ...

closed 2024-11-25 by LarsK1