LarsK1/hass_solvis_control #78: TODO: Home Assistant Integration checklist

LarsK1, 2024-12-29:
Hi everyone,
HomeAssistant recently published new criteria for Integration - therefore the next goal will be to score as high as possible :)

## Bronze
- [X ] `config-flow` - Integration needs to be able to be set up via the UI
    - [ ] Uses `data_description` to give context to fields
    - [ ] Uses `ConfigEntry.data` and `ConfigEntry.options` correctly
- [X] `test-before-configure` - Test a connection in the config flow
- [X] `unique-config-entry` - Don't allow the same device or service to be able to be set up twice
- [ ] `config-flow-test-coverage` - Full test coverage for the config flow
- [ ] `runtime-data` - Use ConfigEntry.runtime_data to store runtime data
- [ ] `test-before-setup` - Check during integration initialization if we are able to set it up correctly
- [X] `appropriate-polling` - If it's a polling integration, set an appropriate polling interval
- [ ] `entity-unique-id` - Entities have a unique ID
- [ ] `has-entity-name` - Entities use has_entity_name = True
- [ ] `entity-event-setup` - Entities event setup
- [X] `dependency-transparency` - Dependency transparency
- [ ] `action-setup` - Service actions are registered in async_setup
- [ ] `common-modules` - Place common patterns in common modules
- [ ] `docs-high-level-description` - The documentation includes a high-level description of the integration brand, product, or service
- [ ] `docs-installation-instructions` - The documentation provides step-by-step installation instructions for the integration, including, if needed, prerequisites
- [ ] `docs-removal-instructions` - The documentation provides removal instructions
- [ ] `docs-actions` - The documentation describes the provided service actions that can be used
- [X] `brands` - Has branding assets available for the integration

## Silver
- [X] `config-entry-unloading` - Support config entry unloading
- [ ] `log-when-unavailable` - If internet/device/service is unavailable, log once when unavailable and once when back connected
- [ ] `entity-unavailable` - Mark entity unavailable if appropriate
- [ ] `action-exceptions` - Service actions raise exceptions when encountering failures
- [ ] `reauthentication-flow` - Reauthentication flow
- [ ] `parallel-updates` - Set Parallel updates
- [ ] `test-coverage` - Above 95% test coverage for all integration modules
- [X] `integration-owner` - Has an integration owner
- [ ] `docs-installation-parameters` - The documentation describes all integration installation parameters
- [ ] `docs-configuration-parameters` - The documentation describes all integration configuration options

## Gold
- [X] `entity-translations` - Entities have translated names
- [X] `entity-device-class` - Entities use device classes where possible
- [X] `devices` - The integration creates devices
- [X] `entity-category` - Entities are assigned an appropriate EntityCategory
- [X] `entity-disabled-by-default` - Integration disables less popular (or noisy) entities
- [ ] `discovery` - Can be discovered
- [ ] `stale-devices` - Clean up stale devices
- [X] `diagnostics` - Implements diagnostics
- [X] `exception-translations` - Exception messages are translatable
- [ ] `icon-translations` - Icon translations
- [X] `reconfiguration-flow` - Integrations should have a reconfigure flow
- [ ] `dynamic-devices` - Devices added after integration setup
- [ ] `discovery-update-info` - Integration uses discovery info to update network information
- [ ] `repair-issues` - Repair issues and repair flows are used when user intervention is needed
- [ ] `docs-use-cases` - The documentation describes use cases to illustrate how this integration can be used
- [ ] `docs-supported-devices` - The documentation describes known supported / unsupported devices
- [ ] `docs-supported-functions` - The documentation describes the supported functionality, including entities, and platforms
- [ ] `docs-data-update` - The documentation describes how data is updated
- [X] `docs-known-limitations` - The documentation describes known limitations of the integration (not to be confused with bugs)
- [ ] `docs-troubleshooting` - The documentation provides troubleshooting information
- [ ] `docs-examples` - The documentation provides automation examples the user can use.

## Platinum
- [ ] `async-dependency` - Dependency is async
- [ ] `inject-websession` - The integration dependency supports passing in a websession
- [ ] `strict-typing` - Strict typing

timmost, 2025-02-02:
Hi @LarsK1 , wie kann man helfen?
VG timmo

LarsK1, 2025-02-02:
Hi @timmost ,
Du könntest super bei den Themen mit docs-* helfen. Am einfachsten klonst Du Dir das Repo und bearbeitest die README.md.

timmost, 2025-02-06:
(Working in my fork)
done:
- docs-high-level-description - The documentation includes a high-level description of the integration brand, product, or service

fkhr79, 2025-02-19:
I was curious what's meant with these criteria and had a look at the dev pages: the checklist on [the developers page](https://developers.home-assistant.io/docs/core/integration-quality-scale/checklist) looks different...?!

LarsK1, 2025-02-21:
@fkhr79 seems to have changed in order, but not regarding it's content...

closed 2025-04-27 by LarsK1