LarsK1/hass_solvis_control #80: [BUG] Your bug title

timmost, 2025-01-05:
**Version of the plugin**
1.2

 **Home-Assistant Version**
Core
2025.1.0
Supervisor
2024.12.3
Operating System
14.1
Frontend
20250103.0

**Enviroment (Docker, HAOS, ...)**
haos

**Version of your Solvis software/hardware**
Version NBG:
Version SC3:

**Describe the bug**
Installation (from scratch) fails with error:
AttributeError: 'ExceptionResponse' object has no attribute 'registers'

**To Reproduce**
Steps to reproduce the behavior:
Resinstall with SC2, HKR2,Solar Collector

**Expected behavior**
No errors ;-)

**Screenshots**
N/A

**Additional context**
N/A
Debug Log:

`AttributeError: 'ExceptionResponse' object has no attribute 'registers'
2025-01-05 12:14:31.664 ERROR (MainThread) [custom_components.solvis_control.coordinator] Unexpected error fetching solvis_control data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 379, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solvis_control/coordinator.py", line 101, in _async_update_data
    decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.BIG)
                                                 ^^^^^^^^^^^^^^^^
AttributeError: 'ExceptionResponse' object has no attribute 'registers'
2025-01-05 12:14:43.346 INFO (MainThread) [custom_components.solvis_control.config_flow] Initialize config flow for solvis_control
2025-01-05 12:14:58.679 DEBUG (MainThread) [custom_components.solvis_control] Migrating configuration from version 1.4
2025-01-05 12:14:58.679 INFO (MainThread) [custom_components.solvis_control] Migration to version 1_4 successful
2025-01-05 12:14:58.679 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Creating Modbus client
2025-01-05 12:14:58.679 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Polling data
2025-01-05 12:14:58.681 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Connected to Modbus for Solvis
2025-01-05 12:14:59.168 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_1_status
2025-01-05 12:14:59.168 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 2
2025-01-05 12:14:59.667 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_2_status
2025-01-05 12:14:59.667 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:00.171 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_3_status
2025-01-05 12:15:00.171 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:00.670 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_4_status
2025-01-05 12:15:00.670 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:01.169 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_5_status
2025-01-05 12:15:01.169 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:01.668 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_6_status
2025-01-05 12:15:01.668 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:02.167 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register gas_power
2025-01-05 12:15:02.167 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:02.671 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register laufzeit_brennerstufe_1
2025-01-05 12:15:02.671 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 11617
[... 113 lines cut ...]
2025-01-05 12:15:32.672 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:33.171 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a01_pumpe_zirkulation
2025-01-05 12:15:33.171 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:33.675 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a02_pumpe_warmwasser
2025-01-05 12:15:33.676 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:34.174 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a03_pumpe_hkr1
2025-01-05 12:15:34.175 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 25600
2025-01-05 12:15:34.674 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a04_pumpe_hkr2
2025-01-05 12:15:34.674 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 25600
2025-01-05 12:15:35.173 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register a12_brennerstatus
2025-01-05 12:15:35.173 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:35.672 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_water_flow
2025-01-05 12:15:35.672 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-05 12:15:36.176 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register solar_leistung/33543
2025-01-05 12:15:36.176 ERROR (MainThread) [custom_components.solvis_control.coordinator] Unexpected error fetching solvis_control data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 379, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solvis_control/coordinator.py", line 101, in _async_update_data
    decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.BIG)
                                                 ^^^^^^^^^^^^^^^^
AttributeError: 'ExceptionResponse' object has no attribute 'registers'
2025-01-05 12:15:36.178 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Finished fetching solvis_control data in 16.044 seconds (success: False)
`

timmost, 2025-01-06:
@LarsK1 
Habe das nochmals mit der Beta 4 getestet, gleiches Problem. Wenn ich die Config für Solar ausschalte, dann läuft es durch. Es scheint, dass in der neuen Homeassistant die Fehlerbehandlung strikter ist - der Fehler tritt ja bei beim Pollen der Solarleistung auf. Das Register gibts nicht beid er SC"...
```
2025-01-05 12:15:36.176 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register solar_leistung/33543
2025-01-05 12:15:36.176 ERROR (MainThread) [custom_components.solvis_control.coordinator] Unexpected error fetching solvis_control data

```

LarsK1, 2025-01-11:
@timmost probiere es noch einmal mit der Code aus der "main"-branch

timmost, 2025-01-11:
Hi @LarsK1 

 ich habd das auch mit der 1.2.0 ausprobiert, Fehler bleibt. Ich denke, es hängt an der Solarleistung / dass es das registern nicht gibt, Log sieht exakt gleich aus. 

VG 
```
2025-01-11 14:58:03.774 DEBUG (MainThread) [custom_components.solvis_control] Migrating configuration from version 1.4
2025-01-11 14:58:03.774 INFO (MainThread) [custom_components.solvis_control] Migration to version 1_4 successful
2025-01-11 14:58:03.774 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Creating Modbus client
2025-01-11 14:58:03.774 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Polling data
2025-01-11 14:58:03.776 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Connected to Modbus for Solvis
2025-01-11 14:58:04.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_1_status
2025-01-11 14:58:04.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 2
2025-01-11 14:58:05.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_2_status
2025-01-11 14:58:05.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 14:58:05.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_3_status
2025-01-11 14:58:05.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 14:58:06.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_4_status
2025-01-11 14:58:06.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 14:58:07.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_5_status
2025-01-11 14:58:07.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 14:58:07.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_6_status
2025-01-11 14:58:07.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 14:58:08.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register gas_power
2025-01-11 14:58:08.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 57
2025-01-11 14:58:09.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register laufzeit_brennerstufe_1
2025-01-11 14:58:09.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 11710
2025-01-11 14:58:10.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register laufzeit_brennerstufe_2
2025-01-11 14:58:10.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 14:58:10.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register outdoor_air_temp
2025-01-11 14:58:10.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: -2
2025-01-11 14:58:11.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register roof_air_temp
2025-01-11 14:58:11.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 59
2025-01-11 14:58:12.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register cold_water_temp
2025-01-11 14:58:12.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 355
2025-01-11 14:58:12.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr1_flow_water_temp
2025-01-11 14:58:12.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 333
2025-01-11 14:58:13.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr2_flow_water_temp
2025-01-11 14:58:13.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 292
2025-01-11 14:58:14.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_temp
2025-01-11 14:58:14.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 412
2025-01-11 14:58:14.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register domestic_water_reheat_start/2322
2025-01-11 14:58:14.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 14:58:15.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_water_temp
2025-01-11 14:58:15.511 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 118
2025-01-11 14:58:16.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_heat_exchanger_in_water_temp
2025-01-11 14:58:16.010 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 158
2025-01-11 14:58:16.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_heat_exchanger_out_water_temp
2025-01-11 14:58:16.762 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 221
[... cut ...]
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solvis_control/coordinator.py", line 101, in _async_update_data
    decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.BIG)
                                                 ^^^^^^^^^^^^^^^^
AttributeError: 'ExceptionResponse' object has no attribute 'registers'
2025-01-11 14:58:25.763 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Finished fetching solvis_control data in 21.989 seconds (success: False)

```

LarsK1, 2025-01-11:
Hi @timmost,
du musst bitte die aktuellste von heute morgen nehmen. Der Bug sollte in [3b4fe06f80f97e926ed342c17d3a4f8e1761ed21](https://github.com/LarsK1/hass_solvis_control/commit/3b4fe06f80f97e926ed342c17d3a4f8e1761ed21) gefixt worden sein.

timmost, 2025-01-11:
@LarsK1 hm... wie lade ich die von HASS? Gibts da eine neue Beta, oder muss ich über github gehen?

LarsK1, 2025-01-11:
@timmost Ich habe gerade 2.0.0-alpha1 für dich erstellt.

timmost, 2025-01-11:
@LarsK1 
zu früh... jetzt habe ich einige Warnings im Log, sind aber nicht unbedingt mit dem vorherigen Thema verbandelt:

```
2025-01-11 15:58:30.720 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_2_status
2025-01-11 15:58:30.720 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 15:58:31.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_3_status
2025-01-11 15:58:31.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 15:58:32.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_4_status
2025-01-11 15:58:32.014 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 15:58:32.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_5_status
2025-01-11 15:58:32.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 15:58:33.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register analog_out_6_status
2025-01-11 15:58:33.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 15:58:34.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register gas_power
2025-01-11 15:58:34.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 91
2025-01-11 15:58:34.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register laufzeit_brennerstufe_1
2025-01-11 15:58:34.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 11711
2025-01-11 15:58:35.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register laufzeit_brennerstufe_2
2025-01-11 15:58:35.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 15:58:36.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register outdoor_air_temp
2025-01-11 15:58:36.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: -6
2025-01-11 15:58:36.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register roof_air_temp
2025-01-11 15:58:36.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 20
2025-01-11 15:58:37.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register cold_water_temp
2025-01-11 15:58:37.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 319
2025-01-11 15:58:38.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr1_flow_water_temp
2025-01-11 15:58:38.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 322
2025-01-11 15:58:38.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register hkr2_flow_water_temp
2025-01-11 15:58:38.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 279
2025-01-11 15:58:39.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_temp
2025-01-11 15:58:39.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 414
2025-01-11 15:58:40.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register domestic_water_reheat_start/2322
2025-01-11 15:58:40.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-01-11 15:58:40.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_water_temp
2025-01-11 15:58:40.760 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 155
2025-01-11 15:58:41.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_heat_exchanger_in_water_temp
2025-01-11 15:58:41.509 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 154
2025-01-11 15:58:42.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_heat_exchanger_out_water_temp
2025-01-11 15:58:42.013 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 234
2025-01-11 15:58:43.617 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer1_water_temp
2025-01-11 15:58:43.617 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register tank_layer1_water_temp: Exception Response(132, 4, GatewayNoResponse)
2025-01-11 15:58:44.510 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer2_water_temp
2025-01-11 15:58:44.510 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 283
2025-01-11 15:58:45.009 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register tank_layer3_water_temp
2025-01-11 15:58:45.009 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 344
[... cut ...]
    await asyncio.shield(awaitable)
  File "/config/custom_components/solvis_control/switch.py", line 183, in async_setup_entry
    entity_registry = await hass.helpers.entity_registry.async_get_registry()
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1532, in __getattr__
    value = getattr(self._module, attr)
AttributeError: module 'homeassistant.helpers.entity_registry' has no attribute 'async_get_registry'
```

timmost, 2025-01-19:
hi @LarsK1 ,

ich glaube, die solar_leistung ist immer noch bei der SSC2 Konfiguration enthalten, das müsste noch disabled werden, die gibt es da nicht. Errorhandling klappt, aber es wirft einen Fehler beim Modbus und eine Division by 0:

```
2025-01-19 16:06:48.271 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading holding register solar_leistung/33543
2025-01-19 16:06:48.272 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register solar_leistung: Exception Response(132, 4, IllegalAddress)
```
```
2025-01-19 16:06:52.519 WARNING (MainThread) [custom_components.solvis_control.sensor] No data available for solar_leistung
2025-01-19 16:06:52.519 WARNING (MainThread) [custom_components.solvis_control.sensor] Division by zero
```
VG

timmost, 2025-02-02:
@LarsK1 lass uns den schliessen, habe die einzelnen Sachen in neue Bugs gepackt. VG timmo

closed 2025-02-02 by timmost