LarsK1/hass_solvis_control #81: [BUG] Removing old entities complains about await and doesn't run

pz-earthling, 2025-01-25:
**Version of the plugin**
2.0.0-alpha2

 **Home-Assistant Version**
2025.1.4

**Enviroment (Docker, HAOS, ...)**
Docker

**Version of your Solvis software/hardware**
Version NBG: "no data available"
Version SC3:  "no data available"

**Describe the bug**
The code for removing old entities seems to be broken in conjunction with latest Home Assistant. The currently checked-in version produces this result:
2025-01-26 00:05:40.293 ERROR (MainThread) [custom_components.solvis_control.sensor] Error removing old entities: object EntityRegistry can't be used in 'await' expression
2025-01-26 00:05:40.313 ERROR (MainThread) [custom_components.solvis_control.number] Error removing old entities: object EntityRegistry can't be used in 'await' expression
2025-01-26 00:05:40.316 ERROR (MainThread) [custom_components.solvis_control.select] Error removing old entities: object EntityRegistry can't be used in 'await' expression
2025-01-26 00:05:40.320 ERROR (MainThread) [custom_components.solvis_control.switch] Error removing old entities: object EntityRegistry can't be used in 'await' expression

Simply removing the "await" just leads to **all** entities being removed and none being added. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start latest HA with latest integration version
4. See error

**Expected behavior**
Old entities are removed without error, new entities are not removed.

**Screenshots**
n/a

**Additional context**
I actually upgraded from an earlier version of the integration. Right after upgrading a lot of unavailable entities with old names were visible in HA. I then removed the await and got all entities of the integration removed, then I re-added the await and got the expected entities with new (I guess the translation changed?) names back.

timmost, 2025-02-02:
Same here with 2.0.0-alpha2

LarsK1, 2025-02-02:
Fixes with alpha3

closed 2025-02-02 by LarsK1