LarsK1/hass_solvis_control #84: [BUG] Domestic Water Flow - SC2/SC3 selection, numeric conversion

timmost, 2025-02-02:
**Version of the plugin**
2.0.0-alpha2

 **Home-Assistant Version**
2025.1.4
supervisor 2024.12.3
OS 14.2
Frontend 2025019.2

**Enviroment (Docker, HAOS, ...)**
HAOS

**Version of your Solvis software/hardware**
Version NBG: 3.00.00
Version SC3: 2.05.08

Hi @LarsK1 

ich habe mal einen separaten Thead für den Warmwasserduchfluss aufgemacht, nachdem ich mich heute nochmals drangesetzt habe. 

Erste Beobachtung:
- im const.py gibt es zwei Einträge für das Register (Unterschied SC2/SC3?) - aber scheinbar werden beide Einträge durchlaufen - also die Abfrage der "supported version" scheint irgendwie nicht beachtet zu werden?
```
    ModbusFieldConfig(  # Durchfluss Warmwasserzirkualation
        name="domestic_water_flow",
        address=33041,
        unit="l/min",
        device_class=None,
        state_class="measurement",
        supported_version=1,
        poll_time=0,
    ),
    ModbusFieldConfig(  # Durchfluss Warmwasserzirkualation
        name="domestic_water_flow",
        address=33041,
        unit="l/min",
        device_class=None,
        state_class="measurement",
        supported_version=2,
        data_processing=2,
        poll_time=0,
    ),
```

Ich habe auch immer zwei Polls im Debug log:
´´´
2025-02-01 16:38:40.889 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow
2025-02-01 16:38:40.889 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 209
2025-02-01 16:38:41.638 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow
2025-02-01 16:38:41.638 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 212
´´´

Ich habe das mal in der Const.py umbenannt (flow1 / flow2):

´´´
    ModbusFieldConfig(  # Durchfluss Warmwasserzirkualation
        name="domestic_water_flow1",
        address=33041,
        unit="l/min",
        device_class=None,
        state_class="measurement",
        supported_version=1,
        poll_time=0,
    ),
    ModbusFieldConfig(  # Durchfluss Warmwasserzirkualation
        name="domestic_water_flow2",
        address=33041,
        unit="l/min",
        device_class=None,
        state_class="measurement",
        supported_version=2,
        data_processing=2,
        poll_time=0,
    ),
´´´
Das wird auch tatsächlich entspreched im Log ausgegeben:
´´´2025-02-02 14:12:17.751 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow1
2025-02-02 14:12:17.752 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-02-02 14:12:18.251 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow2
2025-02-02 14:12:18.251 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
´´´
Ich habe jetzt erstmal die Konfiguration für die SC3 rausgenommen (also teste jetzt mit "domestic_water_flow2"), das funktioniert. 

Die Umrechnung ist aber noch offen, da der Registerwert die Zeit zwischen zwei Impulsen anzeigt, und damit der Multiplier nicht hilft (Kehrwert - kleinere Impulszeit, größerer Flow)

Umrechnung für den Warmwasserstrom:
```
(1 / (value / 60)) * 1000 / 2 /42*60 für l/h für S18.
``` 
oder gekürzt:
```
(5000 / 7) * (1/ value) = l/min für S18.
```

Da der Wert in S18 im Nenner steht, geht die Umrechung mit Multiplier nicht... habe aber noch nicht rausgefunden, wo Du die conversion machst ;-)

Das Gleiche gilt für S17, wo allerdings ein Faktor 2 reinkommt.

VG timmo

LarsK1, 2025-02-02:
Hi,
sorry für die späte Antwort. Schau mal hier: https://github.com/LarsK1/hass_solvis_control/blob/851f282f4438e791d720bdc0a2fe231d10f76e0e/custom_components/solvis_control/sensor.py#L210

timmost, 2025-02-02:
da bin ich gerade dran...habe es auch gerade gefunden,  teste gerade ;-)

timmost, 2025-02-02:
@LarsK1 So jetzt hab ich es...;-)

für den Warmwasserdurchfluss muss die Zeile 212 in Sensor.py
                    self._attr_native_value = ((1 / (response_data / 60)) * 1000) / 42 * 60

ersetzt werden durch 
                    self._attr_native_value = (1 / (response_data / 60)) * 1000 / 2 / 42

der WW sensor braucht den divisor 2, und für l/min muss die 60 weg. Die Klammern sollten egal sein

Wharscheinlich ist es am einfachsten, wenn wir einen case 3 für den Solardurchfluss machen, wo dann die gleiche Formel, aber ohne die division durch 2 drin ist - so dass die Berechnung jeweils in einem Schritt paasiert (ohne hilfe von multipliers)

VG timmo

LarsK1, 2025-02-02:
Hi,
check mal den Commit, ob das so passt :)

timmost, 2025-02-02:
doofe frage - wie lade ich das? in HACS habe ich ein "update information gemacht" - da zeigt er auch Activity 2 min ago. Aber das redownload der Alpha 2 liefert mir nicht deine geänderten files. brauchts da eine neue Alpha, oder kann ich die Dateien direkt aus dem Github updaten?

LarsK1, 2025-02-02:
Entweder nimmst Du die Dateien aus Github und ersetzt die auf deinem System oder Du wartest noch eine bis zwei Stunden dann mache ich einen neuen Release

timmost, 2025-02-02:
alles klar

timmost, 2025-02-02:
Ok, Rechnung klappt.  Die Precision ist aber noch auf Default - ist das richtig (für sensor.solvis_heizung_flow_rate_warm_water_circulation). 

Der Bug mit der Versionsabfrage ist noch da - der domestic waterflow wird immer noch zweimal abgefragt (s.o.) 
Name der Variable evtl auch noch korrigieren, da es nicht die Zirkulation ist. 

Vielen Dank!
VG timmo

LarsK1, 2025-02-03:
@timmost gib gerne mal Feedback mit der Alpha3 - sollten jetzt eigentlich fast alle Fehler weg sein.

timmost, 2025-02-03:
Hi Lars, mache ich, wird aber ein paar Tage dauern (Job…)

Viele Grüße - Timmo

fkhr79, 2025-02-08:
Integration still yields two sensors:

1. "...durchfluss_warmwasser"  and
2. "...durchfluss_warmwasser_2"

Value of 1 is wrong by a factor of 10. Should be scaled by "0.1".
Value of 2 is totally wrong: reading of "11.52073..." - should be "6.2"

Unit of both sensors should be changed to HA-Standard: "L/min". (Yes... I know it's weird)

After a little bit of debugging (and a massive usage of hot water for testing ;) ) I found a solution:

You don't need the data processing (meaning also the case-statement in sensor.py). I set data processing to "0" in my local const.py.
You should also remove the "multiplier=1" in lines 411 and 422 since the standard multiplier value is 0.1 which is exactly the multiplier we want.
To avoid additional tweaking I left both ModbusFieldConfig-entries as they were. But in a final commit you could delete one Config and set supported_version=0.
Since I formerly used SC2 and am now using SC3 I'm quite sure, that the following config should be valid for both versions:

```
    ModbusFieldConfig(  # Volumenstrom Warmwasser (S18)
        name="domestic_water_flow",
        address=33041,
        unit="L/min",
        device_class=None,
        state_class="measurement",
        poll_time=0,
    ),
```
As you can see I also changed the name in the commentary to a more fitting one.

At least in my setup this works and yields correct values - I doublechecked using the modbus-integration.

Version: 2.0.0-alpha5
on
Home-Assistant Version: 2025.2.1 
Enviroment: HA OS 14.2
Solvis Version NBG: 3.1.0
Solvis Version SC3: 3.19.47

timmost, 2025-02-09:
with Alpha 5 the integration behaves correctly with SC2.

```
2025-02-09 15:22:25.133 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register solar_water_flow/33040
2025-02-09 15:22:25.133 DEBUG (MainThread) [custom_components.solvis_control.coordinator] raw value: 0.0
2025-02-09 15:22:25.133 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Supported version: 2 / Register version: 1
2025-02-09 15:22:25.133 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Skipping SC3 entity for SC2 device: solar_leistung/33543
2025-02-09 15:22:25.134 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Supported version: 2 / Register version: 1
2025-02-09 15:22:25.134 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Skipping SC3 entity for SC2 device: domestic_water_flow/33041
2025-02-09 15:22:25.637 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow/33041
2025-02-09 15:22:25.637 DEBUG (MainThread) [custom_components.solvis_control.coordinator] raw value: 0.0
```