LarsK1/hass_solvis_control #85: [BUG] Smaller stuff

timmost, 2025-02-02:
**Version of the plugin**
2.0.0-alpha2

 **Home-Assistant Version**
2025.1.4
supervisor 2024.12.3
OS 14.2
Frontend 2025019.2

**Enviroment (Docker, HAOS, ...)**
HAOS

**Version of your Solvis software/hardware**
Version NBG: 3.00.00
Version SC3: 2.05.08

Hi @LarsK1 
ein paar Sachen, die mir noch in den Variablen / Language-Files aufgefallen sind - evtl muss man da die Variablen umbenennen.

**translations / de.json**
- Zeile 220:
		"cold_water_temp":{
				"name":"Zirkulationsdurchfluss"
Übersetzung - sollte wahrscheinlich "Kaltwassertemp" sein
- 
- Zeile 235
			"solar_water_temp":{
				"name":"Wasserkreislauftemperatur"
Übersetzung - sollte wahrscheinlich "Solarkreis Temparatur" sein

- Zeile 289
 			"domestic_water_flow":{
				"name":"Durchfluss Warmwasserzirkualation"
Übersetzung - sollte wahrscheinlich "Durchfluss Warmwasser" sein

- Zeile 292
			"version_sc3":{
				"name":"Version SC3"
Wäre besser, das mit "Version SC" zu versehen - da Du ja auch SC2 bedienst

**translations / en.json**
Zeile 219
			"cold_water_temp":{
				"name":"Circulation flow rate"
Übersetzung - sollte wahrscheinlich "Temperatur WW Zirkulation" /Temperature Warm Water Circulation" sein
Zeile 235
			"solar_water_temp":{
				"name":"Water circulation temp"
Übersetzung - sollte wahrscheinlich "Temperatur Solar" sein

Zeile 259
			"cold_water_temperatur":{
				"name":"Cold water temp"
Ist die redundant zu 219?

Zeile 265
			"ionisation_voltage":{
				"name":"Ionization current"
Die Variable scheint falsch benannt zu sein (da es um den Strom geht)

VG Timmo

timmost, 2025-02-02:
Oee more topic
file diagnostics.py is misspelled - "diagonstics.py" - might cause errors

LarsK1, 2025-02-02:
Hi @timmost ,
ich habe gerade die dt. korrigiert. Die Englischen müssten wir in der Übersetzungsplattform ändern, sonst werden die automatisch überschrieben.

timmost, 2025-02-02:
ok

closed 2025-02-03 by LarsK1