LarsK1/hass_solvis_control #86: Entities for SC2 and SC3 are shown

LarsK1, 2025-02-02:
> Der Bug mit der Versionsabfrage ist noch da - der domestic waterflow wird immer noch zweimal abgefragt (s.o.) 

 _Originally posted by @timmost in [#84](https://github.com/LarsK1/hass_solvis_control/issues/84#issuecomment-2629462579)_

timmost, 2025-02-03:
Hi @LarsK1 
scheint immer noch der Fall zu sein - bin auf der alpha 3, alles vorher gelöscht etc... 
ein sensor mit "solvis_heizung_flow_rate_warm_water_circulation"
und dann noch einer mit 
"solvis_heizung_flow_rate_warm_water_circulation_2" - der rechnet richtig.-..

ich habe aber noch keinen Schimmer warum der Sensor hier unterschiedlich benannt wird... kann das noch ein verwaister Eintrag sein? Ich hatte allerdings alle Entries gelöscht, neu gestartet etc.. 

Im log wird auch zweimal gecheckt

```
2025-02-03 21:39:56.474 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow/33041
2025-02-03 21:39:56.474 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0
2025-02-03 21:39:56.973 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow/33041
2025-02-03 21:39:56.973 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 0

(... wasserhahn an)

2025-02-03 21:43:38.472 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow/33041
2025-02-03 21:43:38.473 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 225
2025-02-03 21:43:39.225 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow/33041
2025-02-03 21:43:39.225 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Decoded raw value: 225
``` 
in der const.py scheint es richtig angelegt zu sein:

```
    ModbusFieldConfig(  # Durchfluss Warmwasserzirkualation
        name="domestic_water_flow",
        address=33041,
        unit="l/min",
        device_class=None,
        state_class="measurement",
        supported_version=1,
        poll_time=0,
    ),
    ModbusFieldConfig(  # Durchfluss Warmwasserzirkualation
        name="domestic_water_flow",
        address=33041,
        unit="l/min",
        device_class=None,
        state_class="measurement",
        supported_version=2,
        data_processing=2,
        multiplier=1,
        poll_time=0,
    ),
``` 
hhmmm...

LarsK1, 2025-02-04:
Kannst Du bitte einmal das Debugging einschalten, kurz warten. Dann die Integration löschen, neu einrichten und dann noch eine Minute warten und dann das Debugging beenden und mir dann die daraus entstanden Logs zu schicken...

timmost, 2025-02-04:
Yep, habe den Eindruck, dass schon zwei Sensoren angelegt werden. Heute
Abend… Viele Grüße - Timmo

Timmo Sturm
Wittelsbacherstr. 8
85622 Feldkirchen
+49 151 402 71 711

timmost, 2025-02-06:
@LarsK1 
here you are VG timmo

[home-assistant_solvis_control_2025-02-06T17-49-08.710Z.log](https://github.com/user-attachments/files/18694156/home-assistant_solvis_control_2025-02-06T17-49-08.710Z.log)

LarsK1, 2025-02-06:
@timmost please update to latest alpha. Should resolve your issue

LarsK1, 2025-02-08:
@timmost teste gerne noch einmal mit der Alpha5

timmost, 2025-02-08:
@LarsK1 

Hi Lars, teste gerade mit der Alpha5, der Warmwasser Flow wird immer noch doppelt gezogen. Der zweite Wert ist richtig konvertiert, hat aber keine Rundung, der erste ist ohne Umrechnung (also ist die Unterscheidung SC2/3 bei der Auswertung der Daten) richtig.

Kann es sein, dass da schon beim Anlegen ein Fehler gemacht wird und die Version (über "supported version") nicht beachtet wird?  Solar Thermal power wird nämlich auch noch angelegt, obwohl auch über "supported_version=1" richtig parametrisiert ist. 
Im Log habe ich das Device nochmals gelöscht und wieder angelegt,
```
2025-02-08 16:21:08.526 DEBUG (MainThread) [custom_components.solvis_control.sensor] Erstellte unique_id: 33543_1_solar_leistung
2025-02-08 16:21:08.526 DEBUG (MainThread) [custom_components.solvis_control.sensor] Supported version: 2 / Register version: 1
2025-02-08 16:21:08.527 DEBUG (MainThread) [custom_components.solvis_control.sensor] Erstellte unique_id: 33041_1_domestic_water_flow
2025-02-08 16:21:08.527 DEBUG (MainThread) [custom_components.solvis_control.sensor] Supported version: 2 / Register version: 2
2025-02-08 16:21:08.527 DEBUG (MainThread) [custom_components.solvis_control.sensor] Erstellte unique_id: 33041_2_domestic_water_flow
2025-02-08 16:21:08.527 DEBUG (MainThread) [custom_components.solvis_control.sensor] Supported version: 2 / Register version: 0
2025-02-08 16:21:08.527 DEBUG (MainThread) [custom_components.solvis_control.sensor] Erstellte unique_id: 2819_0_hkr1_vorlaufart
2025-02-08 16:21:08.527 DEBUG (MainThread) [custom_components.solvis_control.sensor] Supported version: 2 / Register version: 0
```  

[image]

[image]

[image]

[home-assistant_solvis_control_2025-02-08T15-31-25.869Z.log](https://github.com/user-attachments/files/18719360/home-assistant_solvis_control_2025-02-08T15-31-25.869Z.log)

closed 2025-02-08 by LarsK1

timmost, 2025-02-09:
@LarsK1 mit Alpha 5:
Jetzt läufts, die Selektion nach SC2/3 ist jetzt auch im Log sichtbar
VG timmo
```
2025-02-09 15:49:34.163 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Skipping SC3 entity for SC2 device: solar_leistung/33543
2025-02-09 15:49:34.163 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Supported version: 2 / Register version: 1
2025-02-09 15:49:34.163 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Skipping SC3 entity for SC2 device: domestic_water_flow/33041
2025-02-09 15:49:34.661 DEBUG (MainThread) [custom_components.solvis_control.coordinator] Reading input register domestic_water_flow/33041
2025-02-09 15:49:34.661 DEBUG (MainThread) [custom_components.solvis_control.coordinator] raw value: 0.0
```

commit 1a04ec0a (2025-02-02): Fixes #86 and correct version display
commit 844680a0 (2025-02-08): Fixes #86