LarsK1/hass_solvis_control #89: [BUG] Your bug title

larsneo, 2025-02-04:
**Version of the plugin**
2.0.0-alpha3

 **Home-Assistant Version**
2025.1.4

**Enviroment (Docker, HAOS, ...)**
HA green

**Version of your Solvis software/hardware**
Version NBG:  3.01.00 
Version SC3:  3.19.47 

**Describe the bug**
Misleading name "Solvis Heizung Kaltwasserzirkulation" in alpha3 (formerly: sensor.solvis_heizung_zirkulationsdurchfluss )

**To Reproduce**
upgrade to alpha3 changes the entity name

**Expected behavior**
maybe it's an option to rename the sensors according to the solvis backend (the Sy Sensors in screenshot) to be consistent

**Screenshots**

[image]

**Additional context**
The update itself words like a charm, thx for all your efforts :-)

fkhr79, 2025-02-08:
If "Solvis Heizung Kaltwasserzirkulation" is the measurement of S11: agreed! 
In SC3 you'll find "Zirkulationstemperatur" as well as in the manuals.

Adding the (internal) sensor labels of Solvis to the entity names in general would be great.

Version: 2.0.0-alpha5
on
Home-Assistant Version: 2025.2.1 
Enviroment: HA OS 14.2
Solvis Version NBG: 3.1.0
Solvis Version SC3: 3.19.47

LarsK1, 2025-02-09:
Hi @fkhr79 / @larsneo ,
I would finally like to take the step, and prepend the "solvis"-names Ax, Sx to the names. Could you maybe go through config.py and add a list here of the entity_names with the correct Solvis-names?
Example:
```
ModbusFieldConfig(  # Ionisationsstrom
        name="ionisation_current",
        address=33540,
        unit="nA",
        device_class="current",
        state_class="measurement",
        poll_time=0,
    ),
```
ionisation_current - S12

fkhr79, 2025-02-09:
Did a full review of const.py and added the names as far as I know them. Used commentaries, so that the change is visible. Should possibly be reviewd by another user. 

[const_py__mod_fkhr79_20250209.txt](https://github.com/user-attachments/files/18724364/const_py__mod_fkhr79_20250209.txt)

(Sorry for the inconveniances with this kind of .txt-reply! I'm working on setting up a proper IDE, establish a fitting workflow in my environment and learn to use github. Need's a little bit of time yet.)

LarsK1, 2025-02-23:
Wil be fixed with 2.0.0-alpha.9

closed 2025-02-23 by LarsK1