LarsK1/hass_solvis_control #90: [BUG] Integration error After Update to core 1.2.0

HLueken, 2025-02-06:
**Version of the plugin**
1.2.0

 **Home-Assistant Version**
2025.2.0

**Enviroment (Docker, HAOS, ...)**
The enviroment your HomeAssistant is running in

**Version of your Solvis software/hardware**
Version NBG:
Version SC3: newes7

**Describe the bug**
After update to core 1.2.0 Solvis Integration dies not work anymore, geht the attached Integration error:
Einrichtungsfehler, wird erneut versucht: ModbusClientMixin.read_input_registers() takes 2 positional arguments but 4 were given

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

zsteinmetz, 2025-02-06:
Same here. Reconfiguring or reloading the device won't fix it.

bufu2012, 2025-02-06:
> Same here. Reconfiguring or reloading the device won't fix it.

+1

Even complete reboot did not fix it. Was still unable to configure a new device within the integration

LarsK1, 2025-02-06:
Hi,
thanks for reporting everyone. This has been fixed in https://github.com/LarsK1/hass_solvis_control/commit/b94a7da211e7668765c30c72de0bd8424462ec6b.

Will release a new version probably on the weekend.

LarsK1, 2025-02-06:
2.0.0-alpha4 has been released. This should fix, that the integration can not be setup.

Please note, that it's an already late alpha-release, nearly before beta. But still an alpha - so no guarantee for being bug free...

larsneo, 2025-02-06:
works as expected - thx for your efforts :-)

timmost, 2025-02-06:
@LarsK1 Hi Lars, 

bei mir noch nicht habe noch einen Type Error... Installation läuft nicht durch. Alpha4, SC", HKR2, Solar. Scheinbar ist das neue Release ziemlich "type-pusselig"

VG timmo

[home-assistant_solvis_control_2025-02-06T21-55-05.432Z.log](https://github.com/user-attachments/files/18698150/home-assistant_solvis_control_2025-02-06T21-55-05.432Z.log)

HLueken, 2025-02-07:
Hat bei mir auch nix verbessert. 
Muss ich mehr machen als:
- in HACS neue Version laden
- neu starten

fkhr79, 2025-02-07:
Test Version 2.0.0-alpha4 auf core-2025.2.0, HA OS 6.6.73 leider fehlgeschlagen: 

* GUI-Fehlermeldung bei der Grundeinrichtung: "Unknown error occurred"
* [home-assistant_2025-02-07T17-27-44.095Z.log](https://github.com/user-attachments/files/18710121/home-assistant_2025-02-07T17-27-44.095Z.log)

SC3 Ver. MA3.19.47, NB3.1.0

LarsK1, 2025-02-08:
Thank @fkhr79,
did test your changes and also commit (https://github.com/LarsK1/hass_solvis_control/commit/109bf4761e5319dc289178618aef0417979ceb6a) them to the repo. Will tag a new release shortly...

LarsK1, 2025-02-08:
Release has been published: https://github.com/LarsK1/hass_solvis_control/releases/tag/2.0.0-alpha5

timmost, 2025-02-08:
works for me as well!

LarsK1, 2025-02-09:
@timmost this has also been fixed in alpha6

closed 2025-02-09 by LarsK1

HLueken, 2025-02-09:
Thanks a lot, alpha6 did the job