LarsK1/hass_solvis_control #96: [FEATURE] Add "Ausgang A14" (Umschaltventil Wärmepumpe)

fkhr79, 2025-02-08:
It's another component one want's to monitor.

modbus request parameters (SC3, MA3.19.47):
- address: 33293
- data_type: int16
- input_type: input
- swap: byte
- state_class: measurement

LarsK1, 2025-02-09:
@fkhr79 thanks for opening the bug. Can you please validate the settings in const.py?

fkhr79, 2025-02-09:
Sure: done!

const.py should be

```
    ModbusFieldConfig(  # Umschaltventil Wärmepumpe A14
        name="umschaltventil_waermepumpe",
        address=33293,
        unit="%",
        state_class="measurement",
        device_class="",
        conf_option=4,
        poll_time=0,
#        register=2,
#        edit=False,
    ),
```

- "register" must be 1 (input)
- "edit" is redundant (false is standard)
- "device_class" must not be "power_factor" because in HA "power_factor" is the relation between two types of electrical power (see https://community.home-assistant.io/t/power-factor-change-suggestion-from-100-100-to-1-0-1-0/234246) and might be changed in the future. Leaving it blank shouldn't be a problem.
- value is wrong when activated: it yields 2.560 instead of 100. I think this is due to a missing byte-swap. I'm not sure, but I think "we" have to modify line 107 in coordinator.py: some values (another weird solvis "idea") need another encoding via

`decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.LITTLE)`

Since it's no default it's necessary to do that conditioned. Suggestion: another ModbusFieldConfig struct member "byte_order" with standard 0 (big endian) and option 1 (little endian).

closed 2025-02-14 by LarsK1