LarsK1/hass_solvis_control #101: [BUG] Einrichtungsfehler, wird erneut versucht: ModbusClientMixin.convert_from_registers() got an unexpected keyword argument 'word_order'

larsneo, 2025-02-10:
**Version of the plugin**
2.0.0-alpha6

 **Home-Assistant Version**
2025.2.1

**Enviroment (Docker, HAOS, ...)**
HA green

**Version of your Solvis software/hardware**
Version NBG: 3.01.00
Version SC3: 3.19.47

**Describe the bug**
Updating from alpha-5 to alpha6 breaks with "Einrichtungsfehler, wird erneut versucht: ModbusClientMixin.convert_from_registers() got an unexpected keyword argument 'word_order'."
Trying to add a new setup brings up "Ein Fehler bei der Verbindung ist aufgetreten.". 
Rollback to alpha5 fixes the problem

extract from log
2025-02-10 14:20:04.500 ERROR (MainThread) [custom_components.solvis_control.coordinator] Unexpected error fetching solvis_control data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solvis_control/coordinator.py", line 106, in _async_update_data
    data_from_register = float(self.modbus.convert_from_registers(registers=result.registers, data_type=self.modbus.DATATYPE.INT16, word_order="big"))
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.convert_from_registers() got an unexpected keyword argument 'word_order'

fkhr79, 2025-02-10:
Which HA OS Version are you running on your HA Green?
(See Settings > System > Repairs > 3-dot-menu in the upper right corner: System-Information)

larsneo, 2025-02-10:
Core 2025.2.1
Supervisor 2025.02.0
Operating System 14.2
Frontend 20250205.0

LarsK1, 2025-02-11:
Hi @larsneo ,
try rebooting the underlying HAOS. This release works perfectly fine on my system with exactly the some configuration. According to the docs, the keyword: word_order is existing and needed.

larsneo, 2025-02-15:
it's realy strange an i've no clue about it - the last working version for me is 2.0.0-alpha5. with alpha6 the integration doesn't work anymore. rebooting the underlying HAOS doesn't help (enviroment update to Core 2025.2.4 / Supervisor 2025.02.1 / Operating System 14.2 / Frontend 20250214.0). 
already removed the entire integration and tried updating from stable 1.2 in single steps - updates up to alpha5 work as expected, alpha6 still breaks the integration.

larsneo, 2025-02-15:
eventuell liegt es ja an verschiedenen parallel installierten python libraries für 'pyModbus' - z.b. bei der huawei-installation gab's ja probleme seit 2025.1. nutze zwar kein huawei, eventuell bringt aber z.b. solaredge oder goe aber ähnliche probleme mit sich.

Users that use the native modbus-integration alongside this integration should not upgrade to Home Assistant 2025.1, as it contains an upgrade to an incompatible version 3.7.4 of the pyModbus library. Significant work has already been done to upgrade this integration to support v3.8.0 of this library. We expect this to release alongside Home Assistant 2025.2, which is expected to also upgrade to that latest pyModbus version. This will solve the compatibility issue.

LarsK1, 2025-02-16:
@larsneo ja klingt fast so, als ob irgendeine andere Integration deine pyModbus-Abhängigkeit verändert.

closed 2025-02-16 by LarsK1

larsneo, 2025-02-22:
back in the game - for the records: problem was due to another integration (solaredge-modbus-multi / fixed in https://github.com/WillCodeForCats/solaredge-modbus-multi/pull/756 ). dependencies can really be a pain in the ass...

zsteinmetz, 2025-05-08:
Hey @LarsK1,

First of all, thank you so much for your awesome work on this add-on!

After updating from v2.0.0-alpha6 to v2.0.2, I can't add my device back to HASS (Core 2025.5.0, OS 15.2) due to the same error:

```
Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solvis_control.config_flow
Quelle: custom_components/solvis_control/config_flow.py:209
Integration: Solvis Control (Dokumentation, Probleme)
Erstmals aufgetreten: 17:16:59 (7 Vorkommnisse)
Zuletzt protokolliert: 17:18:49

Unexpected error in config flow: ModbusClientMixin.convert_from_registers() got an unexpected keyword argument 'word_order'
Traceback (most recent call last):
  File "/config/custom_components/solvis_control/config_flow.py", line 209, in async_step_user
    versionsc_raw, versionnbg_raw = await fetch_modbus_value([32770, 32771], 1, user_input[CONF_HOST], user_input[CONF_PORT])
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solvis_control/utils/helpers.py", line 124, in fetch_modbus_value
    value = modbussocket.convert_from_registers(
        data.registers,
        data_type=modbussocket.DATATYPE.INT16,
        word_order=order,
    )
TypeError: ModbusClientMixin.convert_from_registers() got an unexpected keyword argument 'word_order'
```

Strangely, I seem not to have any other modbus add-ons installed. My other HACS repos are Tuya Local, DWD, Adaptive Cover, and SMA Devices Plus. I suspected the latter but their code (https://github.com/littleyoda/ha-pysmaplus) doesn't use pyModbus. 

Any ideas?

fkhr79, 2025-05-08:
Hey @zsteinmetz ,

it seems that the other integrations you mentioned don't use pymodbus either. But it's a pymodbus version issue for sure.
If you didn't already do that: try a clean setup with reboot:

1. remove device and integration
2. reboot the system (not only ha)
3. add integration and device again

zsteinmetz, 2025-05-09:
Thanks for the prompt reply! I tried the steps, unfortunately without success.

One more thing: I had Solvis Modbus (https://github.com/Patrick762/hassio-solvis-modbus) installed before switching to Solvis Control. Maybe those Python dependencies didn't uninstall properly? Is there a way to check and fix this? I don't know the Python internals of HACS that much, yet.

zsteinmetz, 2025-05-09:
Alright, I found the culprit. The HACS add-on [ha-pysmaplus](https://github.com/littleyoda/ha-pysmaplus) is based on the Python package [pysma](https://github.com/littleyoda/pysma) which requires pymodbus. Version requirements in pysma read ambiguous to me. So, I manually updated pymodbus to the current version in HAOS. So far, without problems with pysma .. and I can add my Solvis device again 😍 

I'll open an issue with pysma to ask for an update of their dependencies.

fkhr79, 2025-05-09:
Happy to hear that you could solve the problem and many thanks for reporting back!
I'll update the readme to let other users know.
You don't have a guide for the manual update process of pymodbus in ha by happenstance? Could use that too...
Thx in advance and have a nice weekend! 🙂

zsteinmetz, 2025-05-11:
Sure!

1. Install and configure the Advanced SSH & Web Terminal addon in Home Assistant (see https://github.com/hassio-addons/addon-ssh for instructions).
2. Disable "Protection mode" in the addon settings.
3. Start/restart the addon
4. Open the web terminal or login via SSH.
5. Enter the HA docker container with `docker exec -it homeassistant /bin/bash`
6. Use pip as usual, for instance, search for pymodbus with `pip list | grep pymodbus` or upgrade it with `pip install pymodbus -U`

**Some words of caution:** I think you can break your system pretty easily here. Only change things if you know what you are doing or if you are installing HASSIO for testing purposes only.

zsteinmetz, 2025-05-11:
Ah, one more thing: I heard back from pysma. They suggested that it would be better practice to define compatible version numbers of Python libraries in the manifest.json. This way, inconsistencies and errors are easier to spot.

fkhr79, 2025-05-11:
> Sure!
>...

Thanks a lot! - Just updated the readme.md.

> Ah, one more thing: I heard back from pysma. They suggested that it would be better practice to define compatible version numbers of Python libraries in the manifest.json. This way, inconsistencies and errors are easier to spot.

Yes. Will remedy that shortly! - Thanks!

fkhr79, 2025-05-11:
> Yes. Will remedy that shortly! - Thanks!

Too late: @LarsK1 already fixed that! 🚀