LarsK1/hass_solvis_control #114: [BUG] A12.Brennerstatus: undocumented, wrong identifier? - unplausible unit

fkhr79, 2025-02-13:
**Version of the plugin**
2.0.0-alpha6 incl. commits of main-branch as of 13.02.2025

 **Home-Assistant Version**
2025.2.3

**Enviroment (Docker, HAOS, ...)**
HA OS 14.2

**Version of your Solvis software/hardware**
Solvis Version NBG: 3.1.0
Solvis Version SC3: 3.19.47

**Describe the bug**
A12.Brennerstatus is not documented: what might this be?
"status" does not correspond to unit "%"
value stays constantly at zero (does not mean it is wrong though)

fkhr79, 2025-02-14:
Investigated further: register seems to be 0 if the burner is off and 100 if the burner is on. There are no intermediate values. So the representation should be that of a normal binary sensor ("on"/"off"). Modulation can be found in the register of O1.

fkhr79, 2025-02-14:
Digged deeper... following I assembled all details/results, which may be helpful for solving the issue:

- SC2: Input Reg. 33291, "Ausgang A12", unit "0,50%" (see modbus spec for Solvis Remote / SC2)
- SC3: Input Reg. 33291, "Ausgang A12", Min 0, Max 100, unit "100" (see modbus spec for SC3)
- name:
  -  according to Anlagenschema SolvisMax ALS-MAX-7 Ver. 02.18 & [Anlagenschema SolvisMax 27350-2n, pg 68 f.](https://solvis-files.s3.eu-central-1.amazonaws.com/downloads-fk/solvismax7/27350_ALS-MAX-7.pdf)):
      - "Fernwärme-Ventil" only for SolvisMax Fernwärme
      - "Brenner" for SolvisMax (and more?)
  - seems to be quite harmonized on the different models and/or setups... but it's not safe... and - as said - dependent on the individual setup
- behaviour/value:
  - on SC3, Ver. 3.19.47: switches between 0 ("off") and 1 ("on")
  - on SolvisRemote / SC2: unkown -> 🟡 **Possible ToDo**: Check/Verify on SC2, but: since there are no issues at the moment, this might not be necessary and be left as it is for SC2)
- implementation of ModbusFieldConfig in const.py:
  - `device_class="power_factor"` is not correct (see https://en.wikipedia.org/wiki/Power_factor#Definition_and_calculation)
  - `input_type=0` (sensor, default) is the most appropriate type for the time being. Would better be implemented as binary sensor, which is a bit of work (create binary_sensor.py) -> 🟡 **Possible ToDo**: add binary sensor (at some point in the future...?!)
  - `unit="%"` is the most appropriate uom for the time being. Will become obsolete with implementation of binary sensor
  - `multiplier=0.00390625` results from reading the value with word_order "big". Should be read as little endian. -> 🟡 **Possible ToDo**: change to `word_order=1`and remove `multiplier=0.00390625` (applies to some other sensor configurations as well)

LarsK1, 2025-02-16:
@fkhr79 just fixed the binary_sensor...

I also added a "normal" sensor for debugging, which does show various states[image]
?

fkhr79, 2025-02-16:
This is due to the history not showing the actual (recorded) state but calculated min/max/mean-values: look at the light-blueish areas. They represent the min/max values which are (binary) 0% or 100%. Depending on the number of values and the time windows for mean calculation the mean values range from 0% to 100%. But - mind - it are mean values and not the actual state values. 🙂 
So everthing is fine!

closed 2025-02-16 by LarsK1