LarsK1/hass_solvis_control #117: [BUG] Switch "Warmwasser Nachheizung" behaves as button

fkhr79, 2025-02-14:
**Version of the plugin**
2.0.0-alpha6 incl. commits of main-branch as of 13.02.2025

 **Home-Assistant Version**
2025.2.3

**Enviroment (Docker, HAOS, ...)**
HA OS 14.2

**Version of your Solvis software/hardware**
Solvis Version NBG: 3.1.0
Solvis Version SC3: 3.19.47

**Describe the bug**
Turning the switch "Warmwasser Nachheizung" on correctly starts the heating of hot water but does not stay in the on position during the heating process. Switches back to off immediately. 
Would better be presented as button?!

fkhr79, 2025-02-14:
investigated further via modbus-integration: value of the register stays 1 ("on") - only the switch doesn't represent this

LarsK1, 2025-02-16:
Fixed with 2.0.0-alpha.7

closed 2025-02-16 by LarsK1