LarsK1/hass_solvis_control #125: Error - Integration does not work

Bimmerfriend, 2025-02-15:
Hi,

Unfortunately my Solvis integration (1.2.0) does not work anymore. It o ly gives an error and does not start:
"Einrichtungsfehler, wird erneut versucht: ModbusClientMixin.read_input_registers() takes 2 positional arguments but 4 were given"

Does anybody have a clue what is wrong?

IP did not change, Modus is active in SC3...

Hope anybody can help, pointing me in the right direction...

Best regards!

fkhr79, 2025-02-15:
Hey!
Try updating to version 2.0.0-alpha6. It's an alpha-version... but will solve the problem mentioned.
Otherwise - if you don't want to install this alpha - you'll need to wait for the next release. Should not take long...

duplicate of #90

fkhr79, 2025-02-15:
no need: go to hacs and select another version (release):

[image]

LarsK1, 2025-02-16:
Fixed by 1.2.1
Duplicate of #90

closed 2025-02-16 by LarsK1