LarsK1/hass_solvis_control #138: [CHANGE] [FEATURE] A1 to A14: change sensortypes | A6, A7, A10, A11, A13: add sensors

fkhr79, 2025-02-17:
Hey!
I analyzed all docs I could get my (analog and digital) hands on and combined all the info into the attached table (based on the supported_entities.md). I'd suggest the following changes/additions:
- Change all AX's to binary sensor since they are only "on" (100 / 200) or "off" (0).
- Add A6, A7, A10, A11, A13
- Set the names of the senors to the suggested **defaults** (the standardized ones respectively - see #89) 
- Optional (since it requires structural code additions / new logic): Add the possible altenative namings/meanings as entity attributes 
- VERY (!) optional (since the entities can easily be renamed and ... and...): make the entity names configurable during config (to be honest: I would't invest so much effort in this feature)

Let me know what you think. I'd take over the non-optional parts for a beginning.

| Adresse <br> decimal | Solvis <br> Beschreibung | Funktion (bold: default-suggestion)                                                                                                     | Unterstützt <br> seit Ver. | Sensortyp | Conf.-Option | Bemerkung | Min | Max | Einheit |
| -------------------- | ------------------------ | ------------------------------------------------------------------------------------------------------------------------------------- | -------------------------- | --------- | ------------ | --------- | --- | --- | ------- |
| 33280                | Ausgang A1               | [alle] **Zirkulationspumpe**                                                                                                          | 0.1                        | 0         | 0            |           | 0   | 100 | 100     |
| 33281                | Ausgang A2               | [WP] **Ladepumpe PLAS-WP** <br> [Lino/Solo/Pia] unbenutzt                                                                             | 0.1                        | 0         | 0            |           | 0   | 200 | 200     |
| 33282                | Ausgang A3               | [alle] **Heizkreis 1 - Pumpe**                                                                                                        | 1.0                        | 0         | 0            |           | 0   | 100 | 100     |
| 33283                | Ausgang A4               | [alle] **Heizkreis 2 - Pumpe**                                                                                                        | 1.0                        | 0         | 1            |           | 0   | 100 | 100     |
| 33284                | Ausgang A5               | [alle] **Heizkreis 3 - Pumpe**                                                                                                        | 1.0                        | 0         | 2            |           | 0   | 100 | 100     |
| 33285                | Ausgang A6               | [HK3] **Heizkreis 3 - Mischer auf** <br> [Ost/West-Dach] Ventil 1 <br> [FBK] unbenutzt                                                |                            |           |              |           | 0   | 100 | 100     |
| 33286                | Ausgang A7               | [HK3] **Heizkreis 3 - Mischer zu** <br> [Ost/West-Dach] Ventil 2 <br> [FBK] Ladepumpe                                                 |                            |           |              |           | 0   | 100 | 100     |
| 33287                | Ausgang A8               | [alle] **Heizkreis 1 - Mischer auf**                                                                                                  | 2.0.0                      | 0         | 0            |           | 0   | 100 | 100     |
| 33288                | Ausgang A9               | [alle] **Heizkreis 1 - Mischer zu**                                                                                                   | 2.0.0                      | 0         | 0            |           | 0   | 100 | 100     |
| 33289                | Ausgang A10              | [alle] **Heizkreis 2 - Mischer auf**                                                                                                  |                            |           |              |           | 0   | 100 | 100     |
| 33290                | Ausgang A11              | [alle] **Heizkreis 2 - Mischer zu**                                                                                                   |                            |           |              |           | 0   | 100 | 100     |
| 33291                | Ausgang A12              | [WP/Pia] **Steuerspannung SolvisLea (L/N/PE) Heizstab Stufe 1 und 3 (A12->DHC1)** <br> [Lino/Solo] Brenner <br> [FW] Fernwärme-Ventil | 0.1                        | 0         | 0            |           | 0   | 100 | 100     |
| 33292                | Ausgang A13              | [WP/Pia] **Heizstab Stufe 2 und 3 (->DHC2)** <br> [Lino/Solo] Ladepumpe PLAS Lino / Fremdkessel <br> [Öl] Brenner 2                   |                            |           |              |           | 0   | 100 | 100     |
| 33293                | Ausgang A14              | [WP/Pia] **3-Wege-Umschaltventil** <br> [Lino/Solo] Brenner (potentialfrei) <br> [FW] Fernwärme-Ventil <br> [Öl] Brenner              | 2.0.0                      | 0         | 4            |           | 0   | 100 | 100     |

LarsK1, 2025-02-17:
Hi @fkhr79 ,
thanks for your precious analysis and time ♥️. I finally managed to rename all entities in config.py to a fully-functional naming scheme. Before making any further modification, please take a look at the made changes and create a PR for them. Afterwards I will adjust the icons and translations - which will currently definitely make all tests fail...

LarsK1, 2025-02-17:
## Regarding the rest:

>  Change all AX's to binary sensor since they are only "on" (100 / 200) or "off" (0).

✅, please do so.

> Add A6, A7, A10, A11, A13

✅, please do so as well

> Set the names of the senors to the suggested defaults (the standardized ones respectively - see https://github.com/LarsK1/hass_solvis_control/issues/89)

Please check, the recent modifications first - afterwards we might have to discuss again (a little bit)

> Optional (since it requires structural code additions / new logic): Add the possible altenative namings/meanings as entity attributes
> VERY (!) optional (since the entities can easily be renamed and [...]

Please leave that for a moment out, I'm currently investigating in a logic, which (may) allow a user to fully define his system config, and then show the correct entities and names

fkhr79, 2025-02-17:
> I finally managed to rename all entities in config.py to a fully-functional naming scheme.

Jey! - I was looking forward to that! 👍 

> Before making any further modification, please take a look at the made changes and create a PR for them.

Never did before... but just tried. I hope I did it the right way? Please let me know if I did something wrong or could do better!

> Afterwards I will adjust the icons and translations - which will currently definitely make all tests fail...

A lot of tedious work: I'm not envious! 😉 

> Please check, the recent modifications first - afterwards we might have to discuss again (a little bit)

Check!

> Please leave that for a moment out, I'm currently investigating in a logic, which (may) allow a user to fully define his system config, and then show the correct entities and names

THAT would be cooool! I bow my head in humility! 🙇

fkhr79, 2025-02-18:
Ahh... sorry! I think I messed up: you did not meant it that way. After a good nights sleep I realized that I could have merged the changes into my repo, review/change and THEN create a PR. - Will do now!

fkhr79, 2025-02-19:
Just added another commit to PR#139 which realizes the following processing status of this issue:

| Adresse <br> decimal | Solvis <br> Beschreibung | Funktion                                                                                                              | Unterstützt <br> seit Ver. | Sensortyp | Conf.-<br>Option | Name                             | ToDo                        |
| -------------------- | ------------------------ | --------------------------------------------------------------------------------------------------------------------- | -------------------------- | --------- | ---------------- | -------------------------------- | --------------------------- |
| 33280                | Ausgang A1               | [alle] Zirkulationspumpe                                                                                              | 0.1                        | 4         | 0                | Zirkulationspumpe (A1)           | ✅                          |
| 33281                | Ausgang A2               | [WP] Ladepumpe PLAS-WP <br> [Lino/Solo/Pia] unbenutzt                                                                 | 0.1                        | 4         | 4                | WP Ladepumpe (A2)                | ⚠️verify / check value        |
| 33282                | Ausgang A3               | [alle] Heizkreis 1 - Pumpe                                                                                            | 1.0                        | 4         | 0                | HKR1 Pumpe (A3)                  | ✅                          |
| 33283                | Ausgang A4               | [alle] Heizkreis 2 - Pumpe                                                                                            | 1.0                        | 4         | 1                | HKR2 Pumpe (A4)                  | ✅                          |
| 33284                | Ausgang A5               | [alle] Heizkreis 3 - Pumpe                                                                                            | 1.0                        | 4         | 2                | HKR3 Pumpe (A5)                  | ✅                          |
| 33285                | Ausgang A6               | [HK3] Heizkreis 3 - Mischer auf <br> [Ost/West-Dach] Ventil 1 <br> [FBK] unbenutzt                                    | new: 2.0.1 ?               | 4         | 2                | HKR3 Mischer Heizkreis auf (A6)  | ⚠️verify                      |
| 33286                | Ausgang A7               | [HK3] Heizkreis 3 - Mischer zu <br> [Ost/West-Dach] Ventil 2 <br> [FBK] Ladepumpe                                     | new: 2.0.1 ?               | 4         | 2                | HKR3 Mischer Heizkreis zu (A7)   | ⚠️verify                      |
| 33287                | Ausgang A8               | [alle] Heizkreis 1 - Mischer auf                                                                                      | 2.0.0                      | 4         | 0                | HKR1 Mischer Heizkreis auf (A8)  | ✅                          |
| 33288                | Ausgang A9               | [alle] Heizkreis 1 - Mischer zu                                                                                       | 2.0.0                      | 4         | 0                | HKR1 Mischer Heizkreis zu (A9)   | ✅                          |
| 33289                | Ausgang A10              | [alle] Heizkreis 2 - Mischer auf                                                                                      | new: 2.0.1 ?               | 4         | 1                | HKR2 Mischer Heizkreis auf (A10) | ⚠️verify                      |
| 33290                | Ausgang A11              | [alle] Heizkreis 2 - Mischer zu                                                                                       | new: 2.0.1 ?               | 4         | 1                | HKR2 Mischer Heizkreis zu (A11)  | ⚠️verify                      |
| 33291                | Ausgang A12              | [WP/Pia] Steuerspannung SolvisLea (L/N/PE) Heizstab Stufe 1 und 3 <br> [Lino/Solo] Brenner <br> [FW] Fernwärme-Ventil | 0.1                        | 4         | 0                | Brennerstatus (A12)              | ⚠️verify / test / change name |
| 33292                | Ausgang A13              | [WP/Pia] Heizstab Stufe 2 und 3 <br> [Lino/Solo] Ladepumpe PLAS Lino / Fremdkessel <br> [Öl] Brenner 2                | new: 2.0.1 ?               | 4         | 4                | WP Heizstab Stufe 2 & 3 (A13)    | ⚠️verify                      |
| 33293                | Ausgang A14              | [WP/Pia] 3-Wege-Umschaltventil <br> [Lino/Solo] Brenner (potentialfrei) <br> [FW] Fernwärme-Ventil <br> [Öl] Brenner  | 2.0.0                      | 4         | 4                | Umschaltventil Wärmepumpe (A14)  | ✅                          |

If someone wants to support: unfortunately (due to missing components) I cannot check all of the entries marked with⚠️. So if you could spare some time please check with your setup and sc3/portal/app/whatever. Every hint/info/suggestion is welcome!

closed 2025-03-04 by LarsK1