LarsK1/hass_solvis_control #158: [FEATURE] config-flow-test-coverage / test-coverage

LarsK1, 2025-02-23:
See #78

fkhr79, 2025-03-09:
```
---------- coverage: platform linux, python 3.13.2-final-0 -----------
Name                                                Stmts   Miss  Cover   Missing
---------------------------------------------------------------------------------
custom_components/solvis_control/__init__.py           90     43    52%   53, 59, 94-101, 116-127, 129-134, 136-138, 140-143, 145-152, 154-157
custom_components/solvis_control/binary_sensor.py     137     60    56%   36-37, 49-50, 63-64, 71-72, 117-121, 164, 173-238
custom_components/solvis_control/config_flow.py       177     87    51%   61, 63, 65, 108, 123, 135, 166-168, 182-194, 214-216, 229-231, 233, 235, 242-250, 260, 269-271, 275-326, 334-344, 352-359
custom_components/solvis_control/const.py              49      0   100%
custom_components/solvis_control/coordinator.py       106     76    28%   70-154
custom_components/solvis_control/diagnostics.py        28     28     0%   7-45
custom_components/solvis_control/number.py            138     58    58%   36-37, 59-60, 62-63, 96-100, 159-210, 214-220
custom_components/solvis_control/select.py            139     10    93%   50-51, 64-65, 72-73, 175-176, 178-179
custom_components/solvis_control/sensor.py            149     73    51%   38-39, 64-65, 95-101, 153-243
custom_components/solvis_control/switch.py            135     65    52%   36-37, 49-50, 59-60, 62-63, 90-94, 135-180, 184-192, 196-204
custom_components/solvis_control/utils/helpers.py      72     28    61%   40-41, 72-73, 79-80, 87, 90-91, 101-109, 116-119, 145-154
---------------------------------------------------------------------------------
TOTAL                                                1220    528    57%
```

fkhr79, 2025-03-09:
```
---------- coverage: platform linux, python 3.13.2-final-0 -----------
Name                                                Stmts   Miss  Cover   Missing
---------------------------------------------------------------------------------
custom_components/solvis_control/__init__.py           90     43    52%   53, 59, 94-101, 116-127, 129-134, 136-138, 140-143, 145-152, 154-157
custom_components/solvis_control/binary_sensor.py     137     60    56%   36-37, 49-50, 63-64, 71-72, 117-121, 164, 173-238
custom_components/solvis_control/config_flow.py       177     46    74%   63, 65, 108, 182-194, 229-231, 233, 235, 249-250, 291-302, 316-319, 337-343, 352-359
custom_components/solvis_control/const.py              49      0   100%
custom_components/solvis_control/coordinator.py       110     80    27%   70-169
custom_components/solvis_control/diagnostics.py        28     28     0%   7-45
custom_components/solvis_control/number.py            134     54    60%   36-37, 59-60, 62-63, 96-100, 180-221, 225-238
custom_components/solvis_control/select.py            139     10    93%   50-51, 64-65, 72-73, 175-176, 178-179
custom_components/solvis_control/sensor.py            149     73    51%   38-39, 64-65, 95-101, 153-243
custom_components/solvis_control/switch.py            135     65    52%   36-37, 49-50, 59-60, 62-63, 90-94, 135-180, 184-192, 196-204
custom_components/solvis_control/utils/helpers.py      72     25    65%   40-41, 72-73, 79-80, 87, 90-91, 104-109, 116-119, 145-154
---------------------------------------------------------------------------------
TOTAL                                                1220    484    60%
```

fkhr79, 2025-03-16:
```
---------- coverage: platform linux, python 3.13.2-final-0 -----------
Name                                                Stmts   Miss  Cover   Missing
---------------------------------------------------------------------------------
custom_components/solvis_control/__init__.py           90     39    57%   53, 59, 116-127, 129-134, 136-138, 140-143, 145-152, 154-157
custom_components/solvis_control/binary_sensor.py     137     55    60%   36-37, 49-50, 63-64, 71-72, 117-121, 164, 177-178, 183-238
custom_components/solvis_control/config_flow.py       177     12    93%   251-254, 268-269, 317-320, 378-379
custom_components/solvis_control/const.py              49      0   100%
custom_components/solvis_control/coordinator.py       114     34    70%   43-66, 79-80, 114-116, 142-143, 152-153, 155-156, 164-166, 169-170
custom_components/solvis_control/diagnostics.py        28     28     0%   7-45
custom_components/solvis_control/number.py            134     49    63%   36-37, 59-60, 62-63, 96-100, 181-182, 185-188, 194-221, 225-238
custom_components/solvis_control/select.py            139      8    94%   50-51, 64-65, 72-73, 178-179
custom_components/solvis_control/sensor.py            149     49    67%   38-39, 64-65, 97-101, 164-165, 168-171, 176-179, 183-186, 189-192, 197-228, 230-234, 236-240
custom_components/solvis_control/switch.py            135     58    57%   36-37, 49-50, 59-60, 62-63, 90-94, 145-180, 184-192, 196-204
custom_components/solvis_control/utils/helpers.py      70     21    70%   40-41, 72-73, 79-80, 87, 90-91, 105-106, 113-116, 143-152
---------------------------------------------------------------------------------
TOTAL                                                1222    353    71%

```

fkhr79, 2025-03-22:
```
---------- coverage: platform linux, python 3.13.2-final-0 -----------
Name                                                Stmts   Miss Branch BrPart  Cover   Missing
-----------------------------------------------------------------------------------------------
custom_components/solvis_control/__init__.py           90     39     42      9    45%   53, 59, 88->90, 118-129, 131-136, 138-140, 142-145, 147-154, 156-159
custom_components/solvis_control/binary_sensor.py      82      0     16      0   100%
custom_components/solvis_control/config_flow.py       177     12     28      0    94%   251-254, 268-269, 317-320, 378-379
custom_components/solvis_control/const.py              49      0      0      0   100%
custom_components/solvis_control/coordinator.py       120     36     42     10    72%   45-68, 78, 88->106, 101->106, 112-114, 121->132, 129->132, 138-139, 150-151, 154-155, 163-165, 168-169, 174-177
custom_components/solvis_control/diagnostics.py        28     28      6      0     0%   7-45
custom_components/solvis_control/number.py             86      0     18      0   100%
custom_components/solvis_control/select.py             83      0     14      0   100%
custom_components/solvis_control/sensor.py            103      0     32      0   100%
custom_components/solvis_control/switch.py             83      0     16      0   100%
custom_components/solvis_control/utils/helpers.py     165     37     50     10    77%   45-46, 77-78, 84-85, 92, 95-96, 110-111, 118-121, 148-157, 203-204, 212-217, 223-224, 254-255, 277->287, 283->287, 291-292, 299-300
-----------------------------------------------------------------------------------------------
TOTAL                                                1066    152    264     29    84%
```

fkhr79, 2025-03-23:
```
 ---------- coverage: platform linux, python 3.13.2-final-0 -----------
Name                                                Stmts   Miss Branch BrPart  Cover   Missing
-----------------------------------------------------------------------------------------------
custom_components/solvis_control/__init__.py           90     39     42      9    45%   53, 59, 88->90, 118-129, 131-136, 138-140, 142-145, 147-154, 156-159
custom_components/solvis_control/binary_sensor.py      82      0     16      0   100%
custom_components/solvis_control/config_flow.py       172      4     28      0    98%   312-315
custom_components/solvis_control/const.py              49      0      0      0   100%
custom_components/solvis_control/coordinator.py       120     36     42     10    72%   45-68, 78, 88->106, 101->106, 112-114, 121->132, 129->132, 138-139, 150-151, 154-155, 163-165, 168-169, 174-177
custom_components/solvis_control/diagnostics.py        28     28      6      0     0%   7-45
custom_components/solvis_control/number.py             86      0     18      0   100%
custom_components/solvis_control/select.py             83      0     14      0   100%
custom_components/solvis_control/sensor.py            103      0     32      0   100%
custom_components/solvis_control/switch.py             83      0     16      0   100%
custom_components/solvis_control/utils/helpers.py     165     37     50     10    77%   45-46, 77-78, 84-85, 92, 95-96, 110-111, 118-121, 148-157, 203-204, 212-217, 223-224, 254-255, 277->287, 283->287, 291-292, 299-300
-----------------------------------------------------------------------------------------------
TOTAL                                                1061    144    264     29    85%
```

Nearly done - but need to change the exception handling in config_flow and fetch_modbus_value... <sigh>

fkhr79, 2025-03-23:
```
---------- coverage: platform linux, python 3.13.2-final-0 -----------
Name                                                Stmts   Miss Branch BrPart  Cover   Missing
-----------------------------------------------------------------------------------------------
custom_components/solvis_control/__init__.py           90     39     42      9    45%   53, 59, 88->90, 118-129, 131-136, 138-140, 142-145, 147-154, 156-159
custom_components/solvis_control/binary_sensor.py      82      0     16      0   100%
custom_components/solvis_control/config_flow.py       179      0     22      0   100%
custom_components/solvis_control/const.py              49      0      0      0   100%
custom_components/solvis_control/coordinator.py       120     35     42     10    72%   45-68, 78, 88->106, 101->106, 112-114, 121->132, 129->132, 138-139, 150-151, 154-155, 163-165, 169, 174-177
custom_components/solvis_control/diagnostics.py        28     28      6      0     0%   7-45
custom_components/solvis_control/number.py             86      0     18      0   100%
custom_components/solvis_control/select.py             83      0     14      0   100%
custom_components/solvis_control/sensor.py            103      0     32      0   100%
custom_components/solvis_control/switch.py             83      0     16      0   100%
custom_components/solvis_control/utils/helpers.py     155     32     50     10    79%   45-46, 77, 83, 90, 93, 109-112, 138-147, 193-194, 202-207, 213-214, 244-245, 267->277, 273->277, 281-282, 289-290
-----------------------------------------------------------------------------------------------
TOTAL                                                1058    134    258     29    85%
```

Finally! - Config flow is at 100%. Jey! 🙂 
Base structure for (further) testing is prepared and stable.

ToDo: Increase the overall test coverage to 95%.

fkhr79, 2025-03-23:
```
---------- coverage: platform linux, python 3.13.2-final-0 -----------
Name                                                Stmts   Miss Branch BrPart  Cover   Missing
-----------------------------------------------------------------------------------------------
custom_components/solvis_control/__init__.py           90      0     42      0   100%
custom_components/solvis_control/binary_sensor.py      82      0     16      0   100%
custom_components/solvis_control/config_flow.py       179      0     22      0   100%
custom_components/solvis_control/const.py              49      0      0      0   100%
custom_components/solvis_control/coordinator.py       120     17     42     10    83%   78, 88->106, 101->106, 112-114, 121->132, 129->132, 138-139, 150-151, 154-155, 163-165, 169, 174-177
custom_components/solvis_control/diagnostics.py        28     28      6      0     0%   7-45
custom_components/solvis_control/number.py             86      0     18      0   100%
custom_components/solvis_control/select.py             83      0     14      0   100%
custom_components/solvis_control/sensor.py            103      0     32      0   100%
custom_components/solvis_control/switch.py             83      0     16      0   100%
custom_components/solvis_control/utils/helpers.py     155     32     50     10    79%   45-46, 77, 83, 90, 93, 109-112, 138-147, 193-194, 202-207, 213-214, 244-245, 267->277, 273->277, 281-282, 289-290
-----------------------------------------------------------------------------------------------
TOTAL                                                1058     77    258     20    92%
```

fkhr79, 2025-03-24:
```
---------- coverage: platform linux, python 3.13.2-final-0 -----------
Name                                                Stmts   Miss Branch BrPart  Cover   Missing
-----------------------------------------------------------------------------------------------
custom_components/solvis_control/__init__.py           90      0     42      0   100%
custom_components/solvis_control/binary_sensor.py      82      0     16      0   100%
custom_components/solvis_control/config_flow.py       179      0     22      0   100%
custom_components/solvis_control/const.py              49      0      0      0   100%
custom_components/solvis_control/coordinator.py       120     17     42     10    83%   78, 88->106, 101->106, 112-114, 121->132, 129->132, 138-139, 150-151, 154-155, 163-165, 169, 174-177
custom_components/solvis_control/diagnostics.py        28      0      6      0   100%
custom_components/solvis_control/number.py             86      0     18      0   100%
custom_components/solvis_control/select.py             83      0     14      0   100%
custom_components/solvis_control/sensor.py            103      0     32      0   100%
custom_components/solvis_control/switch.py             83      0     16      0   100%
custom_components/solvis_control/utils/helpers.py     155     32     50     10    79%   45-46, 77, 83, 90, 93, 109-112, 138-147, 193-194, 202-207, 213-214, 244-245, 267->277, 273->277, 281-282, 289-290
-----------------------------------------------------------------------------------------------
TOTAL                                                1058     49    258     20    95%
```

fkhr79, 2025-03-26:
```
---------- coverage: platform linux, python 3.13.2-final-0 -----------
Name                                                Stmts   Miss Branch BrPart  Cover   Missing
-----------------------------------------------------------------------------------------------
custom_components/solvis_control/__init__.py           90      0     42      0   100%
custom_components/solvis_control/binary_sensor.py      82      0     16      0   100%
custom_components/solvis_control/config_flow.py       179      0     22      0   100%
custom_components/solvis_control/const.py              49      0      0      0   100%
custom_components/solvis_control/coordinator.py        97      0     24      0   100%
custom_components/solvis_control/diagnostics.py        28      0      6      0   100%
custom_components/solvis_control/number.py             86      0     18      0   100%
custom_components/solvis_control/select.py             83      0     14      0   100%
custom_components/solvis_control/sensor.py            103      0     32      0   100%
custom_components/solvis_control/switch.py             83      0     16      0   100%
custom_components/solvis_control/utils/helpers.py     161      0     52      0   100%
-----------------------------------------------------------------------------------------------
TOTAL                                                1041      0    242      0   100%
```

fkhr79, 2025-03-26:
Could yet be refactored a little bit... but... for now: DONE!!! 💯

closed 2025-03-26 by fkhr79