LarsK1/hass_solvis_control #172: [FEATURE] Extend ability to use own temperature sensors for HKR2 and HKR3

fkhr79, 2025-02-28:
The new feature allowing the use of custom temperature sensors for HKR1 (see #10) should be extended to HKR2 and HKR3.

Additionally, the configuration dialog should be enhanced to include a separate dialog for configuring this behavior.

sneekerzzz, 2025-03-01:
I’ve got something interesting from solvis maybe this is something you can use [

[Modbus_SmartEnergy.xlsx](https://github.com/user-attachments/files/19037923/Modbus_SmartEnergy.xlsx)

]

fkhr79, 2025-03-01:
@sneekerzzz : Cool! Thanks!!!
Moved it to a new issue for further discussion / examination: #173 
Will further comment over there.

LarsK1, 2025-03-02:
@fkhr79 see https://github.com/LarsK1/hass_solvis_control/commit/a3974e08f89a895904838d6bdd09d93203e30681

What did you think of, in regards to configuration?

fkhr79, 2025-03-02:
@LarsK1 
Don't know if you get notified: left two comments at https://github.com/LarsK1/hass_solvis_control/commit/a3974e08f89a895904838d6bdd09d93203e30681

I think we should add another config form for config of the hkrx_room_temp configuration: let's say up to three select / dropdown boxes. one for each active/selected hkr (so it depends on the selected hkr's). options in every box are: read, write.

sneekerzzz, 2025-03-03:
I tested the temperature control, and it can be adjusted from the Devices page using the slider, as well as through an automation, as shown in the example.

However, when clicking on the entity and opening the More Info dialog, the slider cannot be moved to the desired temperature.

In the version I tested, only HKR1 was available, while HKR2 was missing, so I was unable to test it.

LarsK1, 2025-03-03:
@fkhr79 yes, I'll be notified. I'll pushed a fix for this just now.

LarsK1, 2025-03-04:
Hi @sneekerzzz ,
best place for this would have been a discussion, but no worries. The reason for this, is that the SolarEdge Modbus integration is running a much older version oy pymodbus. You may try updating to https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V2.0.3, which as far as I know, uses the latest pymodbus version. Please mind, that you do have to reboot the complete system for this to take not place, not only HA Core

sneekerzzz, 2025-03-05:
> Hi [@sneekerzzz](https://github.com/sneekerzzz) , best place for this would have been a discussion, but no worries. The reason for this, is that the SolarEdge Modbus integration is running a much [...]

Hi @LarsK1,

Thanks, I appreciate it! Unfortunately, I’m already using the latest version: [V2.0.3](https://github.com/binsentsu/home-assistant-solaredge-modbus/releases/tag/V2.0.3).

For now, I’m running Solvis Control on a secondary instance, connecting to my production server. This works for the time being.

fkhr79, 2025-03-05:
Hey @sneekerzzz : we had the same issue a few weeks ago. See https://github.com/LarsK1/hass_solvis_control/issues/101#issuecomment-2676087572 - Perhaps the solution works for you too!? Good Luck! 🙂

sneekerzzz, 2025-03-09:
he writing of the temperature values to the SC3 works well, but the written temperature is not visible in the entity. I have added an image.

[https://imgur.com/a/psyF3Br](url)

fkhr79, 2025-03-09:
[home-assistant_solvis_control_2025-03-09T15-20-11.885Z.log](https://github.com/user-attachments/files/19150430/home-assistant_solvis_control_2025-03-09T15-20-11.885Z.log)
Here's my debug log.
Even if I put hkr1_fix_day_temp and hkr1_room_temp in the same polling group, hkr1_room_temp gets never updated?! Very strange!

fkhr79, 2025-03-09:
Finally found the main bug! - There's another smaller bug to solve... but I think this'll not take long... 🙂

fkhr79, 2025-03-09:
Done.
Thanks, @sneekerzzz for testing! This discovered some other things as well so that they could be fixed to not popup any time later.
Even if I'm pretty sure to have solved the issue I cannot test the feature due to the missing sensors (the modbus registers in my setup are ... let's say stubborn!). So if you'd test again, please? Thanks in advance!

Just a short hint: the temperature sensors for HKRx are in the default polling group. Therefore their values are updated every default polling interval. If you set a value to write via modbus it may take up to the default polling intervals timespan to update the value in the ui. Just saying... 😉

sneekerzzz, 2025-03-10:
> Done. Thanks, [@sneekerzzz](https://github.com/sneekerzzz) for testing! This discovered some other things as well so that they could be fixed to not popup any time later. Even if I'm pretty sure to [...]

Running since yesterday and it looks and works good 

Many thanks to both of you guys!

fkhr79, 2025-03-10:
You're welcome!
Do you get this message frequently or just once? Is it reproducable? If so: would you mind providing a debug log? Thx in advance!

sneekerzzz, 2025-03-10:
> You're welcome! Do you get this message frequently or just once? Is it reproducable? If so: would you mind providing a debug log? Thx in advance!

Ive included the log 

[home-assistant_2025-03-10T20-46-56.270Z.log.zip](https://github.com/user-attachments/files/19170648/home-assistant_2025-03-10T20-46-56.270Z.log.zip)

fkhr79, 2025-03-18:
Hey @sneekerzzz !
Thx for the log. Don't know if I could solve the remaining issue since I cannot test it myself with those registers. Did some consistency improvements and some things more. The modbus connection should be more stable now. Perhaps You'd like to test the current implementation again? Looking forward to any info (or log) I can get! 😉

sneekerzzz, 2025-03-20:
> Hey [@sneekerzzz](https://github.com/sneekerzzz) ! Thx for the log. Don't know if I could solve the remaining issue since I cannot test it myself with those registers. Did some consistency [...]

I’ve been running this update since its release, and it has been running smoothly.

The error [hkr1_room_temp_write] Couldn't connect to Modbus device: Modbus Error: [Connection] Not connected [AsyncModbusTcpClient] appeared only once since then.

Another issue is that I’ve translated everything into Dutch, but most entities are still in German or English.

fkhr79, 2025-03-20:
Thanks for your feedback! Happy to hear that it's running smoothly!

Just added a new issue for the missing translation to Dutch: see #233 - Let's handle

Since this feat is implemented and there seem to be no problems left, I'll close this issue now.
If you encounter any new failures You're welcome to open a new issue. Thanks in advance!