LarsK1/hass_solvis_control #173: TODO: Evaluate / Check registers in Excel-Table

fkhr79, 2025-03-01:
[Modbus_SmartEnergy.xlsx](https://github.com/user-attachments/files/19038308/Modbus_SmartEnergy.xlsx)

> I’ve got something interesting from Solvis maybe this is something you can use 
> 
> Attached, you will find a table with all SC3 registers available via Modbus/TCP. There are still a few errors, which will be corrected in the next update.
> 
> Additionally, this table already includes groups 131 and 135, which are not yet part of the current 3.20.16 version. However, version 3.20.16 is now running stably with external access, for example, from a Raspberry Pi, Home Assistant, and the corresponding SC3 integration (various HACS available).
> 
> The 3.21.xx software is expected in March, incorporating all changes—unless new ones are added again.
> 
> [Modbus_SmartEnergy.xlsx](https://github.com/user-attachments/files/19037923/Modbus_SmartEnergy.xlsx)
> 
>  

 _Originally posted by @sneekerzzz in [#172](https://github.com/LarsK1/hass_solvis_control/issues/172#issuecomment-2692067721)_

sneekerzzz, 2025-03-01:
No, this is a document I received without any modifications from my side. I also received a UDP file, Sc3_MA_3.20.16A. I will share it as well when I’m home and at my laptop.

fkhr79, 2025-03-01:
Great! Thanks!! 👍

fkhr79, 2025-03-02:
Just had a first deeper look inside: quite funny. I think I now what happend to them. They got a little lost in registers and mixed up reading and holding registers. This is why some registers which should yield known values seem to be flawed: they were just accidentally overwritten by other values. Will check later.
And: at least a few registers of group 131 are actually available already (and supported by this integration of course! 😉 ).

fkhr79, 2025-03-02:
> Thanks for sharing, looks really interesting. Do you have any ability to adress wishes/suggestions? It would be very helpful (at least for me) if it would be possible to write the modus for Zirkulation . This would it make possible to easily switch warmwater Zirkulation off e.g. during vacation time.

Your wish is my command! 😉
Unfortunately there is no holding register for warmwater circulation: only reading is possible (at least at the moment and as far as it is documented).

fkhr79, 2025-03-02:
input registers to change/add:
- [ ] (33546) "aktuelle Leistung Wärmeerzeuger 2  (thermisch)"
- [ ] (33547) "aktuelle Leistung Wärmeerzeuger 2  (elektrisch)"
- [ ] (33549) "aktuelle Leistung Warmwasser (thermisch)"
- [ ] (33552) "Laufzeit Solarpumpe 1"
- [ ] (33553) "Laufzeit Solarpumpe 2"
- [ ] (33792) "Meldungen Anzahl"

holding registers to change/add:
- [ ] (838) "Bivalenztemp. für WW"
- [ ] (839) "Bivalenztemp. für Heizung"

fkhr79, 2025-03-02:
until further needs: done.

closed 2025-03-02 by fkhr79