LarsK1/hass_solvis_control #184: [INFO] entity name changed from solvis_heizung_ww_solltemperatur to solvis_heizung_warmwasser_solltemperatur

larsneo, 2025-03-02:
**Version of the plugin**
2.0.0-alpha.10

 **Home-Assistant Version**
 Core 2025.2.5
Supervisor 2025.02.4
Operating System 14.2
Frontend 20250221.0

**Enviroment (Docker, HAOS, ...)**
HA green

**Version of your Solvis software/hardware**
Firmware: 3.19.47
Hardware: 3.02.01 

**Describe the bug**
the entity "solvis_heizung_ww_solltemperatur" changed in alpha10 to "solvis_heizung_warmwasser_solltemperatur"

**To Reproduce**
update to alpha10

**Additional context**
just as an information, one needs to keep in mind that some automations might rely on entities ;-)

fkhr79, 2025-03-02:
It's not a bug. It's a feature! 😉 
Yes, we know that there might be some inconveniances to migrate to the new versions. But please keep in mind, that this is still an alpha version still. There might (possibly will) be further changes to the entity id's due to a complete consolidation and a new naming convention. The benefit will be, that the many many many entities are correctly and consistently named and fit to the Solvis naming.
Sorry for that... but it's for a better future. 😉 After this consolidation there shouldn't be any changes on the id's any more...

closed 2025-03-02 by fkhr79

larsneo, 2025-03-02:
i'm perfectly fine with this idea - especially since it brings a bright future :-) it was just meant as an reminder that ppl upgrading the plugin needs to keep in mind those changes. 
anyway, it might be a useful addition to add some unique_id to the entities also. i've to re-update automations with every version...

LarsK1, 2025-03-03:
@larsneo there are unique_id's in place. The reason here, is that the unique_id's have also been changed in order to complete a standardised naming scheme.

larsneo, 2025-03-03:
@LarsK1 thx, makes sense. but i'm still unsure about a specific problem - to be honest: i've no idea if a unique_id _could_ solve the problem that the following automation breaks with each update
[image]

fkhr79, 2025-03-03:
@larsneo : could you provide the automation details (the yaml-code), please?

larsneo, 2025-03-03:
`alias: Warmwasser Nachheizung ein
description: ""
triggers:
  - device_id: 24b487c09d939ff40ff2e2015838c7b5
    domain: zha
    type: remote_button_short_press
    subtype: button_1
    trigger: device
  - device_id: 9c6819f3f212fba3be12286168e607d2
    domain: zha
    type: remote_button_short_press
    subtype: button_1
    trigger: device
conditions: []
actions:
  - type: turn_on
    device_id: b7222c0ef52b0c3605aa3767117160bf
    entity_id: 3f5ff53caf4f792eb4226252d8e051f8
    domain: switch
mode: single
`

fkhr79, 2025-03-04:
Thx @larsneo !
Investigated and thought about it for a while: as long as the implementation regarding the entity naming is not changed (any more - and at the moment it doesn't feel for me to be necessary again) and you don't remove the device between upcoming version changes, the automation _should_ be stable as the entity id's _should_ not change. At least for upcoming versions.