LarsK1/hass_solvis_control #199: [FEATURE] Enable SG-Ready for dynamic energy prices or solar using

churge82, 2025-03-04:
**Is your feature request related to a problem? Please describe.**
Ich habe einen dynamischen Stromtarif und möchte in Phasen von niedrigen Strompreisen meine Wärmepumpe im verstärkten Betrieb fahren, um den Warmwasserspeicher aufzufüllen.

**Describe the solution you'd like**
Ich möchte einen Schalter in HASS, den ich ansteuern kann, wenn der Preis günstig ist.

**Additional context**
Ich habe im Einsatz:

- SolvisBen mit SolvisLea 8,3 Premium und einem Solvis TIM 3 kW
- Stromanbieter tibber

Unser Heizungsmonteur hat uns für diese Lösung einen Smart Energy Broker und einen Shelly Pro3EM ins Angebot geschrieben. 
[Infos zu SG-Ready](https://www.waermepumpe.de/normen-technik/sg-ready/)
[Code Snippet](https://www.photovoltaikforum.com/thread/212186-zweimal-sg-ready-an-w%C3%A4rmepumpe/?pageNo=2), bei dem man vielleicht abschauen kann.
Es gibt scheinbar 4 Smart Grid Modi. Interessant wäre das Umschalten zwischen den Modi 2-4 aus HASS heraus.

churge82, 2025-03-04:
Ich fürchte, ganz ohne manuellen Aufwand wird es nicht gehen. Was ich so lese, klingt nicht so, als ob man das per SW schalten kann. Aber ich lasse mich gerne verbessern.
Gibt einen Schaltplan von Solvis, welche Kontakte beschaltet werden müssen, damit die WP die Betriebsart ändert.
[Schaltplan](https://www.photovoltaikforum.com/core/attachment/545009-solvis-sg-ready-platine-schaltplan-pdf/)

fkhr79, 2025-03-04:
Didn't try yet... but there are fitting holding registers to set via modbus, I think. Will check later. :-)

fkhr79, 2025-03-04:
@churge82 
Sooo: jetzt habe ich Zeit. Ich wechsele mal in's Deutsche... geht etwas schneller 😉 
Habe mir gerade mal genauer die zuletzt erhaltene Excel-Tabelle mit den Registern angesehen: leider (leider, leider) gibt es softwareseitig keine dort dokumentierte Möglichkeit, die SmartGrid-Modi zu setzen. Allenfalls die Ziel-Temperaturen für die Modi wären per Modbus (was wir bei dieser Integration nutzen) möglich.

Ja, es gibt 4 SmartGrid Betriebszustände für die WP: An, Aus (Sperre), verstärkter Betrieb und Maximalbetrieb.

Mit dem Schaltbild von Dir (bzw. weiteren - das muss ja zu Deiner Anlage passen...) kann man sicher eine gute und stabile Schaltung hinbekommen. Mein Tipp (falls benötigt): ESP32 mit ESPHome und ein paar Relais. Fertig. Zumindest gedanklich...

Aber: Ich sprach erst jüngst noch mit einem Solvis-Servicetechniker über das Thema. Bedenke bitte bei der Nutzung von erhöhten Betriebsarten, dass die WP zwar grundsätzlich hohe Temperaturen **kann** - dem Verdichter dieser Betrieb aber nicht wirklich gut tut: er verschleißt VIEL schneller... und so ein Verdichter ist kein Schnäppchen. Insofern würde ich vermutlich eher beim Normalbetrieb bleiben: SO günstig KANN kein Stromtarif sein (ich wechsele sehr regelmäßig), so dass sich Aufwand, Risiko und mögliche Kosten rentieren. Ist aber selbstverständlich auch immer individuell zu beurteilen und zu entscheiden: bei Wärmepumpen schon fast eine Glaubensfrage.

Das Ticket mache ich jetzt zu, da wir leider nichts tun können. Vielleicht aber ändert sich das in Zukunft? Who knows!

Viel Erfolg!

closed 2025-03-04 by fkhr79