LarsK1/hass_solvis_control #232: [FEATURE] Additional config step for HKRx Room Temp read/write configuration

fkhr79, 2025-03-20:
**Is your feature request related to a problem? Please describe.**
source: #172

**Describe the solution you'd like**
I think we should add another config step for config of the hkrx_room_temp configuration: let's say up to three select / dropdown boxes. one for each active/selected hkr (so it depends on the selected hkr's). options in every box are: read, write.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

fkhr79, 2025-04-03:
Hey @LarsK1 
did it as suggested. Created a new branch so that you can try it out before merging into main. Tests are also adjusted already.

LarsK1, 2025-04-06:
Hi @fkhr79 ,
thanks again for your precious time. I'm unsure about, whether the default should be disabled. I think rather that the default should be read only. Also we should respect, that the options for HKR2/3 are only shown, once the HKRx is enabled. 

Where I'm unsure is about the migration, since we're changing used constants into new ones, which denies any backwards compatibility. But it's probably okay, since 2.0.0 will definitely break many things 😜.

Feel free to merge into main, once that's done.

fkhr79, 2025-04-07:
Hey @LarsK1 ,
welcome back! 🙂 

> I'm unsure about, whether the default should be disabled. I think rather that the default should be read only. Also we should respect, that the options for HKR2/3 are only shown, once the HKRx is enabled.

It's already implemented exactly like that. Have a look at this:

```
def get_solvis_roomtempsensors(data: ConfigType) -> Schema:
    schema_fields = {
        vol.Required("hkr1_room_temp_sensor", default="1"): SolvisRoomTempSelect,
    }
    if data.get(CONF_OPTION_1, False):
        schema_fields[vol.Required("hkr2_room_temp_sensor", default="1")] = SolvisRoomTempSelect

    if data.get(CONF_OPTION_2, False):
        schema_fields[vol.Required("hkr3_room_temp_sensor", default="1")] = SolvisRoomTempSelect

    return vol.Schema(schema_fields)
```

All schema_fields default to "1" (=reading) and are shown only if the corresponding CONF_OPTION is set. 

> Where I'm unsure is about the migration, since we're changing used constants into new ones, which denies any backwards compatibility. But it's probably okay, since 2.0.0 will definitely break many things 😜.

Also thought about that and implemented it as follows:

```
if current_version == 2 and current_minor_version == 3:
        _LOGGER.info(f"Migrating from version {current_version}_{current_minor_version}")
        if CONF_OPTION_6 not in new_data:
            new_data[CONF_OPTION_6] = False
        else:  # keep read-setting for all sensors, if read was set
            new_data[CONF_OPTION_9] = new_data[CONF_OPTION_6]
            new_data[CONF_OPTION_11] = new_data[CONF_OPTION_6]
        if CONF_OPTION_7 not in new_data:
            new_data[CONF_OPTION_7] = False
        else:  # keep write-setting for all sensors, if write was set
            new_data[CONF_OPTION_10] = new_data[CONF_OPTION_7]
            new_data[CONF_OPTION_12] = new_data[CONF_OPTION_7]
        if CONF_OPTION_9 not in new_data:
            new_data[CONF_OPTION_9] = False
        if CONF_OPTION_10 not in new_data:
            new_data[CONF_OPTION_10] = False
        if CONF_OPTION_11 not in new_data:
            new_data[CONF_OPTION_11] = False
        if CONF_OPTION_12 not in new_data:
            new_data[CONF_OPTION_12] = False
        current_minor_version = 4
```

Consequently, there shouldn't be any problems with the migration: everything will remain the same for users updating to this version.
The existing setting of CONF_OPTION_6 (previously: read) results in the same configuration for CONF_OPTION_6, _9, and _11 (all now: read hkr1, 2, and 3).
The same applies to the previous setting of CONF_OPTION_7 (write), which is now transferred to CONF_OPTION_7, _10, and _12.
I think this is the key aspect for most users.

Yes, it's not backward compatible. But I believe that's negligible: who wants to use an old version when they can get their hands on this masterpiece of control??? 😉

PR #256 (2025-04-03): Feat: Add new config step for hkrx room temp sensors
Fix failing options_update_listener @ __init__.py

Fix #232
commit 29cef37b (2025-04-03): Feat: Add new config step for hkrx room temp sensors

Fix failing options_update_listener @ __init__.py

Fix #232
PR #259 (2025-04-07): Feat: Add new config step for hkrx room temp sensors
Fix failing options_update_listener @ __init__.py

Fix #232
closed 2025-04-07 by fkhr79