LarsK1/hass_solvis_control #233: [BUG] Dutch translations missing

fkhr79, 2025-03-20:
**Version of the plugin**
2.0.0-alpha.11 +

 **Home-Assistant Version**
unknown

**Enviroment (Docker, HAOS, ...)**
unknown

**Version of your Solvis software/hardware**
Version NBG: unknown
Version SC3: unknown

**Describe the bug**
"Another issue is that I’ve translated everything into Dutch, but most entities are still in German or English."

**To Reproduce**
./.

**Expected behavior**
Dutch translation for all entities

**Screenshots**
./.

**Additional context**
Source: @sneekerzzz at #172

LarsK1, 2025-03-20:
Hi,
for me https://github.com/LarsK1/hass_solvis_control/blob/main/custom_components%2Fsolvis_control%2Ftranslations%2Fnl.json looks pretty Netherlands.

Please do note, that language files are not updated between releases, so only the next alpha, beta, stable will update the language files in your instance.

@sneekerzzz Therefore in the next update, this should resolve it selve.

fkhr79, 2025-03-20:
Hey!
Even if he uses the current main branch version?
Yes: I saw the Dutch translations... but thought to better check twice for any other issue...

LarsK1, 2025-03-21:
Probably yes, since afaik the reload only happens on version string changes.

LarsK1, 2025-03-31:
Fixed with 2.0.0-beta.1

closed 2025-03-31 by LarsK1