LarsK1/hass_solvis_control #238: [CHANGE] Modbus registers scanned by diagnostics

fkhr79, 2025-03-22:
Currently, the diagnostic method async_get_config_entry_diagnostics scans a very large number of Modbus registers:

    range(2000, 5000), unit=1
    range(32000, 34000), unit=1
    range(1000, 4000), unit=2
    range(34000, 35000), unit=2

Most of these registers are not readable, which results in numerous warnings and errors being logged by pymodbus.(My Modbus connection was smoking... and my test system wasn't happy either.)

Was this intentional? Is there a specific purpose behind it?
Or could the register scan be limited to commonly used registers only?

LarsK1, 2025-03-23:
Hi @fkhr79 ,
the idea behind this "masterpiece" 😜, was to provide users with an easy way to scan the modbus range for new data values.

I'm still not sure, if that's the ha-way or if we really should migrate that into a new action call and really make this one a diagnostic section.

While writing I think, it's best to really clean it up - diagnostic contains some kind of diagnostic tests, logs, etc and on top we add a scan-Action (for v2.1)

What do you think about that?

fkhr79, 2025-03-23:
Hey @LarsK1 ,

I'm with you: the primary goal is to read/write register values, not to search for new or unknown ones. If it were a minor issue, it wouldn’t matter. However, this scan routine is quite extensive and, in my opinion, could confuse users.
I very much like your idea of implementing a specific action for it! The functionality is available through integration, but it shouldn’t be easily or accidentally triggered by an unsuspecting user.

fkhr79, 2025-03-24:
@LarsK1 
Since we are apparently of the same opinion, I've taken the liberty of implementing the change directly: still need to add tests to finally achieve the test coverage...

PR #242 (2025-03-24): Refactor(diagnostics): scan only registers defined in REGISTERS
Update scan_modbus_registers to filter registers based on the REGISTERS constant in const.py

Fix #238
commit b775267e (2025-03-24): Refactor(diagnostics): scan only registers defined in REGISTERS

Update scan_modbus_registers to filter registers based on the REGISTERS constant in const.py

Fix #238
closed 2025-03-24 by fkhr79