LarsK1/hass_solvis_control #264: Configuaration hangs at Modbus discovery

timmost, 2025-04-12:
**Version of the plugin**
2.0.0-alpha.11

 **Home-Assistant Version**
Core
2025.4.2
Supervisor
2025.04.0
Operating System
15.1
Frontend
20250411.0

**Enviroment (Docker, HAOS, ...)**
HAOS

**Version of your Solvis software/hardware**
Device info
Solvis Control 2
by Solvis
Firmware: 2.05.08
Hardware: 3.00.00

**Describe the bug**
When setting up the Solvis, Modbus discover fails. 
Works with Alpha 10

**To Reproduce**
Delete device
Redownload to version
setup device
after entry of ip, modbus fails with error message

**Expected behavior**
full config completed like in alpha10

**Screenshots**

**Additional context**
Add any other context about the problem here.

2025-04-12 15:56:46.863 INFO (MainThread) [custom_components.solvis_control.config_flow] Initialize config flow for solvis_control
2025-04-12 15:56:51.087 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:56:51.227 WARNING (MainThread) [homeassistant.components.kodi.media_player] Unable to connect to Kodi via websocket
2025-04-12 15:56:51.473 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.55', 502)
2025-04-12 15:56:51.574 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:56:51.575 ERROR (MainThread) [custom_components.solvis_control.config_flow] Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.178.55:502]
2025-04-12 15:56:56.710 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:56:57.208 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.55', 502)
2025-04-12 15:56:57.309 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:56:57.309 ERROR (MainThread) [custom_components.solvis_control.config_flow] Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.178.55:502]
2025-04-12 15:57:15.578 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:57:15.957 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.55', 502)
2025-04-12 15:57:16.058 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:57:16.058 ERROR (MainThread) [custom_components.solvis_control.config_flow] Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.178.55:502]
2025-04-12 15:57:18.207 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:57:18.707 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.55', 502)
2025-04-12 15:57:18.809 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:57:18.809 ERROR (MainThread) [custom_components.solvis_control.config_flow] Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.178.55:502]
2025-04-12 15:58:32.007 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:58:32.457 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.55', 502)
2025-04-12 15:58:32.557 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] Connected to Modbus for Solvis
2025-04-12 15:58:32.558 ERROR (MainThread) [custom_components.solvis_control.config_flow] Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.178.55:502]

LarsK1, 2025-04-12:
Hi @timmost ,
this is an error we did encounter some times. In most cases it helped to completely take the solvis heating device from power (on newer devices press the power button on the front) and then wait a minute and repower it again.

timmost, 2025-04-12:
@lars - let me check, but this problem does not happen with alpha 10 - but consistently with alpha 11

timmost, 2025-04-12:
hi @LarsK1 - nein - kompletter Restart aller Komponenten einschl. Solvis... mit Alpha 11 hängts. 

```
2025-04-12 16:27:32.686 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:27:32.686 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register heating_circuits_energy/33541: ExceptionResponse(dev_id=1, transaction_id=11, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:27:33.437 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:27:33.438 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register heat_generator_2_power_thermal/33546: ExceptionResponse(dev_id=1, transaction_id=12, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:27:34.442 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:27:34.443 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register heat_generator_2_power_electric/33547: ExceptionResponse(dev_id=1, transaction_id=13, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:27:35.192 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:27:35.192 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register warm_water_power/33549: ExceptionResponse(dev_id=1, transaction_id=14, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:27:35.691 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:27:35.691 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register solar_pump_primary_runtime/33552: ExceptionResponse(dev_id=1, transaction_id=15, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:27:36.442 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:27:36.443 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register solar_pump_secondary_runtime/33553: ExceptionResponse(dev_id=1, transaction_id=16, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:27:40.362 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.55', 502)
2025-04-12 16:27:40.463 ERROR (MainThread) [custom_components.solvis_control.config_flow] Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.178.55:502]
2025-04-12 16:28:16.066 WARNING (MainThread) [homeassistant.components.kodi.media_player] Unable to connect to Kodi via websocket
2025-04-12 16:28:26.204 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:28:26.205 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register heat_generator_2_power_thermal/33546: ExceptionResponse(dev_id=1, transaction_id=1, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:28:26.704 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:28:26.704 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register heat_generator_2_power_electric/33547: ExceptionResponse(dev_id=1, transaction_id=2, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:28:27.455 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:28:27.456 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register warm_water_power/33549: ExceptionResponse(dev_id=1, transaction_id=3, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:28:27.456 ERROR (MainThread) [custom_components.solvis_control.sensor] Invalid data from coordinator: <class 'dict'> expected
2025-04-12 16:28:27.456 ERROR (MainThread) [custom_components.solvis_control.sensor] Invalid data from coordinator: <class 'dict'> expected
2025-04-12 16:28:27.456 ERROR (MainThread) [custom_components.solvis_control.sensor] Invalid data from coordinator: <class 'dict'> expected
2025-04-12 16:28:37.956 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:28:37.956 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register heat_generator_2_power_thermal/33546: ExceptionResponse(dev_id=1, transaction_id=1, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:28:38.455 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:28:38.455 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register heat_generator_2_power_electric/33547: ExceptionResponse(dev_id=1, transaction_id=2, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:28:38.954 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:28:38.954 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register warm_water_power/33549: ExceptionResponse(dev_id=1, transaction_id=3, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:28:38.956 ERROR (MainThread) [custom_components.solvis_control.sensor] Invalid data from coordinator: <class 'dict'> expected
2025-04-12 16:28:38.956 ERROR (MainThread) [custom_components.solvis_control.sensor] Invalid data from coordinator: <class 'dict'> expected
2025-04-12 16:28:38.956 ERROR (MainThread) [custom_components.solvis_control.sensor] Invalid data from coordinator: <class 'dict'> expected
[... cut ...]
2025-04-12 16:29:03.457 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register heat_generator_2_power_electric/33547: ExceptionResponse(dev_id=1, transaction_id=7, address=0, count=0, bits=[], registers=[], status=1)
2025-04-12 16:29:03.956 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-12 16:29:03.957 ERROR (MainThread) [custom_components.solvis_control.coordinator] Modbus error reading register warm_water_power/33549: ExceptionResponse(dev_id=1, transaction_id=8, address=0, count=0, bits=[], registers=[], status=1)
``` 

mit alpha 10 gehts wieder...

LarsK1, 2025-04-12:
@timmost thanks, now I've think I see the issue

@fkhr79 ich glaube ich hab's - die Register für die Version/Anzahl HKR scheint es in der SC2 nicht zu geben, und wir haben diesen Fehler nicht abgefangen. Könnte das sein?

timmost, 2025-04-12:
Welches Register ist das? let me check

fkhr79, 2025-04-12:
Ahoi! Kann gerade nicht schauen. Später oder morgen.
Ein HKR-Fehler sollte aber eigentlich kein Problem sein: es sollte nur ein Warning geworfen werden. Der Flow sollte weiterlaufen. Eigentlich. ... und wenn ich mich recht erinnere.
Aber: muss ich mir ansehen...

timmost, 2025-04-13:
Servus, nochmals ausführlichere Log, hier mit der Beta1. Alles gelöscht, Integration im HACS gelöscht, neu geladen, neu aufgesetzt. 

[image]

[image]
```
2025-04-13 08:11:19.619 WARNING (SyncWorker_10) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:19.620 WARNING (SyncWorker_10) [homeassistant.loader] We found a custom integration dwd_weather which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:19.621 WARNING (SyncWorker_10) [homeassistant.loader] We found a custom integration bwt_perla which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:19.621 WARNING (SyncWorker_10) [homeassistant.loader] We found a custom integration e3dc_rscp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:19.622 WARNING (SyncWorker_10) [homeassistant.loader] We found a custom integration device_tools which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:19.623 WARNING (SyncWorker_10) [homeassistant.loader] We found a custom integration homematicip_local which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:32.269 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:32.270 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration dwd_weather which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:32.271 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration solvis_control which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:32.271 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration bwt_perla which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:32.272 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration e3dc_rscp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:32.272 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration device_tools which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:32.273 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration homematicip_local which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
s6-rc: info: service legacy-services: stopping
[06:11:46] INFO: Home Assistant Core finish process exit code 0
[06:11:46] INFO: Home Assistant Core service shutdown
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun home-assistant (no readiness notification)
s6-rc: info: service legacy-services successfully started
2025-04-13 08:11:52.848 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:52.849 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dwd_weather which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 08:11:52.850 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solvis_control which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience  [... cut ...]
2025-04-13 08:15:36.622 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] [fetch_modbus_value] Closing Modbus connection: AsyncModbusTcpClient 192.168.178.55:502
2025-04-13 08:15:36.622 DEBUG (MainThread) [custom_components.solvis_control.utils.helpers] [fetch_modbus_value] Modbus connection closed
2025-04-13 08:15:36.622 ERROR (MainThread) [custom_components.solvis_control.config_flow] ConnectionException: Modbus Error: [Connection] [fetch_modbus_value] Failed to connect to Modbus device at 192.168.178.55:502
``` 
Kann das mit dem Umbau (?) der Modbusabfrage zum Helper zusammenhängen (hatte eine Veränderung im Gitthub gesehen)?

VG timmo

fkhr79, 2025-04-13:
Moinmoin @timmost !
Danke, dass Du direkt die beta.1-logs bereitgestellt hast - danach hätte ich vermutlich gefragt. 🙂 

Ich kann es mit der SC2 leider nicht testen. Ich vermute allerdings stark, dass es mit den schnellen Verbindungsauf- und abbauen zusammenhängt: der erste Aufbau funktioniert und das Register 32770 wird erfolgreich ausgelesen. Der nächste Verbindungsaufbau geht in die Hose - ich vermute, weil die Remote etwas lahm/träge ist. Die Fehlermeldung `[pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.55', 502)` jedenfalls besagt, dass die Verbindung aktiv abgelehnt wird: entweder weil der Server nicht antwortet/bereit ist/Verbindungen akzeptiert, der Port nicht offen ist, der Zielhost nicht bereit ist oder die Anfrage sonstwie blockiert wird. Da der erste Aufruf durchgeht, kann es eigentlich nur daran liegen, dass die Remote nach der letzten Verbindung noch nicht wieder bereit ist, eine neue Verbindung aufzubauen.

Jetzt könnte man einerseits mit einer kleinen Verzögerung arbeiten oder aber die Abfragefunktion umbauen. Tatsächlich fand ich es bisher immer schon etwas unelegant, jedes Mal die Verbindung vollständig ab und dann wieder aufzubauen. War bisher aber nie ein Problem, daher hatte ich es einfach dabei belassen.
Habe jetzt den Anlass genutzt und die Abfragefunktion umgebaut, so dass mit EINER Verbindung BEIDE Werte ausgelesen werden - sprich: der zweite Call fällt weg und damit hoffentlich auch der zweite fehlerhafte Connect. Ich schaue jetzt noch mal eben, ob es noch andere Stellen gibt, die das Problem provozieren könnten. Dann passe ich noch das Testing an (wenn erforderlich - muss gleich mal sehen) und dann könntest Du nochmal probieren...?! DANKE VORAB DAFÜR!!

Und: ja, dass hängt wahrscheinlich mit dem Umbau zusammen. Und mit dem Einführen der Tests: währenddessen - so erinnere ich mich vage - hatte ich an eben dieser Stelle einen Bug aufgedeckt und entfernt. Ich meine, dass es so war, dass vor dem Umbau ein inkonsistentes Abfrageverhalten existierte, dass nicht auffiel, nicht wirklich problematisch war, aber dessen Behebung (jetzt richtige und konsistente Abfrage) bei Dir jetzt zu diesem Fehler führt (der vorher hätte auch schon auftauchen müssen, wenn die Abfragen korrekt gewesen wären). Ist letztendlich aber auch egal: sollte jetzt behoben sein.

LarsK1, 2025-04-13:
@fkhr79 ,
beim lesen fällt mir gerade noch auf. Wir machen die Verbindung zwar auf, aber es gibt kein `modbus.close()` 😅

fkhr79, 2025-04-13:
@LarsK1 

Aso? - Du meinst in der alten Version, oder? - Habe gerade vollkommen schockiert nochmal nachgesehen, weil ich mich bei den Tests recht intensiv mit den close-calls beschäftigt habe: im finally-block steht extra ein synchrones modbussocket.close(). 
Vorher war es teilweise auch asynchron implementiert, was in bestimmten Situationen Fehler produziert hat (die einem dann beim Testen schwer auf die Füße fallen...).

LarsK1, 2025-04-13:
@fkhr79 Oh mann, ja Du hast natürlich Recht. Das war nur in der Commit-Ansicht nicht sichtbar. Sorry - alle Mann zurück

fkhr79, 2025-04-13:
So, @timmost ,
wenn Du wollen würdest, könntest Du mit der aktuellen Version im main-branch einmal testen, ob der Fehler noch auftaucht #daumendrück

Danke Dir schonmal sehr vorab für Deine Mühe!!

@LarsK1 ,
Puh... ich dachte schon! 😉 Mir wurde gerade schon heiß und kalt, dass ich womöglich einen kapitalen Bock geschossen hätte...

timmost, 2025-04-13:
Hi @LarsK1 
ich erinnere mich, das die Modbus-Schnittstelle des Solvis auch bei FHEM sauer wurde, wenn sie zu lange offen war oder gar aus verschiedenen Richtungen angesprochen wurde. 

Aber jetzt scheints durchzulaufen, config geht mit dem neuen Main durch. 

Allerdings - Ich habe jetzt aber noch eine Reihe Errors im Log - die scheinen ähnliche Modbus Issues zu sein. macht ihr die TCP-Verbindung wieder zu?

Es gibt auch noch eine "Division by zero" beim Waterflow:  Division by zero for warm_water_volume_flow_s18 with value 0

VG timmo

```
services-up: info: copying legacy longrun home-assistant (no readiness notification)
s6-rc: info: service legacy-services successfully started
2025-04-13 15:25:42.902 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 15:25:42.903 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dwd_weather which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 15:25:42.903 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solvis_control which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 15:25:42.904 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration bwt_perla which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 15:25:42.904 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration e3dc_rscp which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 15:25:42.905 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration device_tools which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 15:25:42.905 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration homematicip_local which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-04-13 15:26:04.017 WARNING (MainThread) [homeassistant.components.kodi.media_player] Unable to connect to Kodi via websocket
2025-04-13 15:27:30.221 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-13 15:27:30.221 ERROR (MainThread) [custom_components.solvis_control.coordinator] [heating_circuits_energy | 33541] Modbus error while reading register: ExceptionResponse(dev_id=1, transaction_id=11, address=0, count=0, bits=[], registers=[], status=1)
2025-04-13 15:27:30.725 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-13 15:27:30.725 ERROR (MainThread) [custom_components.solvis_control.coordinator] [heat_generator_2_power_thermal | 33546] Modbus error while reading register: ExceptionResponse(dev_id=1, transaction_id=12, address=0, count=0, bits=[], registers=[], status=1)
2025-04-13 15:27:31.223 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-13 15:27:31.224 ERROR (MainThread) [custom_components.solvis_control.coordinator] [heat_generator_2_power_electric | 33547] Modbus error while reading register: ExceptionResponse(dev_id=1, transaction_id=13, address=0, count=0, bits=[], registers=[], status=1)
2025-04-13 15:27:31.723 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-13 15:27:31.723 ERROR (MainThread) [custom_components.solvis_control.coordinator] [warm_water_power | 33549] Modbus error while reading register: ExceptionResponse(dev_id=1, transaction_id=14, address=0, count=0, bits=[], registers=[], status=1)
2025-04-13 15:27:32.226 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-13 15:27:32.227 ERROR (MainThread) [custom_components.solvis_control.coordinator] [solar_pump_primary_runtime | 33552] Modbus error while reading register: ExceptionResponse(dev_id=1, transaction_id=15, address=0, count=0, bits=[], registers=[], status=1)
2025-04-13 15:27:32.726 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-13 15:27:32.727 ERROR (MainThread) [custom_components.solvis_control.coordinator] [solar_pump_secondary_runtime | 33553] Modbus error while reading register: ExceptionResponse(dev_id=1, transaction_id=16, address=0, count=0, bits=[], registers=[], status=1)
2025-04-13 15:28:11.806 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-13 15:28:11.807 ERROR (MainThread) [custom_components.solvis_control.coordinator] [heat_generator_2_power_thermal | 33546] Modbus error while reading register: ExceptionResponse(dev_id=1, transaction_id=1, address=0, count=0, bits=[], registers=[], status=1)
2025-04-13 15:28:12.556 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-04-13 15:28:12.556 ERROR (MainThread) [custom_components.solvis_control.coordinator] [heat_generator_2_power_electric | 33547] Modbus error while reading register: ExceptionResponse(dev_id=1, transaction_id=2, address= [... cut ...]
2025-04-13 15:29:53.344 ERROR (MainThread) [custom_components.solvis_control.coordinator] [warm_water_power | 33549] Modbus error while reading register: ExceptionResponse(dev_id=1, transaction_id=8, address=0, count=0, bits=[], registers=[], status=1)
2025-04-13 15:30:11.094 WARNING (MainThread) [custom_components.solvis_control.sensor] Division by zero for solar_volume_flow_s17 with value 0
2025-04-13 15:30:11.095 WARNING (MainThread) [custom_components.solvis_control.sensor] Division by zero for warm_water_volume_flow_s18 with value 0
```

LarsK1, 2025-04-13:
Hi @timmost ,
kannst du bitte noch einmal die Debug-Logd für die Integration einschalten und ein volles Slow-Intervall abwarten?

timmost, 2025-04-13:
via config?.. das ist viel...

[logfile beta1 Modbus.txt](https://github.com/user-attachments/files/19725634/logfile.beta1.Modbus.txt)

vG timmo

fkhr79, 2025-04-13:
Der Fehler entsteht aufgrund einiger von der SC2 nicht unterstützter Register ("Exception response 132 / 0" von pymodbus).
Möglich ist, dass weitere Exceptions/Probleme dabei sind, die aufgrund der Trägheit der Remote und der Lesefehler entstehen: das müssen wir noch sehen.
Ich habe gerade einen Fix gemerged, der die Register umkonfiguriert, so dass die SC2 sie nicht mehr versucht abzurufen. Damit sollten alle 132'er-Exceptions erledigt sein.
Sollten noch Fehler auftauchen: dann bitte nochmal ein neues, vollständiges Log.

timmost, 2025-04-13:
Hi @LarsK1 , mache ich in den nächsten Tagen - ist dass dann der Main branch?
VG timmo

timmost, 2025-04-13:
konnte nich warten ;-) schaut sehr gut aus. 
Wie sind die Meldungen "Skipping update: no data available in coordinator. Skipped update!?!" zu interpretieren?

[logfile beta1 Modbus 22_03 SC2 fixes.txt](https://github.com/user-attachments/files/19727029/logfile.beta1.Modbus.22_03.SC2.fixes.txt)
Herzlichen Dank für die Spätschicht...
VG timmo

fkhr79, 2025-04-13:
Bin zwar nicht der Lars, aber: gern.

Das sind reine debug-Meldungen: tauchen i. d. R. auf, wenn beim Polling
eine Entität übersprungen wird, weil sie noch nicht dran ist.

timmost, 2025-04-14:
@fkhr79 Welche Version soll ich testen - ich sehe noch einige Fehler im Log, bin mir aber nicht sicher, welcher Branch jetzt der relevante ist? 
RE neue Register - klar, macht aber nix - solange die Solvis laüft und keine Probleme im HASS liefert. 
V G timmo

fkhr79, 2025-04-14:
@timmost 
Dann mache ich das Ticket mal wieder auf. 😆 
Gerne die aktuellste Version im main-branch (das ist die beta mit allen commits, die danach noch kamen).

... und wenn Du gerade beim Testen bist: ich habe gerade die Doku der gestrigen Änderungen nachziehen wollen und bin ein wenig irritiert, dass gestern auch die Register 

- 3840 (Analog Out 1 Status), 
- 33036 (Temp S13), 
- 33037 (Temp S14) und
- 33038 (Temp S15) 

Probleme gemacht haben. Die hatte ich für die SC2 ebenfalls entfernt - waren vorher aber eigentlich immer unproblematisch. Es könnte allerdings sein, dass die Fehler bei diesen Registern nur Nebeneffekte waren. Solltest Du also Testlust verspüren (😜) könntest Du bei Dir lokal in der const.py (unter /custom_components/solvis_control/) bei den entsprechenden Registern mal den Eintrag `supported_version=1` durch `supported_version=0` ersetzen und schauen, ob die Register noch Fehler produzieren....?! #ganzliebschau
Nach den Änderungen musst Du HA immer einmal neustarten, damit die Änderungen greifen.

timmost, 2025-04-14:
so , doch noch die Register eingepflegt. 

Eine dunkle Vermutung re register... ich glaube, die Monteure hatten zumindest einen Sensor anders belegt... irgendwie ging der Drucksensor für die Solaranlage nicht. Aber die Erinnerung ist ziemlich vage, muss ich recherchieren, aber ich glaube nicht dass es vier Stück waren.

 Könnte sein, dann wir evtl eine kurze Anleitung für diesen Fall einbauen sollten "wenn Deine Monteure kreativ waren"

System läuft mit dem neuen Download. Melde mich morgen
VG timmo

fkhr79, 2025-04-14:
Puh! Da habe ich mich wohl zu früh gefreut...
Wärest Du so freundlich, dann bitte das gesamte Log (nicht nur die Ausschnitte) als Datei hochzuladen? Weniger Aufwand für Dich - mehr Info für mich. 😉 
Außerdem könntest Du einmal die Diagnosedaten herunterladen und hier anhängen (Dreipunkt-Menü bei dem Gerät -> Diagnosedaten herunterladen). Da werden dann einmal alle Register angefragt und ausgelesen: würde mich mal interessieren, was dabei herauskommt. Vielleicht eine gute Übersicht?
Ansonsten: morgen geht's weiter.

-

Das andere Belegen der Sensoren in Deiner Anlage ist eigentlich kein Problem. Das ist nur die Hardware-Seite. Die Software stellt die Register so oder so bereit (oder halt eben nicht). So habe ich z. B. diverse Temperaturwerte von Sensoren die nicht angeschlossen sind. Auslesen kann man die (Null)Werte aber.

Noch ne Frage: wie wär's mit ner SC3?! 😜 #duckundweg

timmost, 2025-04-14:
SC3 ... nee - das kostet einen Tausi bis der Moonteur wieder weg ist... ;-) 

Schon mal der Anlagenstatus - S14 ist offline, ich lade aber noch die Diagnosedaten runter. 

Log mache ich 

[image]

timmost, 2025-04-15:
@fkhr79 So, jetzt mal nach 20 std Logging...
Das erscheint im log... sind halt 300 mb... wenn wr das einschänken können, gerne
VG timmo

fkhr79, 2025-04-15:
N'abend @timmost 

... ich mag auch nicht so große Dateien... 😉 

Okay. Habe mal angefangen und sammele hier erstmal meine Erkenntnisse. Fertig werde ich heute damit nicht - aber anfangen kann man ja schonmal:

* Die "Division by zero" ist zum Glück unproblematisch: da wird nur an der falschen Stelle eine unnötige Warnung ausgegeben... nämlich immer dann, wenn der Wert 0 ist. In dem Fall wird aber auch gar keine Division durchgeführt, die problematisch sein könnte. Insofern: unschöne Warnung, die einfach zu beseitigen sein wird. Problem zum Glück keines.
* Irgendwann - warum auch immer - taucht zu Beginn eines neuen coordinator-zyklus `[pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.178.55', 502)` auf. Der Fehler besagt "Connection Refused". Daraus folgend stammen dann auch die ganzen Lesefehler in dem Zyklus: es ist ja keine Modbus-Verbindung da. Insofern scheitern auch die Leseversuche. - Warum der Fehler auf einmal entsteht: keine Ahnung. Liegt vermutlich an der trägen SC2?! - Muss mir mal ansehen, ob man da in der Fehlerbehandlung etwas tun kann und vielleicht testweise eine kurze Verzögerung für SC2's einbauen. Bei späteren Verbindungsversuchen ist dann nämlich alles wieder in Ordnung.
* Die Register 3840 (Analog Out 1 Status), 33036 (Temp S13), 33037 (Temp S14) und 33038 (Temp S15), die Du wieder aktiviert hast, werden sauber gelesen. Hatte ich mir schon fast gedacht: waren also Folgefehler. Können wir also wieder aufnehmen.
* `[pymodbus.logging] Exception response` konnte ich in dem Logging nicht finden. Schau mal, ob das noch vorkommt, bitte. Dann nochmal ein Log von einem entsprechenden Part.

Bei Gelegenheit: gerne auch einmal die Diagnosedaten...
Schicken Abend!

fkhr79, 2025-04-15:
Mit dem commit sollten die division by zero Fehler verschwinden. Auch die benannten Register sind wieder im Ursprungszustand.
(branch: main)

timmost, 2025-04-16:
Mal die erste Ladung nach 45 min
VG timmo

[home-assistant_2025-04-16T19-42-11.060Z.log](https://github.com/user-attachments/files/19783663/home-assistant_2025-04-16T19-42-11.060Z.log)

fkhr79, 2025-04-20:
Hey @timmost ,

habe mir das 45 Minuten Log gerade mal angesehen: keinerlei Fehler. Einerseits gut. Andererseits weniger gut: Du hast ja offenbar noch Fehler... und für deren Analyse wäre ein vollständiges Log schon hilfreich. In dem Abschnitt, den Du oben gepostet hast, sind zwar die Exception Responses zu sehen - aber leider nicht das Verhalten der Integration davor/danach.

Ganz allgemein (auch zu meiner eigenen Doku): 

Eine Modbus-Exception‑Antwort besteht aus genau zwei Bytes (im Modbus-Payload):

1. Exception‑Funktionscode = Ursprungs‑Funktionscode + 128 (0x80).
2. Exception‑Code (Error Code) = Code-Nr. aus der Spezifikation

- "Exception response 132 / 0" : 132 dez (= 0x84) = 4 + 0x80 → Exception bei Funktion 4 „Read Input Registers“
- "Exception response 131 / 0" : 131 dez (= 0x83) = 3 + 0x80 → Exception bei Funktion 3 „Read Holding Registers“
- Der nachfolgende Wert (/ 0) im Log ist eine Pymodbus‑Log‑Ungenauigkeit; das eigentliche Exception‑Code‑Feld findet sich im Objekt als status=1 → Illegal Function.

Werde mir jetzt mal den Quelltext in Ruhe beschauen und sehen, ob sich am Fehlerhandling etwas machen lässt... oder ob sonstwo noch Probleme provoziert werden.

timmost, 2025-04-20:
Hi @fkhr79 

hier nochmals das aktuelle log von heute - komplett als zip... hatte vergessen, den logger abzuschalten.

Ich erinnere mich bie FHEM, das wir da timeouts egetrzt hatten, und die TCP-Verbindung regelmäßig zu schliessen war. 

Lass mich wissen, ob ich da nochmals wa tun so.. VG timmo

[home-assistant.zip](https://github.com/user-attachments/files/19825251/home-assistant.zip)

timmost, 2025-04-20:
Hi @fkhr79 ... die ersten 15 min nach deinem dev brach schauen gut aus ;-)

[home-assistant.log](https://github.com/user-attachments/files/19826398/home-assistant.log)

fkhr79, 2025-04-20:
@timmost ,
soso... Du lebst also gern gefährlich und probierst unfertigen code? 😉 

Jaaaaa... dachte ich auch. Bis ich in die Tests eingestiegen bin: leider zerstörte die Änderung das ein oder andere. Baue gerade nochmal etwas reduzierter um.

fkhr79, 2025-04-20:
Sooooo @timmost ,

da Du ja offensichtlich weißt, wie Du den Code aus einem bestimmten Branch in einem geforkten Repo holst, kann ich mir etwas Arbeit sparen.
Bevor ich die Testabdeckung wieder vervollständige und einen sauberen Commit in den hiesigen Main-Branch mache, wäre es prima, wenn Du den aktuellen Code testen könntest: er liegt [in meinem Fork im dev-branch](https://github.com/fkhr79/hass_solvis_control/tree/dev) . 
Dabei habe ich erstmal darauf verzichtet, für die SC2 Wartezeiten einzubauen. Statt dessen wird jetzt durchgehend nur _eine_ Modbus-Verbindung verwendet und nicht andauernd wieder neu verbunden. Das sollte eigentlich helfen. - Mal abgesehen davon, dass das ohnehin eleganter ist.
Weiterhin verwende ich nun die reconnect-Möglichkeiten der pymodbus-Bibliothek: sollte die Verbindung abreißen oder ein Fehler auftauchen, übernimmt pymodbus (bis zu einem gewissen Grad) das Fehlerhandling und versucht mehrfach, zu verbinden bzw. Werte abzurufen.
Hoffen wir mal, dass das nun reicht. - Bin gespannt. - DANK vorab für's Testen!! 🙂

timmost, 2025-04-20:
Neugier ;-)
ich denke, das mit der einen Modbusverbindung kann es sein. Ich hatte mit Fhem situationen, wo ich über zwei Mechanismen auf das Solvis zugegriffen habe (frage mich aber nicht, was das war... ), und dann gab es eine Lösung mit "close after connection"
so noch schnell einspielen... 
dann bis morgen!
VG timmo

timmost, 2025-04-21:
hi @fkhr79  so, hier mal die erste log-datei... sind heute den tag über unterwegs.. VG timmo

[home-assistant.zip](https://github.com/user-attachments/files/19829659/home-assistant.zip)

fkhr79, 2025-04-21:
Danke, @timmost !
Mehr brauche ich gerade auch erstmal nicht.
Leider tauchen immer noch Exceptions auf. Recht willkührlich und nicht viele. Aber ab und an steigt er aus - warum auch immer. 
Ich baue dann mal weiter um...

fkhr79, 2025-04-21:
Well, well, well,...
@timmost ,
ich habe noch weiter umgebaut und eine neue Client-Option (mit toleranteren Parametern) für die SC2 vorgesehen. Außerdem wird jetzt an zwei Stellen ein Handler der pymodbus-Bibliothek verwendet, der das Verhalten automatisiert und stabiler machen sollte. Ansonsten ein paar Kleinigkeiten... schauen wir mal.
Du wirst wieder in meinem Fork im Dev-Branch fündig: wenn Du wieder testen würdest...?! - Allerliebst! 😉

timmost, 2025-04-21:
@fkhr79 alles klar - eingespielt...

timmost, 2025-04-21:
[home-assistant.zip](https://github.com/user-attachments/files/19838812/home-assistant.zip)

@fkhr79 hier nochmals nach ein paar std das Logfile... nicht wundern, wenn dallas-Sensoren auch vorkommen, da hatte ich in HW problem

VG timmo

fkhr79, 2025-04-21:
N'abend @timmost ,
so langsam geht mir die SC2 an/auf dunkle Stellen... 😠 

In dem Log sind leider immer noch Exceptions. Treten nur sporadisch auf. Ein Muster kann ich leider nicht erkennen. Warum auch immer scheint die SC2 dann und wann mal einfach keine Lust mehr zu haben. Vorher und nachher tut sie dann wieder wie ein Unschuldslamm. Doofe Kuh! 😉 

Mangels besserer/weiterer Ideen habe ich für die SC2 jetzt einmal kurze Pausen nach den Leseversuchen (0,1 s) eingebaut. Vielleicht hilft das der SC2, nicht so in Stress zu geraten...?! Wir werden sehen.

Same procedure: bitte gerne testen (wieder mein Fork, dev-Branch) und dann sehen wir weiter.

Falls Du möchtest, kannst Du die Länge der Pause(n) auch selber verändern und schauen, ob das etwas bringt: in coordinator.py findest Du zwei Stellen...:
```
# slow down on SC2-devices
if int(self.supported_version) == 2:
     _LOGGER.debug(f"[{register.name} | {register.address}] Sleep after read (SC2)")
     await asyncio.sleep(0.1)
```
Dort kannst Du die Schlaf-Dauer anpassen, indem Du "0.1" änderst. Bis zu 0,5 Sekunden könnte man probieren. Danach können wir die Werte auch händisch von der SC2 abschreiben und in HomeAssistant eintippen 😜 

Viel Erfolg und schicken Abend!

timmost, 2025-04-22:
@fkhr79 na wenigstens verlierst Du den Humor nicht... ;-) Die SC2 ist von Solvis schon etwas lieblos implementiert... ich hoffe es klappt, sonst leben wir halt mit den connects. Ist eingespielt, ich schaue heute abend nach, wird aber später. 

vG

timmost, 2025-04-22:
@LarsK1 
Latest version ..

VG timmo
[home-assistant.zip](https://github.com/user-attachments/files/19856061/home-assistant.zip)

fkhr79, 2025-04-22:
@timmost 
Danke für den Test & das Log!
Hmpf. Immer noch Fehler. Wenige. Aber Fehler. Darüber muss ich nachdenken...: Systematiken konnte ich keine finden. Auch zeitlich nicht. ...

timmost, 2025-04-23:
@fkhr79  
so hier nochmals für heute... ich habe jetzt den delay mal wie oben geschireben auf 0.5 sec gesetzt. 
Die Outages um 3 Uhr bitte ignorieren, das ist ein reboot des Interface, den ich schon für FHEM gemacht habe (habe ich jetzt ausgeschaltet)
VG timmo

[log2.zip](https://github.com/user-attachments/files/19873001/log2.zip)

[home-assistant.zip](https://github.com/user-attachments/files/19873000/home-assistant.zip)

fkhr79, 2025-04-23:
Danke, @timmost !
Hat leider alles nichts gebracht: es sind bei beiden Logfiles immer noch Exception Responses enthalten.

Ich bin jetzt mal noch tiefer in die Modbus-Bibliothek eingestiegen, habe ein paar Diagnose-Outputs hinzugefügt sowie die debug-messages von pymodbus aktiviert. Vielleicht bringt das ein paar mehr Informationen, was da los ist.
ACHTUNG: das Log wird dadurch ziemlich schnell ziemlich voll. Ich brauche aber auch nicht viel (also nicht mehrere Stunden). Ein Log mit einem Fehler reicht erstmal. Mehr als 0,5 bis 1 Stunde sollte es vermutlich nicht dauern. Schau bitte mal, was geht.

Schicken Abend!

timmost, 2025-04-24:
so ist eingespielt, nach 2 min schon 400 kb... mei, funkt der durch die Gegend...

timmost, 2025-04-24:
[home-assistant.zip](https://github.com/user-attachments/files/19898289/home-assistant.zip)

@fkhr79  here you are VG timmo

timmost, 2025-04-25:
@fkhr79 ein Gedanke nochmals - machst Du zwischen durch einen Reset der Modbus TCP  Verbindung? Irgendwie war das damals auch ein Lösungsweg.

fkhr79, 2025-04-27:
Schönen guten Sonntag, @timmost !

Habe mir das gerade mal angesehen: leider unterstützt die SC2 wohl auch die Diagnose-Möglichkeiten von pymodbus nicht. Fehler sind nach wie vor vorhanden.

> [@fkhr79](https://github.com/fkhr79) ein Gedanke nochmals - machst Du zwischen durch einen Reset der Modbus TCP Verbindung? Irgendwie war das beim tunen mit FHEM ein Ansatz

Genau DAS hatte ich zuletzt geändert, da die SC2 durch die schnellen Verbindungsaufbauten offenbar "gelähmt" wurde.
Abgesehen davon ist eine dauerhafte Verbindung für die SC3 ebenfalls besser (weniger Traffic).
Jedenfalls habe ich jetzt einen Reconnect bei jedem Lesezyklus der SC2 eingebaut, der auch schöööön langsaaaaam vonstatten geht. Vielleicht hilft das ja?!
Danke für Deine Geduld: teste gerne die Version an bekannter Stelle im dev branch in meinem fork. - Mehr als 15 Minuten brauchts vermutlich nicht: die Fehler treten in der Regel nach wenigen Minuten schon auf.

#daumendrück

timmost, 2025-04-27:
Hi @fkhr79 ist eingespielt... momentan ärgert mich mein ESP32 mit den Dallas Sensoren mindestens ebenso ;-( vielleicht haben wir hier ja Erfolg!
V Gimmo

timmost, 2025-04-27:
@fkhr79 hm, scheint immer noch zu motzen... Mist....so langsam recherchiere ich SC3 preise ;-)

[home-assistant.zip](https://github.com/user-attachments/files/19931344/home-assistant.zip)

fkhr79, 2025-05-01:
@timmost 

So, jetzt habe ich wieder etwas Zeit für dieses Problem.

> .so langsam recherchiere ich SC3 preise ;-)

Spiele nicht mit mir! 😉 
Im Übrigen eine hervorragende Idee: Die SC3 hat ein sensationell schönes, farbiges Display (mindestens 16 Farben!!!), gut zu bedienen... ein Traumgerät!!! ... unterstützt übrigens auch Modbus-Verbindungen! Läuft äußerst stabil. Kurz: ein Must-Have! 😜 

Habe das Log gerade gesichtet: ich verstehe es nicht. Die Reconnects funktionieren bestens. Nur aus irgendwelchen Gründen steigt die Remote zu beliebigen Zeitpunkten einfach mal aus. Das passiert teilweise direkt nach dem Reconnect; teilweise auch erst nach einiger Zeit. Vor und nach dem Fehler ist aber direkt wieder alles i. O. . Am Register hängt es ebenfalls nicht. Ein Rätsel. 
Ein paar restliche Ideen habe ich aber noch:

- Nur sicherheitshalber: Du hast in Deinem Netzwerk nicht zufällig noch irgendwelche "Besonderheiten", die halbwegs regelmäßig die Modbus-TCP-Kommunikation mit der Remote stören könnten? Andere Modbus-Anfragen an die Remote? Starker Traffic irgendwoher? - Nicht, dass die Fehler durch andere Komponenten / Seiteneffekte provoziert werden...?!
- Ich setze - letzte verzweifelte Idee im bestehenden Code - die Wartezeit nach einem Lesevorgang jetzt gleich nochmal hoch (von 0.1 auf 0.3). Ich glaube zwar nicht, dass das etwas bringt... aber: man weiß ja nie. Findest du gleich an bekannter Stelle.
- Schlussendlich werde ich gleich mal in die tiefere Analyse der Änderungen von alpha.10 nach alpha.11 einsteigen: vielleicht finde ich da etwas. Die alpha.10 funktioniert bei Dir ja offenbar fehlerfrei.

timmost, 2025-05-01:
Super - bin diese Tage unterwegs, melde mich…

vis_control#264)

fkhr79, 2025-05-01:
> * Schlussendlich werde ich gleich mal in die tiefere Analyse der Änderungen von alpha.10 nach alpha.11 einsteigen: vielleicht finde ich da etwas. Die alpha.10 funktioniert bei Dir ja offenbar fehlerfrei.

Habe mir die Änderungen jetzt nochmal zu Gemüte geführt: es ist und bleibt mir ein Rätsel, warum die alpha.10 bei Dir fehlerfrei läuft, die alpha.11 hingegen nicht. An dem Modbus-Abfrageverhalten ist quasi nichts geändert worden. Zumindest nicht im coordinator, in dem die Fehler ja nachher auftauchen.

@LarsK1 ... hast Du vielleicht noch Ideen?

@timmost 
Wie viel Motivation hast Du noch? Ich würde als nächstes auf die alpha.11 (bzw. alpha.10) zurückgehen, um Effekte von Änderungen auszuschließen, die danach erfolgt sind. Dann anhand der Änderungen von alpha.10 zu alpha.11 versuchen herauszufinden, welche Änderung zu den Problemen führt. - Das wäre allerdings mit einer gewissen Menge Testaufwand Deinerseits verbunden. Kein Drama... aber halt etwas Aufwand. Fraglich, ob Du bereit bist, den aufzubringen...?
... und bevor wir damit starten sollten, wäre es gut, wenn Du vorher nochmal verifizieren würdest, dass die alpha.10 tatsächlich fehlerfrei läuft. Davon wäre dann debug-Protokoll hilfreich - quasi als Nullmessung/Ausgangspunkt.

timmost, 2025-05-02:
Ich schaue es mir ab morgen an - Motivation ist vorhanden! 👍

timmost, 2025-05-03:
Hi @fkhr79 

ich glaube der Hänger war damals unabhängig vom Modbus... da gings um die Konfigs. 

hier ist das Log von Alpha 9

[home-assistant_2025-05-03T15-45-48.876Z.log](https://github.com/user-attachments/files/20023997/home-assistant_2025-05-03T15-45-48.876Z.log)

ich fahre das jetzt mal mit Alpha 6

vG timmo

fkhr79, 2025-05-03:
N'abend @timmost 

> hier ist das Log von Alpha 9
Das das Log auch exception responses enthält, unterstelle ich mal, dass die v2.0.0-alpha.10 nicht fehlerfrei gelaufen ist? ... und Du daher v2.0.0-alpha.9 getestet hast?

> ich fahre das jetzt mal mit Alpha 1.2.1
Es wäre gut, wenn Du mir die genau Versionsbezeichnung geben würdest, damit es keine Verwechslungen gibt.
Abgesehen davon: Ja, suche gerne die jüngste Version, die fehlerfrei funktionierte. - Gab es denn einmal eine Version, die keine Fehler produzierte?

timmost, 2025-05-04:
@fkhr79 - checke gerade die Versionen… ich werde den Verdacht nicht los, dass die Fehler immer da waren, und einfach unter dem Radar geblieben sind. Bin auch die Anleitungen durchgegangen, es gibt im Werksmodus (:-)) sogar einen “Testmodus” für Modbus - allerdings ist nichts dokumentiert… habe ich eben gefunden (ganz legal..-)

Ich habe jetzt tatsächlich meine Monteur angefunkt wg Kosten SC3, das Teil geht mir tierisch auf den Zeiger… wahrscheinlich ist das Teuerste, den Monteur zu schicken, der sich dann durch die Anleitung quält… mal schauen…

timmost, 2025-05-04:
@fkhr79  so, hier das Protokoll der 1.2.1 (kein Alpha) - einfach mal zurück zum Start. 

Nach dem Log zu urteilen gibts da auch schon Exceptions 132 und Verbindungsabbrüche... arghh. 

Ich habe mir auch mal das 5 Minutenprotokoll der Solvis-Remote angeschaut - was da auffällt, dass da 5-min Intervalle übersprungen werden, aber kein Muster zu erkennen ist. Und wenn ich Remote zugreife, dann hängt die Web-App der Solvis Remote immer wieder, um dann nach ein paar Sekunden wieder zu reagieren. Die Anlage selber immer ist immer responsive, reagiert ohne Verzögerung auf Tastendrücke. Das legt nahe, dass die Problematik in der Solvis-Remote Box liegt, wo die Anbindung ans Heimnetz passiert, und nicht in der Steuerung selber. 

[home-assistant.zip](https://github.com/user-attachments/files/20026485/home-assistant.zip)

timmost, 2025-05-04:
@fkhr79 ich teste gerade die aktuelle 2.0.1. Auch da gibts die Fehler... 
Macht es sinn, deine Dev mal auf die 2.0.1. hochzuziehen und da nochmals die Muster zu untersuchen?

[15 more replies]

commit 09903b1e (2025-05-11): Merge pull request #298 from fkhr79/dev

fix(modbus): improve connection handling and error management

closes #264
closed 2025-05-11 by fkhr79