LarsK1/hass_solvis_control #273: [BUG] No data available for...some entity

stefan088, 2025-04-23:
**Version of the plugin**
1.2.1

 **Home-Assistant Version**
2025.4.3

**Enviroment (Docker, HAOS, ...)**
nuc supervisor

**Version of your Solvis software/hardware**
Version NBG: 3.01.00
Version SC3: 319.47

**Describe the bug**
No data available for hkr1_warmwasser_vorrang No data available for hkr1_betriebsart
No data available for hkr2_betriebsart 
 data available for hkr2_heizkurve_temp_tag_1
No data available for hkr2_heizkurve_temp_tag_2
No data available for hkr2_heizkurve_temp_tag_3
No data available for hkr2_heizkurve_temp_absenkung
No data available for hkr2_heizkurve_steilheit 
No data available for number_gas_burner_start
No data available for hkr1_vorlaufart
No data available for hkr2_vorlaufart
No data available for version_sc3
No data available for version_nbg

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

fkhr79, 2025-04-23:
Hey @stefan088 ,

thanks for reaching out!
Would it be possible for you to install version 2.0.0-beta1? (install via HACS, select version...). Don't fear: it's quite stable!
If the errors are gone: fine! Otherwise - same if you stay with version 1.2.1 - we'll need a log-file to help you out. Could you provide one covering approx 5 to 10 minutes?
Are you a first time user? ... or did the errors occur spontaneously? If so: did you change anything? The more context you can give us the better!

Thanks in advance!

stefan088, 2025-05-05:
i's ok now thk

fkhr79, 2025-05-05:
Good to hear! Thanks!
Will close the issue now.

closed 2025-05-05 by fkhr79