LarsK1/hass_solvis_control #274: [BUG] Einheit Durchflussmenge Solar nicht korrekt.

RedVision81, 2025-04-27:
Plugin: 1.21

HA: 2025.4.4

Version NBG: 3.1.0
Version SC3: MA3.19.47

**Describe the bug**
 Durchflussmenge Solar ist in l/min müsste aber l/h sein

[image]

[image]

fkhr79, 2025-04-27:
Hey @RedVision81 ,

Danke, dass Du Dich meldest!
Der Fehler ist in der aktuellen Version 2.0.0-beta.1 bereits behoben. Die Version ist bis auf die Dokumentation quasi fertig, so dass sie bedenkenlos genutzt werden kann. Wähle bei der Installation aus HACS einfach
[image]
Damit ist das Problem (und eine Menge anderer Dinge) behoben.

Lass gerne hören, ob Du erfolgreich warst. Falls Du weitere Hilfe benötigst: melde Dich gern!

P.S.: Mit dem Versionswechsel ändern sich die Namen der Entitäten grundlegend. Allerdings nur einmalig, so dass einmal ein wenig Anpassungsaufwand entsteht.

RedVision81, 2025-04-27:
Danke für die schnelle Antwort, werde ich die Tage dann mal einspielen, will heut nicht wieder alles umbauen müssen :)

closed 2025-04-27 by RedVision81