LarsK1/hass_solvis_control #282: [FEATURE] (Typ)-Änderung Sensor elektrische Wärmepumpenleistung

RedVision81, 2025-04-30:
Um die elektrische Wärmepumpenleistung ins Energie-Dashboard mit aufzunehmen, hatte ich mir einen Helfer -> Integralsensor erstellt umso die Summe errechnen lassen zu können, da der bisherige Sensor: "sensor.solvismax_elektrische_warmepumpenleistung" sich nicht direkt ins Energie-Dashboard einbinden hat lassen.

Der Sensor wurde in Version 2.0.1 nun umbenannt: "sensor.solvismax_warmepumpe_leistungsaufnahme_elektrisch" lässt sich aber leider nicht mehr mit dem Integralsensor verwenden und auch nicht direkt ins Energie-Dashboard einbinden.

Zur Aufzeichnung / Auswertung des tatsächlichen Stromverbrauchs der WäPu wäre das allerdings gut, wenn man ihn entweder wieder für den Integralsensor oder aber direkt ins Energie-Dashboard einbinden könnte.

LarsK1, 2025-05-01:
Hi @RedVision81 ,
rein vom Sensor her hat sich außer dem Namen nichts geändert. Ich kann bei mir auch einen Integralsensor dafür erstellen...

Probier es gerne noch einmal und gib dann Feedback.

RedVision81, 2025-05-01:
Hallo Lars,

danke für die schnelle Rückmeldung, hab gestern echt alles versucht, aber es wollte nicht klappen. Beim bestehenden Integralsensor konnte ich die neue Entität nicht auswählen. Habe den Helfer jetzt gelöscht und beim neu anlegen ging es jetzt plötzlich. 

Ist also wieder alles fein 😉👍

Gruß

Peter

closed 2025-05-01 by RedVision81