LarsK1/hass_solvis_control #283: [BUG] Wärmepumpe Heizstab zeigt Ein obwohl nicht aktiv

RedVision81, 2025-04-30:
**Version of the plugin**
2.0.1

 **Home-Assistant Version**
2025.4.4

**Enviroment (Docker, HAOS, ...)**
HAOS

**Version of your Solvis software/hardware**
Version NBG: 3.01.00
Version SC3: 3.19.47

**Describe the bug**
SolvisMax Wärmepumpe Heizstab Stufe 2 & 3 (A13) zeigt ein obwohl Aus:

[image]

[image]

LarsK1, 2025-05-01:
Hi @RedVision81 ,
vielen Dank für deine Meldung und herzlich Willkommen in der Solvis-Familie! 😀.

Zu deiner Meldung, bisher haben wir diese Werte nur theoretisch eingefügt, hatten aber noch keinen Besitzer mit dieser Komponente.

Könntest Du bitte einmal die Debug-Protokollierung einschalten und dann (wenn möglich) den Heizstab einschalten und nach ca. 2 Minuten wieder ausschalten.

Könntest Du uns zudem bitte noch die Diagnosedaten deiner HA-Instanz zukommen lassen?[image]

RedVision81, 2025-05-01:
Hallo Lars, 
kann die logs erzeugen , wenn ich wieder am PC sitze, am Handy ist das alles etwas mühsam 😉 aber ich glaube den Heizstab der WäPu kriege ich manuell gar nicht eingeschaltet. 

Beim Brennerstatus A12 habe ich übrigens ein ähnliches Phänomen, der zeigt auch EIN, obwohl mittlerweile gar kein Brenner mehr verbaut ist.

Gruß Peter

fkhr79, 2025-05-01:
Moinmoin @RedVision81 & @LarsK1 ,

wünsche einen schönen 1. Mai! 

Klasse, dass Du Dich gemeldet hast! Danke! - So bekommen wir an der Stelle hoffentlich endlich mal etwas mehr Licht in's Dunkel: die Dokumentationen bzw. Anlagenkonfigurationen sind an der Stelle höchst unterschiedlich. Für die weitere Lösung habe ich gerade mal die mir bekannten Fakten zusammengesucht und sammele sie hier:

- Hinzugefügt haben wir A13 im Rahmen der Bearbeitung von #138 
- Laut  [Anlagenschema SolvisMax - ALS-MAX7 Ver. 27350-2n](https://solvis-files.s3.eu-central-1.amazonaws.com/downloads-fk/solvismax7/27350_ALS-MAX-7.pdf) gibt es die folgenden Belegungen für A13
  - bei **SolvisBen / SolvisMax Gas/Öl und Gas-/Öl-Hybrid**
    - Öl / Öl-Hybrid: Brenner 2
    - Gas / Gas-Hybrid: unbenutzt
  - bei **Solvis Max Solo mit Lea/Lino/Mia/Pia oder Fremdkessel**
    - Lino / Solo: Ladepumpe PLAS Lino / Fremdkessel
    - WP / Pia: Heizstab Stufe 2 und 3 (->DHC2)
- In meiner älteren, analogen Version des Anlagenschemas (02.18 ALS-MAX-7) finde ich
  - bei **SolvisMAX Gas, Öl, FW und Solo**
    - alle: Ladepumpe LI-3/4 bzw. Fremdkessel (bei SolvisMax Solo) ODER Brenner 2 (bei SolvisMax Öl) ODER unbelegt

Es wäre daher super, wenn Du so freundlich wärest, uns die Konfiguration bzw. das Setup Deiner Anlage mitzuteilen.
Du schreibst "...obwohl mittlerweile gar kein Brenner mehr verbaut ist.". Heißt: Du hattest einen Brenner? ... und die Anlage ist umgerüstet worden?

> aber ich glaube den Heizstab der WäPu kriege ich manuell gar nicht eingeschaltet.

Vielleicht schon. Wenn Du das Installateursmenü aufrufst (Nutzerwechsel -> Installateur (Code 0064) -> Ausgänge -> 2. Seite) kannst Du die einzelnen Ausgänge auswählen und händisch schalten. Bei mir sieht das so aus (den kleinen Grafik-Fehler bitte ignorieren):

[image]

[image]

Vielleicht findest Du so heraus, wie bei Dir A13 belegt ist bzw. was passiert?!

RedVision81, 2025-05-01:
Hi zusammen, 
Werde versuchen euch die Infos zu schicken, und auch mal testen wegen Heizstab der WäPu. Mein Heizungsbauer schimpft mich nur immer wenn ich in den Installateur Modus gehe 😅

Ansonsten kurz zur Anlage:
Ist ein SolvisMax 750 mit SC3 vorher Öl und seit 1,5 Jahren jetzt Lea Premium, Brenner wurde ausgebaut. Solarthermie und Solvis Tim 6kw ist auch verbaut, aber den Steuer ich extern über 3 Shelly Pro 2 PM Relais, weil mir das Lummel Gedöns zu teuer war und nicht sinnvoll erschien.

Viele Grüße Peter

fkhr79, 2025-05-01:
> Hi zusammen, Werde versuchen euch die Infos zu schicken, und auch mal testen wegen Heizstab der WäPu. Mein Heizungsbauer schimpft mich nur immer wenn ich in den Installateur Modus gehe 😅

Er "schimpft" mit Dir, wenn Du in **Deiner** Anlage ein Softwaremenü aufrufst??? ... Unglaublich.

> Ansonsten kurz zur Anlage: Ist ein SolvisMax 750 mit SC3 vorher Öl und seit 1,5 Jahren jetzt Lea Premium, Brenner wurde ausgebaut. 

Okay: dann müsste laut Anlagenschema A13 bei Dir tatsächlich mit "Heizstab Stufe 2 und 3 (->DHC2)" belegt sein.

> Solarthermie und Solvis Tim 6kw ist auch verbaut, aber den Steuer ich extern über 3 Shelly Pro 2 PM Relais, weil mir das Lummel Gedöns zu teuer war und nicht sinnvoll erschien.

Das ist spannend! Was meinst Du mit "Lummel Gedöns"? Wie genau ist der Tim angeschlossen? Wie genau wird der gesteuert? - Danke Dir für eine kurze Erläuterung!!!

Gruß
Frank

RedVision81, 2025-05-01:
Naja der Installateur Modus ist ja nicht für den Endkunden gedacht, und wenn man dann wieder rumspielt und was verstellt und hinterher nix mehr geht, hab ich da schon verständnis. Bin selbst im IT-Bereich unterwegs, und man kennt einfach seine Pappenheimer :-D

Normal brauchst Du ja für die Heizstabsteuerung die Leistungselektronik / Leistungsmeser zum Anschließen. Da ist der Lummel verbaut, siehe auch Montageanleitung Solvis Tim im Fachkunden Downloadbereich. Das war mir aber wie gesagt zu teuer, zu altbacken und zu aufwendig. Daher hab ich es jetzt so gelöst, das ich jede Phase von dem Heizstab an ein Shelly Relais angeschlossen habe. Somit kann ich 2, 4 oder 6kw einzeln schalten, je nach PV Überschuss. Ist also Ergo ein dummer Heizstab, von dem die SC3 quasi gar nichts weis. 
Das ganze ist aktuell als Testphase in das Energiemanagement System EMMA von Huawei eingebunden, und die schaltet dann die 3 Shellys je nach PV-Überschuss einzeln ein. Werde das aber noch komplett auf Homeassistant umstellen, da ich ja dann, wenn der Speicher irgendwann mal 60-65 Grad erreicht hat die ja dann wieder ausgeschaltet kriegen muß. Zumal ja auch noch parallel die Solarthermie mitläuft und den Kessel hochheizt. Im Prinzip hätten wir den Heitstab auch gar nicht gebraucht, aber durch die ganzen Umbaumaßnahmen und Förderung wars auch schon egal :-D

RedVision81, 2025-05-01:
@LarsK1 @fkhr79 
hier jetzt mal die Log-Files, ich hoffe das hilft weiter.
Gegen 14:43/44 Uhr habe ich den A13 in HAOS deaktiviert, 5 Sekunden gewartet und dann wieder aktiviert.

[config_entry-solvis_control-01JT3QJ2TDTDNKXZPT8AJRSQ91.json](https://github.com/user-attachments/files/19999882/config_entry-solvis_control-01JT3QJ2TDTDNKXZPT8AJRSQ91.json)

[home-assistant_solvis_control_2025-05-01T12-44-18.968Z.log](https://github.com/user-attachments/files/19999883/home-assistant_solvis_control_2025-05-01T12-44-18.968Z.log)

fkhr79, 2025-05-01:
@RedVision81 

Dank Dir für die Rückmeldungen! Insbesondere auch für die Hinweise betr. SolvisTim! - Hätte ich genauso gelöst wie Du (mit den Shellys). 

Ansonsten ist an Deinem SC3-Screen erkennbar, dass A12 und A13 in Deinem System nicht installiert / genutzt / konfiguriert sind. Warum auch immer. - Ich bin mal auf ein System gespannt, bei dem A12 und A13 belegt sind...

Lange Rede, kurzer Sinn: die Entitäten sind nutzlos. Kannst Du deaktivieren.

> Gegen 14:43/44 Uhr habe ich den A13 in HAOS deaktiviert, 5 Sekunden gewartet und dann wieder aktiviert.

Die Deaktivierung der Entität in HA ist in den Logs nicht zu sehen. Gemeint war von Lars die anlagen-seitige (De)Aktivierung.
Brauchst Du m. E. aber jetzt nicht mehr machen: die Ausgänge sind ja nicht belegt.

@LarsK1 
Was meinst Du: wollen wir A13 standardmäßig deaktivieren? Wäre vermutlich meine Wahl bis wir Hinweise darauf haben, dass ihn jemand braucht (der ihn dann ja auch händisch aktivieren könnte).

LarsK1, 2025-05-02:
@fkhr79 Ja, ist vielleicht nicht die schlechteste Option. Andererseits kriegen wir dann evtl. kein Feedback ob (oder nicht :)) es funktioniert.

fkhr79, 2025-05-02:
@LarsK1 
Stimmt auch wieder.
Dann lassen wir es erstmal so und warten mal noch ein wenig, ob und ggf. was noch kommt. Können dann ja immer noch anpassen.
Habe das Ticket auf jeden Fall mal in der supported_entities vermerkt. Dann haben wir es auf dem Schirm...

fkhr79, 2025-05-02:
Thema ist damit erledigt -> Ticket geschlossen

closed 2025-05-02 by fkhr79