LarsK1/hass_solvis_control #284: [BUG] Update to 2.0 Verbindet sich nicht mit SC3

nevyen, 2025-04-30:
**Version of the plugin**
1.2.1

 **Home-Assistant Version**
25.4.4

**Enviroment (Docker, HAOS, ...)**
Docker

**Version of your Solvis software/hardware**
Version NBG: 3.2.1
Version SC3: MA 3.19.47

**Describe the bug**
Bei dem Versuch auf die Version 2.0 zu gehen, kann ich mich nicht mehr mit der SC3 verbinden. Als Fehler kommt immer wieder. "Fehler bei der Modbus Kommunikation "

Ein Downgrade zur Version 1.2.1 lässt die Verbindung wieder zu. 

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the Update-Guide ( Deleted -> Update -> Restart)
2. Ttry to add a new Konfiguration 
3. See error in the Screen with IP Config

**Expected behavior**
Create a new "Device" without any issues. 

**Additional context**

LarsK1, 2025-05-01:
Hallo @nevyen ,
vielen Dank für deine Meldung! Kannst Du bitte versuchen deine Heizung einmal komplett auszuschalten, ca. 1 Minute warten, dann wieder einschalten. Nun sollte auch die Verbindung wieder klappen.

Ansonsten müsstest Du bitte einmal die Logs mitschicken (siehe README)

nevyen, 2025-05-01:
Moin und vielen Dank für die schnelle Rückmeldung. 

Wenn ich das nächste Mal Zeit finde, werde ich das direkt probieren. 

Ich muss nur etwas mehr Zeit einplanen, da alle Skripte angepasst werden müssen. 

VG

Dominic Warzok
Adelebser Straße 11
37139 Adelebsen

Tel.: 05506 9993714
Mobil: 0171 2014463

nevyen, 2025-05-03:
Ich habe mich nochmal dran gesetzt. 

Das Log gibt folgendes aus: 

```2025-05-03 19:16:58.981 INFO (MainThread) [custom_components.solvis_control.config_flow] Initialize config flow for solvis_control
2025-05-03 19:17:02.478 DEBUG (MainThread) [custom_components.solvis_control.config_flow] calling get_mac for 192.168.15.30
2025-05-03 19:17:05.494 DEBUG (MainThread) [custom_components.solvis_control.config_flow] get_mac returned: None```

Die Methode scheint mit dem 2.0 Release neu zu sein. Deswegen wird auch das 1.2.1 noch so problemlos funktionieren. 

Wenn ich bei meinem Windows PC - `arp -a` aufrufe, dann bekomme ich eine MAC zu der IP 
Auf dem Docker Host auch.
ABER in dem Container nicht. Das ist auch total sinnvoll. Der Container läuft nicht als Network-Host sondern in seinem eigenen Netzwerk. Dieses gibt natürlich nur den Port für Homeassistant frei. Ein ARP wird also nichts anderes finden als sich selbst.

Vielleicht kann man die alte Methode ja wieder nutzen. Oder seine MAC händisch eingeben, damit der Fehler nicht kommt. 

VG
Nevyen

LarsK1, 2025-05-03:
Hallo Nevyen (@nevyen ),
danke für deine Analyse - wir hatten dieses Thema schon beim Programmieren hin und her diskutiert, hatten uns aber aufgrund der HA Vorgaben eigentlich dagegen entschieden. Wenn das natürlich aber so ausartet....

Hmm, ich denke nocheinmal über eine mögliche Lösung nach...

@fkhr79 meinst Du es wäre sinnvolle, wenn wir als Optionsfeld (nur für solche Fälle) die MAC eingebbar machen und dann die arp-Prüfung wegfällt?

nevyen, 2025-05-03:
Ich habe gerade einen Pull-Request eingestellt. Keine Ahnung ob dass das Problem löst. 
Dafür habe ich zu wenig Ahnung von den HA Vorgaben. Und auch nicht so richtig von Python.

fkhr79, 2025-05-03:
@LarsK1 
Ich überlege auch schon die ganze Zeit, wie wir das lösen. Haben wir ja in #207 schon kommen sehen. 

Habe gerade auch nochmal geschaut, ob wir per Modbus einen Identifikator oder Seriennummer oder vergleichbar bekommen können. Modbus bietet dafür die `read_device_information`-methode. Leider (und wenig überraschend) unterstützt Solvis diesen Modbus-Standard nicht (hatte das gerade mal fix in die diagnostics eingebaut - wirft leider einen fehler).

Problem an der UUID-Lösung von @nevyen ist, dass doppelte Einträge des gleichen Geräts nicht verhindert werden... und das ist ja eigentlich der Zweck, den HA mit der Vorgabe verfolgt.

Laut https://developers.home-assistant.io/docs/config_entries_config_flow_handler/#unique-id-requirements käme noch die Verwendung der Geolocation in Frage. Wäre eine Variante.

Schlussendlich noch die vorgeschlagene Lösung mittels Eingabe der MAC per config flow. 
Das finde ich eher unschön und nicht mehr niedrigschwellig für bestimmte Nutzergruppen. - Ich selbst hätte auch keine Lust, die MAC herauszusuchen und einzugeben... nur für einen Identifikator.

Fazit:
Wenn auch nicht schön oder elegant... aber ich würde der Einfachheit halber tatsächlich einfach die Geolocation nutzen. Ist schnell implementiert und zumindest ein wenig mehr als eine rein zufällige UUID.

nevyen, 2025-05-04:
Guten Morgen,

wie gesagt ich hatte quasi keine Ahnung wo für das ist. Eure Erklärungen dazu sind aber verständlich. 

Aber wer hat schon mehr als einen Max zu Hause stehen?

Bei der Geolocation hätte ich bedenken, dass die nicht genau genug ist. Wenn ich mir noch eine Gerät z.B. auf dem Hof in ein Nebengebäude oder z.B. Poolhaus Stelle, dass dann dort ein Konflikt entsteht. 

Müsste nicht die IP oder der Hostname als Identifyer reichen. Meine Solvis hat als Hostname z.B. `SOLVIS-546FBD` ich würde davon ausgehen, dass das 546FBD eindeutig ist. 

Und im selben Netzwerk kann keine zweite Solvis Anlage mit der Selben IP existieren. 

Man könnte die IP ja auch nur nutzen, wenn man keine Mac bekommt.

LarsK1, 2025-05-04:
@fkhr79 meine Überlegung wäre wie folgt. Bei den allermeisten Setups geht die MAC-Erkennung ja, daher hätte ich das erst einmal so gelassen und wenn ein Fehler aus der MAC-Erkennung heraus entsteht, ein Muss-Feld für die MAC-Adresse im ConfigFlow angezeigt. (Das müsste für mein Verständnis eigentlich gehen). Dann haben definitiv eine UniqueID und müssen auch sonst keine übermäßig großen Verrenkungen machen.

Was hälst Du davon?

fkhr79, 2025-05-04:
@LarsK1 
Gehen tut das in jedem Fall. Mein "Problem" damit ist letztlich die zusätzliche Flow-Komplexität nebst Test-Komplexität (letztendlich der Aufwand dafür)... und das für ein so kleines Problem. Vielleicht male ich aber auch gerade schwarz und es ist nicht so schlimm... (bin viellleicht noch ein wenig wegen der ganzen Tests sensibilisiert, die mich ganz schön angenervt haben).
Wie wäre es, standardmäßig ein Pflichtfeld für die MAC vorzusehen, welches - falls die Abfrage funktioniert - vorbefüllt wird. Wie bei den HKR auch. Das wäre - glaube ich - einigermaßen einfach machbar und entspräche fast Deiner Vorstellung.

LarsK1, 2025-05-04:
@fkhr79 Ja, das ginge auf jeden Fall. Das bräuchten nur halt noch einen Schritt - weil wir können ja nicht das Feld befüllen, ohne die IP-Adresse zu kennen. Und nach meiner Kenntnis geht das nur bei einem Fehler, oder wenn der Absenden-Button gedrückt wird.
Oder hast Du da andere Kenntnisse?

fkhr79, 2025-05-04:
@LarsK1 
Ach... verfl... - Ja, Du hast Recht: wir machen das ja schon im ersten Step. Ich war gedanklich einen Schritt weiter...

Dann halt volles Programm: bei MAC-Fehler ein neuer (weiterer) schritt, in dem die MAC eingegeben werden muss? ... oder ein erweitertes bzw. anderes data_schema im gleichen Step?

nevyen, 2025-05-04:
Ich will ja kein Spielverderber sein, aber wenn man die Mac selber eingeben kann, dann ist das ja nicht besser als die IP Adresse. Dann kann ich das selbe Gerät ja nochmal konfigurieren und denke mir nur eine neue Mac aus.

fkhr79, 2025-05-04:
> Ich will ja kein Spielverderber sein, aber wenn man die Mac selber eingeben kann, dann ist das ja nicht besser als die IP Adresse. Dann kann ich das selbe Gerät ja nochmal konfigurieren und denke mir nur eine neue Mac aus.

Naja... es ist so nahe an einer "korrekten" Lösung wie möglich. - Wie schon geschrieben: wirklich befriedigend ist das alles nicht... aber es wird eine funktionierende Lösung geben.
Solltest Du weitere Ideen haben, die besser funktionieren und HA-konformer sind: gerne jederzeit her damit!

nevyen, 2025-05-05:
Ideen hoffentlich sinnvoll nach Machbarkeit und Sinnigkeit sortiert: 

### Hostname: 

Ich habe nochmal in die Doku geschaut: 
> **Sometimes acceptable sources for a unique ID for local devices**
> Hostname: If a subset of the hostname contains one of the acceptable sources, this portion can be used
> 
Wenn meine Annahme von oben richtig ist, dürfte man laut den Regeln den Teil vom Hostname benutzen. 
Der Hostname kann ja auch nicht in der Anlage angepasst werden. Damit sollte das den Vorgaben entsprechen. 

Ja irgendwie kann ich in meiner Fritzbox den Hostname ändern. Keine Ahnung was dann passiert, weil es ändert sich ja nicht der "originale" Hostname. Aber wenn man das auf die Spitze treibt, kann ich mit entsprechender Ahnung auch eine MAC fälschen. 

> **Unacceptable sources for a unique ID**
> IP Address
> Device Name
> Hostname **if it can be changed by the user** 
> URL
 Den Hostname kann man nicht in der Anlage anpassen. Damit erfüllt er die Kriterien für den Part oben 

### Solvis Portal

Eventuell kann man irgendwie aus dem Solvis Portal die Anlagen-Id auslesen:
Diese wird zumindest im graphQL per JSON übermittelt.
Könnte allerdings sein, dass man sich dann noch in HA im Portal anmelden muss. Das wird zum programmieren wahrscheinlich ziemlich 👎 

### Solvis Analagen ID
Wenn man es nicht automatisch auslesen kann, könnte man diese eventuell eher in der Eingabemaske nutzen, als die MAC-Adresse. 
Die ID kann einfach aus dem Portal kopiert werden und das sollte niederschwelliger sein, als die MAC heraus zu finden. 

### MAC-Adresse des Hosts nutzen
Man könnte die MAC des Hosts mit einem vorangestellten SOLVIS-[HOST-MAC] nutzen. Damit hätte man aber das Problem, dass man immer nur eine Anlage konfigurieren kann.

PS: 
Vielen Dank euch beiden, das ihr mit so viel Mühe guten Support und allem was dazu gehört so eine geniale Integration anbietet. 
Jetzt müssen wir nur noch Solvis überreden, dass sie ein wenig mehr Zeit in ihren Modbus investieren.

LarsK1, 2025-05-05:
@nevyen die Änderungen sind nun vorgenommen. Ich warte nur noch auf @fkhr79 (zwei paar Augen sehen immer mehr als ein paar :)). Dann sind die in der Master-branch. Bzw. im nächsten Release mit drinnen.

Damit kannst Du dann die MAC-Adresse selber setzen und damit den Fehler umgehen. 😀

nevyen, 2025-05-05:
Krass bist du schnell 👍 

Dann bin ich mal gespannt und freue mich auf das nächste Release :-)

nevyen, 2025-05-05:
Ich habe auch eine Blick auf die Änderung geworfen. Von dem was ich bisher verstehe sieht das ganz gut aus. 

Vielen Dank nochmal für eure ganze Arbeit.

fkhr79, 2025-05-05:
So, Review ist durch. Sollte zügig im Release landen. 🙂 

> Jetzt müssen wir nur noch Solvis überreden, dass sie ein wenig mehr Zeit in ihren Modbus investieren.

Ich nervte zuletzt "meinen" Servicetechniker damit, doch bitte einen Kontakt zu den Entwicklern (bzw. DEM Entwickler) herzustellen: tatsächlich hatten wir schon ein paar Dinge gefunden, die ihm hätten weiterhelfen können - meine ich zumindest. Leider bis jetzt ohne Erfolg: ich warte auf den nächsten Besuch (steht noch aus). Da werde ich wieder erinnern... 😈 
Ansonsten "hört man", dass Solvis gerade GANZ andere Sorgen als die Modbus-Verbindung hat...: nicht ohne Grund ist in die Ver. 3.20.16 z. B. ein neuer Statusscreen mit Diagnosedaten eingezogen. Angeblich gibt es unerklärbare Reboots.
Aber: geben wir die Hoffnung nicht auf! - Vielleicht versteht Solvis ja irgendwann, dass sie durch eine Zusammenarbeit nur gewinnen können und die Modbus-Schnittstelle für viele Nutzenden von elementarer Bedeutung ist...

closed 2025-05-08 by LarsK1

nevyen, 2025-05-08:
Works like a charm :-) 

Vielen Dank euch beiden