LarsK1/hass_solvis_control #285: [BUG] Schreibfehler in Grafik & inkonsistente Benamung der Integration

fkhr79, 2025-05-01:
**Version of the plugin**
2.0.1

**Describe the bug**
In der neuen Grafik fehlt ein "l", ist ein "r" zu viel und ein weiteres "r" ist ... "irgendwie komisch":
[image]

Im Zusammenhang damit:
In der Grafik heißt die Integration "Solvis Control Integration for Home Assistant".
Die erste Überschrift in der readme.md benennt die Integration als "Solvis Heating Integration for Home Assistant".
Im HACS heißt die Integration "Solvis Control".

Der Name sollte konsistent genutzt werden. M. E. ist "Solvis Control" als Name (quasi "Produktname") prima. Als "Unterüberschrift" könnte man ergänzen "Home Assistant Integration for Solvis Heating devices".

fkhr79, 2025-05-31:
@LarsK1 

Ich habe mal die Grafik ändern lassen:

[image]

Falls Du einverstanden bist, könnten wir die austauschen...?

LarsK1, 2025-06-01:
Hi @LarsK1 ,
sorry, dass gerade alles etwas länger dauert....

Jay bin ich völlig fein damit 😊